awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #85: gst_element_factory_make ("nvcamerasrc", NULL); returns NULL

sergeystoyan, 2018-06-14:
Hi,

I'm developing a kinesis video producer for nvidia TX2 camera. They use their driver nvcamerasrc instead of v4l2src. This driver was tested with gst-launch-1.0 e.g.

`gst-launch-1.0 nvcamerasrc fpsRange="60 60" intent=3 ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)I420, framerate=(fraction)30/1' ! omxh264enc control-rate=2 bitrate=6000000 ! h264parse ! flvmux streamable=true ! queue ! filesink location=test2.mkv`

It works well.

So I installed amazon-kinesis-video-streams-producer-sdk-cpp package and in kinesis_video_gstreamer_sample_app.cpp in gst_element_factory_make replaced "v4l2src" with "nvcamerasrc". But it cannot create it and returns NULL.

What is the app missing about nvcamerasrc? Can you give an idea why gst-launch-1.0 can see nvcamerasrc but the sample app cannot?
Thank you!

sergeystoyan, 2018-06-14:
Yes, gst-launch-1.0 was already installed.

#1 I copied libgstnvcamera.so which is located in usr/lib/aarch64-linux-gnu/gstreamer-1.0 to <...>/kinesis-video-native-build/downloads/local/lib

no help

#2 I did 
export LD_LIBRARY_PATH=/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp-1.4.1/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH:/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0

(where tegra contains nvidia libs)
no help

#3 I did 
export GST_PLUGIN_PATH=/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0

where tegra contains nvidia libs

and it worked!

Thank you!

unicornss, 2018-06-14:
Closing this issue as it has been resolved.

closed 2018-06-14 by unicornss

sergeystoyan, 2018-06-14:
BTW here all the issues I stumbled over while deploying Kinesis SDK to TX2 camera system. Hope it will be helpful for somebody else.

Installation and running amazon-kinesis-video-streams-producer-sdk-cpp1.4.1 ISSUES:

INSTALLATION ISSUES:

#1

'__NR_eventfd' undeclared

remedy: 

in install-script to command ./config add no-afalgeng flag: 

>./config  no-afalgeng --prefix=$DOWNLOADS/local/ --openssldir=$DOWNLOADS/local/

#2

checking build system type... ./config.guess: unable to guess system type

remedy:

in install-script after command 
    

>cd $DOWNLOADS/m4-1.4.10	

add

>wget -O config.guess 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD'
>wget -O config.sub 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD'

in order to overwrite the obsolete scripts

#3

curl: (77) error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem

remedy:

sudo mkdir -p /etc/ssl
sudo cp /etc/ssl/certs/ca-certificates.crt /etc/ssl/cert.pem

RUNNING APPLICATION ISSUES:

#4

error while loading shared libraries: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory
 
remedy:

>export LD_LIBRARY_PATH=/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp-1.4.1/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH
//export LD_LIBRARY_PATH=/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp-1.4.1/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH:/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0
//export LD_LIBRARY_PATH=/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0

#5

gst_element_factory_make ("nvcamerasrc", NULL); returns NULL 

remedy:

>export GST_PLUGIN_PATH=/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0 
//>export GST_PLUGIN_PATH=/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0

MushMal, 2018-06-14:
@sergeystoyan Thanks a lot for reporting these issues! BTW, you could send a PR :)

yuxiny2586, 2020-03-09:
> BTW here all the issues I stumbled over while deploying Kinesis SDK to TX2 camera system. Hope it will be helpful for somebody else.
> 
> Installation and running amazon-kinesis-video-streams-producer-sdk-cpp1.4.1 ISSUES:
> 
> INSTALLATION ISSUES:
> 
> #1
> 
> '__NR_eventfd' undeclared
> 
> remedy:
> 
> in install-script to command ./config add no-afalgeng flag:
> 
> > ./config  no-afalgeng --prefix=$DOWNLOADS/local/ --openssldir=$DOWNLOADS/local/
> 
> #2
> 
> checking build system type... ./config.guess: unable to guess system type
> 
> remedy:
> 
> in install-script after command
> 
> > cd $DOWNLOADS/m4-1.4.10
> 
> add
> 
> > wget -O config.guess 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD'
> > wget -O config.sub 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD'
> 
> in order to overwrite the obsolete scripts
> 
> #3
> 
> curl: (77) error setting certificate verify locations:
> CAfile: /etc/ssl/cert.pem
> 
> remedy:
> 
> sudo mkdir -p /etc/ssl
> sudo cp /etc/ssl/certs/ca-certificates.crt /etc/ssl/cert.pem
> 
> RUNNING APPLICATION ISSUES:
> 
> #4
> 
> error while loading shared libraries: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory
> 
> remedy:
> 
> > export LD_LIBRARY_PATH=/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp-1.4.1/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH
> 
> #5
> 
> gst_element_factory_make ("nvcamerasrc", NULL); returns NULL
> 
> remedy:
> 
> > export GST_PLUGIN_PATH=/usr/lib/aarch64-linux-gnu/tegra/:/usr/lib/aarch64-linux-gnu/gstreamer-1.0

This post was extremely helpful when I was setting up my Jetson Xavier. Thanks a lot @sergeystoyan

MushMal, 2020-03-09:
Folks, this is all fantastic info. We strongly encourage sending pull requests with changes that can get merged into master branch. This information can go into a section in Readme for example