awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #87: Pushing live video from pc webcam to kinesis stream

anmol1ratn, 2018-06-15:
@chehefen  and @MushMal. The problem which I'm encountering while pushing the live feed from my webcam is:

INFO - 
getStreamingTokenResultEvent(): Get streaming token result event.
DEBUG - streamReadyHandler invoked
DEBUG - Kinesis Video stream my-stream is Ready.
**DEBUG - Stream is ready**
**Unable to set the source to ready state.**
INFO - 
freeKinesisVideoClient(): Freeing Kinesis Video Client
INFO - 
commonHeapDebugCheckAllocator(): Heap is initialized
INFO - 
commonHeapDebugCheckAllocator(): Heap limit: 				536870912
INFO - 
commonHeapDebugCheckAllocator(): Heap size: 				0
INFO - 
commonHeapDebugCheckAllocator(): Number of allocations: 				0
INFO - 
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				(nil)
INFO - 
aivHeapDebugCheckAllocator(): *******************************************
INFO - 
aivHeapDebugCheckAllocator(): *******************************************
INFO - 
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x7f9c41b7a010
INFO - 
aivHeapDebugCheckAllocator(): *******************************************
INFO - 
aivHeapDebugCheckAllocator(): Block:	0x7f9c41b7a010		size:	536870879
INFO - 
aivHeapDebugCheckAllocator(): *******************************************
INFO - 
heapRelease(): Freeing native heap.
INFO - Freeing Kinesis Video Stream my-stream
INFO - 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
INFO - Curl shutdown.

Can you please tell me how to set the source to ready state? Which file do I have to modify to push live video from my pc webcam to the kinesis stream and what do I have to modify in those files to get it working?

**PS: I'm using Lubuntu 18.04 OS**

unicornss, 2018-06-15:
Hi @anmol1ratn  

Could you   `export GST_DEBUG=4` before running the demo application. This will provide detailed logs on camera source and its supported resolutions.

You can also try using the command line options (as given in the [Readme](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md#1-gstreamer-webcam-demo-application) ).

anmol1ratn, 2018-06-15:
@unicornss, thanks for your prompt response. I did what you said and here's the result:

DEBUG - Kinesis Video stream my-stream is Ready.
DEBUG - Stream is ready
DEBUG - streamReadyHandler invoked
0:00:03.912912360  1894 0x55cebffebc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/anmol/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:03.912957195  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "capsfilter" named "source_filter"
0:00:03.913169320  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x55cec00800e0> adding pad 'sink'
0:00:03.913201524  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x55cec00800e0> adding pad 'src'
0:00:03.913268588  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "capsfilter" named "encoder_filter"
0:00:03.913296916  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x55cec0080420> adding pad 'sink'
0:00:03.913321825  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x55cec0080420> adding pad 'src'
0:00:03.920567196  1894 0x55cebffebc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/anmol/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstapp.so" loaded
0:00:03.920623082  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "appsink" named "appsink"
0:00:03.920934469  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSink@0x55cec01beed0> adding pad 'sink'
0:00:04.069629685  1894 0x55cebffebc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/anmol/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstvideoparsersbad.so" loaded
0:00:04.069753319  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "h264parse" named "h264parse"
0:00:04.069906063  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x55cec01f1460> adding pad 'sink'
0:00:04.069978176  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x55cec01f1460> adding pad 'src'
0:00:04.070074379  1894 0x55cebffebc00 INFO               baseparse gstbaseparse.c:3944:gst_base_parse_set_pts_interpolation:<GstH264Parse@0x55cec01f1460> PTS interpolation: no
0:00:04.070138273  1894 0x55cebffebc00 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "vtenc_h264_hw"!
0:00:04.070190768  1894 0x55cebffebc00 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "omxh264enc"!
0:00:04.096084308  1894 0x55cebffebc00 INFO                 x264enc gstx264enc.c:2894:plugin_init: linked against x264 build: 148
0:00:04.116683396  1894 0x55cebffebc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/anmol/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstx264.so" loaded
0:00:04.116816925  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "x264enc" named "encoder"
0:00:04.117054962  1894 0x55cebffebc00 INFO                 x264enc gstx264enc.c:224:gst_x264_enc_add_x264_chroma_format: 8-bit depth supported
0:00:04.117216410  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@0x55cec01fbbd0> adding pad 'sink'
0:00:04.117303981  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@0x55cec01fbbd0> adding pad 'src'
0:00:04.147914809  1894 0x55cebffebc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/anmol/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:04.147950257  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "v4l2src" named "source"
0:00:04.148718327  1894 0x55cebffebc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSrc@0x55cec0210120> adding pad 'src'
0:00:04.148746440  1894 0x55cebffebc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "pipeline" named "test-pipeline"
0:00:04.148830558  1894 0x55cebffebc00 WARN                    v4l2 **v4l2_calls.c:612:gst_v4l2_open:<source> error: Cannot identify device '/dev/video0'.**
0:00:04.148839036  1894 0x55cebffebc00 WARN                    v4l2 v4l2_calls.c:612:gst_v4l2_open:<source> error: system error: No such file or directory
[... cut ...]
INFO - 
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x7f11c2f2e010
INFO - 
aivHeapDebugCheckAllocator(): Block:	0x7f11c2f2e010		size:	536870879
INFO - 
heapRelease(): Freeing native heap.
INFO - Freeing Kinesis Video Stream my-stream
INFO - 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
INFO - Curl shutdown.

Please tell which file do I have to modify to set the source correctly and how to configure this source object so that I will be able to push live feed from my pc webcam?

unicornss, 2018-06-15:
The demo application is trying to access the video from the source at [/dev/video0]( https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L453). You can update the device based on your setup. 

Could you tell us which camera and OS you are trying to use (USB camera? or webcam?); also details on the camera & laptop model will help us debug your issue.

You can run `ls -ltr /dev/video*` to see whether your video device is detected.

You can also refer past issues in this repository (closed) for similar questions and debug steps e.g https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/33 https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/20

anmol1ratn, 2018-06-15:
@unicornss 

I'm using hp laptop, more specifically hp **ay011tx**. I want to stream with the built in camera that comes with the laptop and I'm using Lubutnu 18.04 OS inside the vbox (using vbox should not make a difference in my opinion).

I tried `ls -ltr /dev/video* ` but the output was `cannot access '/dev/video*': No such file or directory`.

MushMal, 2018-06-15:
@anmol1ratn not sure if we tried the setup you describe but could you ensure the device is exposed in the virtual OS. Not sure what virtualization software you are using but there might be options to enable the USB connected devices to the guest OS.

unicornss, 2018-06-20:
As @MushMal has mentioned that the media pipeline should be available for getting the video to be sent to Kinesis  If video source (webcam) is not available within the virtual box, then you can try IP cameras (RTSP demo ) or USB cameras.

MushMal, 2018-06-21:
@anmol1ratn have you been successful streaming on your guest OS?

anmol1ratn, 2018-06-30:
@MushMal and @unicornss , I apologize for the late reply.
I'm able to successfully push live stream from pc webcam after dual booting my pc with ubuntu. I think the problem was that Virtual Box was not providing the support for underlying hardware to the guest OS. Also, I ordered the IP camera and it has just arrived. I'll be testing that too. :)

Thanks, @unicornss  and @MushMal

closed 2018-06-30 by anmol1ratn