awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #88: Building the container image under ubuntu

roccomuso, 2018-06-18:
The docker build stops with this error

```
       # ------------------------- GSTREAMER - for demo application --------#
        GStreamer, GStreamer plugins, yasm, gst-libav and x264.
        ------------------------------------------------------------------------
        *** NOTE *** for MacOS High Sierra: Use 
            brew install autoconf  
            brew install bison before running this script
        ------------------------------------------------------------------------
        ========================================================================
        This install script downloads and builds the dependent libraries from  
        source in the ./kinesis-video-native-build/downloads directory only. 
        ========================================================================
 

      Y - Confirm to proceed with Producer SDK build 
      N - Exit Producer SDK build

Enter selection: The command '/bin/sh -c ./install-script a' returned a non-zero code: 1
```

roccomuso, 2018-06-18:
Is this a typo? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docker_native_scripts/Dockerfile#L33

There's a missing dash peraphs?
`RUN ./install-script a` should be `RUN ./install-script -a`

unicornss, 2018-06-18:
@roccomuso  Thanks for noticing, yes its a typo; we will update the Dockefile to include the latest install-script change and other platforms soon.

roccomuso, 2018-06-18:
@unicornss btw would be good to differentiate between a dev container and a prod container, right now it's a 2.8GB container. It's too big.

Again, do you know if there's a way to do a producer with Node.js? https://github.com/aws/aws-sdk-js/issues/2093

unicornss, 2018-06-18:
@roccomuso thanks for the suggestion. We will try to release two separate containers.

Regarding the Node.js support, we only have CPP and Java for Producer SDK for now. Could you help us understand your use case and environment - how do you plan to send video data through Node.js?

roccomuso, 2018-06-18:
For example I'm using node and the module `node-fluent-mmpeg` to get a rtsp video and get a raw stream. Then I'd like to push that mvk video stream to kinesis-video.

It should be possible to process the frames with a `Transform` stream and do a `PUT_MEDIA` call.
In a way like the standalone kinesis does, with the `putRecords` method.

But right now as u can see from the issue 2093 on aws-sdk-js, the put media method is not available.
The best node api would be to have a `createReadStream` to get the video stream and a `createWriteStream` to have a writable stream to write your video to.
Something like [kinesis-streams](https://github.com/crccheck/kinesis-streams) but for kinesis-video.

unicornss, 2018-06-19:
@roccomuso  Could you elaborate what you mean by  

> This the built files and dir i'm taking from the builder image,

The log configuration file should exist  in the folder -  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/kvs_log_configuration

roccomuso, 2018-06-19:
I'm trying to build a new image without the dev dependencies.
Only using the built binaries so that I can have a "lite" container. The original one is over 2.8GB.

unicornss, 2018-06-20:
@roccomuso if you would like to build the image with only built libraries and binaries, then you can try removing subfolders within kinesis-video-native-build/downloads except local. Example:

```
WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
RUN rm -rf $(ls -I local)
```
and try building docker image using `docker build --squash -t yourimagetag` .

There are few related postings you can refer to : https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/71  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/57

roccomuso, 2018-06-20:
Yeah but it's still too big `1.4GB`.
I'm trying to get just the required files but i'm having this error:

```
(kinesis_video_gstreamer_sample_rtsp_app:7): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Not all elements could be created:
	rtspsrc (gst-plugins-good)
	rtph264depay (gst-plugins-good)
	appsink (gst-plugins-base)
[INFO ][2018-06-20 11:23:00] 
freeKinesisVideoClient(): Freeing Kinesis Video Client
[INFO ][2018-06-20 11:23:00] 
commonHeapDebugCheckAllocator(): Heap is initialized
[INFO ][2018-06-20 11:23:00] 
commonHeapDebugCheckAllocator(): Heap limit: 				536870912
[INFO ][2018-06-20 11:23:00] 
commonHeapDebugCheckAllocator(): Heap size: 				0
[INFO ][2018-06-20 11:23:00] 
commonHeapDebugCheckAllocator(): Number of allocations: 				0
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				(nil)
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x7fa5aa5ff010
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): Block:	0x7fa5aa5ff010		size:	536870879
[INFO ][2018-06-20 11:23:00] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-06-20 11:23:00] 
heapRelease(): Freeing native heap.
[INFO ][2018-06-20 11:23:00] Freeing Kinesis Video Stream test-stream
[INFO ][2018-06-20 11:23:00] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[INFO ][2018-06-20 11:23:00] Curl shutdown.
```

unicornss, 2018-06-20:
The error

> Not all elements could be created:
> 	rtspsrc (gst-plugins-good)
> 	rtph264depay (gst-plugins-good)
> 	appsink (gst-plugins-base)

suggests that it does not have the required GStreamer libraries or cant find it in the LD_LIBRARY_PATH / GST_PLUGIN_PATH.
You could remove the plugins that are not required for your use case as  the docker  build example is generic for all users. you are welcome to send pull request for optimal docker build just for RTSP runtime. 

Another _suggestion_ is that you could extract the built libraries and dependencies
` 
RUN tar -cvf kinesis-video-producer-sdk-libs.tar.gz /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib
`

`
RUN tar -cvf kinesis-video-producer-sdk-demo.tar.gz /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/kinesis*  
 `

and copy them to host;  then create another Docker build (your custom Dockerfile) directly with these pre-built binaries.

Example:

`docker run --rm -iv${PWD}/sdk:/sdk sdk16-04 sh -s << EOF`

`cp -vd kinesis-video-producer-sdk-libs.tar.gz /sdk/kinesis-video-producer-sdk-16-04-libs.tar.gz`

`cp -vd kinesis-video-producer-sdk-demo.tar.gz /sdk/kinesis-video-producer-sdk-16-04-demo.tar.gz`
`EOF`

Hope this helps.

unicornss, 2018-07-13:
Closing this issue for now as it has been answered. If further support is required please reopen.

closed 2018-07-13 by unicornss

rmtuckerphx, 2023-01-25:
> For example I'm using node and the module `node-fluent-mmpeg` to get a rtsp video and get a raw stream. Then I'd like to push that mvk video stream to kinesis-video.
> 
> It should be possible to process the frames with a `Transform` stream and do a `PUT_MEDIA` call. In a way like the standalone kinesis does, with the `putRecords` method.
> 
> But right now as u can see from the issue 2093 on aws-sdk-js, the put media method is not available. The best node api would be a `createReadStream` to get the video stream and a `createWriteStream` to have a writable stream to write your video to. Something like [kinesis-streams](https://github.com/crccheck/kinesis-streams) but for kinesis-video.

@roccomuso Did you ever find a workflow that worked for Node?