awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #89: Log level is not set, always DEBUG messages are seen.

sergeystoyan, 2018-06-18:
In kinesis_video_gstreamer_sample_app.cpp im main() I set
`PropertyConfigurator::doConfigure("kvs_log_configuration");`
where kvs_log_configuration is the following: 
`og4cplus.rootLogger=INFO, KvsConsoleAppender

#log4cplus.logger.MemoryCheck=INFO, KvsConsoleAppender

#KvsConsoleAppender:
log4cplus.appender.KvsConsoleAppender=log4cplus::ConsoleAppender
log4cplus.appender.KvsConsoleAppender.layout=log4cplus::PatternLayout
log4cplus.appender.KvsConsoleAppender.layout.ConversionPattern=[%-5p][%d] %m%n `

and still there are messages like 
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0

Then I added 
`LOG_CONFIGURE_STDOUT("INFO");`
	
and even
` Logger::getRoot().setLogLevel(0);`
       
but still debug messages are seen. Please advise how they can be suppressed?
Thank you!

chehefen, 2018-06-18:
Hi, we already did `PropertyConfigurator::doConfigure("kvs_log_configuration");` in `gstreamer_init()` in kinesis_video_gstreamer_sample_app.cpp. If you start from a clean workspace, you just need to set `log4cplus.rootLogger=INFO, KvsConsoleAppender` in kinesis-video-native-build/kvs_log_configuration and then run the install-script again. You should see the modified log level.

sergeystoyan, 2018-06-19:
I did what you said in your previous post and
`make kinesis_video_gstreamer_sample_app.cpp`
to recompile the sample. But no change in logging. 

Is running the whole install-script the only way to make it work? Logger.h seems not to be changed since I deployed the SDK so why I have to recompile everything?
Maybe I just need to destroy some caches?

unicornss, 2018-06-19:
Could you try running `rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles` then run `./install-script `

As mentioned in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md 

> The sub-sequent run will be building just the modified SDK components or applications which is much faster.

unicornss, 2018-07-27:
Closing this issue as it has been answered. Please reopen if needed. Thanks.

closed 2018-07-27 by unicornss