awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #92: Segmentation fault in libgstaccurip.so after GStreamer plugin installation

stylate, 2018-06-21:
Hello, after running ./install-script and following instructions based on [https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html](url), I get the following error:

```
ERROR: Caught a segmentation fault while loading plugin file:
/Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstaccurip.so

Please either:
- remove it and restart.
- run with --gst-disable-segtrap --gst-disable-registry-fork and debug.
```

Would this indicate that something went wrong after running ./install-script, leading to a segmentation fault in libgstaccurip.so? As a workaround, I was hoping to be able to use my own GStreamer instead of the one installed from the script and to manually install the `kvssink` plugin, but this option seems to be of no avail.

stylate, 2018-06-21:
Sure thing! My OS is macOS High Sierra 10.12.6. The error occurs on any command that I run related to gstreamer, so running something as simple as `gst-inspect-1.0` would trigger this error only if I had already set the `GST_PLUGIN_PATH` and/or `LD_LIBRARY_PATH`. 

The main folder was cloned from this repository, but renamed to 'cpp' for convenient access.

unicornss, 2018-06-22:
Could you tell us if you were able to launch the `gst-inpsect-1.0` from the `kinesis-video-native-build` directory or `/Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/bin/gst-inspect-1.0`?

Also there are demo applications in the `kinesis-video-native-build` directory. Were you able to start streaming with those sample applications `kinesis_video_gstreamer_sample_app` or `kvs_producer_plugin_demo`  ?

stylate, 2018-06-22:
Before setting my `GST_PLUGIN_PATH`, I am able to call `gst-inspect-1.0`, as well as any other gstreamer command as deemed appropriate in both folders. After setting my `GST_PLUGIN_PATH` to `/Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/bin`and `LD_LIBRARY_PATH` to `/Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/lib`, I reproduce the same segmentation fault error.

unicornss, 2018-06-22:
> After setting my GST_PLUGIN_PATH to /Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/bin, I reproduce the same segmentation fault error.

the plugins will be in the ` /Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/lib` not `/Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/bin` .

The PATH can be set to /Users/alanng/work/cpp/kinesis-video-native-build/downloads/local/bin if  you are planning to use the GStreamer built and linked against  during the install-script. 

If you are planning to use the system GStreamer plugin then you have to update the CMakeLists.txt with the correct path (for the GStreamer and its base plugins) .  

Could you elaborate on how are you planning to use system GStreamer plugin (hope I understand this correctly), source element and kvsplugin .

stylate, 2018-06-22:
Woops, I meant `.../downloads/local/lib`, not bin--my apologies! By system GStreamer, you mean the one that isn't installed by `./install-script`, correct? I think I'll be updating CMakeLists.txt if this segmentation error can't be resolved/debugged. 

As for using the plugin with AWS, I plan on feeding an HLS live stream (the source element) towards KVS, with kvssink as the sink element.

unicornss, 2018-06-22:
Thanks @stylate  for the details.

Yes, you have to build the plugin elements using the same GStreamer libraries to be used from the gst-launch-1.0.

Would you be able to share the pipeline and plugin elements you are using for us to debug further?  You can also send them in email to kinesis-video-support@amazon.com if preferred.

stylate, 2018-06-22:
Thank you! I was wondering how sharing the pipeline and plugin elements I would be using would help debug the segmentation fault issue though, since the issue occurs after running `./install_script` and correctly setting the `GST_PLUGIN_PATH`. For reference, my intended pipeline and plugin elements would be the following:

`gst-launch-1.0 -v souphttpsrc location="sample m3u8 link here" ! hlsdemux ! tsdemux ! h264parse ! avdec_h264 ! autovideosink ! kvssink stream-name="livestream" storage-size=512 access-key="secret" secret-key="secret" `.

unicornss, 2018-06-25:
Hi @stylate  

**Approach 1.** 
Here is the **sample**   [CMakeLists.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2133528/CMakeLists.txt) you could try to compile the kvssink plugin with your GStreamer installation (other than the one installed as part of KVS sdk install-script).

1. Change current working directory to /Users/alanng/work/cpp/kinesis-video-gstreamer-plugin/
` cd /Users/alanng/work/cpp/kinesis-video-gstreamer-plugin/`
2. Replace the CMakeLists.txt that is in the /Users/alanng/work/cpp/kinesis-video-gstreamer-plugin/ with this one. 
2. Run 
  `PKG_CONFIG_PATH=/Users/alanng/work/cpp/downloads/local/lib/pkgconfig cmake .  `
4. Run
 `make`

you can copy the resulting `libgstkvssink.so`  into your plugin directory (where all other plugins are installed in your GStreamer).

Hope this provides guidance on how to compile the kvssink plugin with custom GStreamer installation.

**Approach 2**
 
This **sample** `install-script-with-httpsoupsrc-plugin.sh` will also install the required plugin **souphttpsrc**

It requires two additional dependencies (python3 and intltool)  that you could install 

1. `
  brew install python
`
2.  `
  export PATH=/usr/local/Cellar/python/3.6.5_1/bin:$PATH. 
`
3. `
brew install intltool
`
4. `
 export PATH=/usr/local/Cellar/intltool/0.51.0/bin:$PATH
 ` 

[install-script-with-httpsoupsrc-plugin.sh.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2133568/install-script-with-httpsoupsrc-plugin.sh.zip)

You can verify after installation using `gst-inspect-1.0` 
`
/Users/alanng/work/cpp/downloads/local/bin/gst-inspect-1.0  souphttpsrc 
`

_Another note:_
There is no need for  autovideosink in the GStreamer pipeline   - video streams can be  directly sinked  it to the kvssink .

stylate, 2018-06-25:
Hi @unicornss, I tried Approach 1 and altered CMakeLists.txt to require `log4cplus` to be version 2.0.1 instead of 1.2.0. After being able to build the appropriate files and run `make`, I get the following errors:

[image]

Would it be necessary to have log4cplus be version 1.2.0?

unicornss, 2018-06-25:
@stylate
If you are able to get the libgstkvssink.so built  (using the the newer CMakeLists.txt ) and copied to the system GStreamer plugin location - could you paste the output of the command
`GST_DEBUG=4 gst-inspect-1.0 kvssink` ?

For more details on the plugin path please refer GStreamer documentation [here](https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gstreamer/html/gst-running.html)

stylate, 2018-06-25:
@unicornss 

This works now, thank you so much! I'm now having an issue with my pipeline, though: I get the following errors (while removing autovideosink):

[image]

Would this be related to the .m3u8 file that I'm using? Right now, I'm currently using a sample YouTube livestream and the big bunny VoD stream.

chehefen, 2018-06-25:
can you try replacing avdec_h264 with "video/x-h264,alignment=au,stream-format=avc"? I think decoding the stream loses the cpd.

unicornss, 2018-06-28:
Closing this issue as @stylate  is able to get kvssink compiled against the system GStreamer. For any new issues or questions pl. open a separate thread. thanks.

closed 2018-06-28 by unicornss

akommireddi, 2018-10-17:
@unicornss please help as I am unable to compile gstreamer with the install-script you have provided i get the following error. Please help.

PS: i am having issues trying to install **libsoup**

ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     xmlcatalog
  CCLD     xmllint
./.libs/libxml2.so: undefined reference to `libiconv'
./.libs/libxml2.so: undefined reference to `libiconv_close'
./.libs/libxml2.so: undefined reference to `libiconv_open'
collect2: error: ld returned 1 exit status
./.libs/libxml2.so: undefined reference to `libiconv'
./.libs/libxml2.so: undefined reference to `libiconv_close'
./.libs/libxml2.so: undefined reference to `libiconv_open'
collect2: error: ld returned 1 exit status
Makefile:1115: recipe for target 'xmlcatalog' failed
make[2]: *** [xmlcatalog] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:1119: recipe for target 'xmllint' failed
make[2]: *** [xmllint] Error 1
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/libxml2-2.9.8'
Makefile:1442: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/libxml2-2.9.8'
Makefile:861: recipe for target 'all' failed
make: *** [all] Error 2

akommireddi, 2018-10-18:
I was able to build the producer for hls, thanks to your script. However i get this error.

```
Pipeline is PREROLLING ...
DEBUG - streamReadyHandler invoked
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Secure connection setup failed.
Additional debug info:
gstsouphttpsrc.c(1279): gst_soup_http_src_parse_status (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
TLS/SSL support not available; install glib-networking (6), URL: https://XXXXXXXXXXXXXXXXXX
ERROR: pipeline doesn't want to preroll.
2018-10-18 12:18:22 [140458943223552] Setting pipeline to NULL ...
Freeing pipeline ...
```

mstallmo, 2018-11-29:
I'm working through a similar issue and I'm trying to find where in the CMakeLists.txt file I should update to point at the system GStreamer rather than the downloaded GStramer. Any help would be awesome! 

Thanks!