awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #93: Need more clarity on the callbacks the SDK emits

ameyat2812, 2018-06-25:
Hi @MushMal 

Can you please shade some light on following queries regarding callbacks the SDK emits. Let me also write my understanding about them

1. StreamErrorReportFunc
 - Is this an absolute fatal error report notification? Maybe for errors which SDK can't recover from. The application has to take appropriate actions like resetting stream or in my case calling stop - free - and start on the stream where data present in buffer is flushed.

2. StreamClosedFunc 
- My understanding is that this is a notification to the application that the stream is closed and the stream handle is free to be destroyed. I reckon it is only called when stop is called on a stream so that the stream resources can be freed. Is there any other case when this is triggered and if yes what is expected from application?

3. StreamConnectionStaleFunc/ StreamLatencyPressureFunc/ DroppedFrameReportFunc
 - Is the application supposed to take any actions on these events or the SDK handles all state changes internally? And does the SDK trigger streamClosed or any other CB if these callbacks are triggered consecutively.

Regards,
Ameya

blechatellier, 2018-06-27:
@MushMal @turb0kat anyone in your team that can help with this question please?

zhiyua-git, 2018-06-27:
Hi @ameyat2812 ,

Some related sample code for callbacks is here:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/KvsSinkStreamCallbackProvider.cpp
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/ConnectionStaleStateMachine.cpp
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/KvsSinkStreamCallbackProvider.cpp

Below are some information about these callbacks. 
1. StreamErrorReportFunc
This callback will return [Error Code](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html) from core state machine, both fatal error and recoverable error will show up here.
2. StreamClosedFunc
This is a notification to the application that the stream is closed. This will only triggered when StopStream() is called. This is a signal that all the remaining data in SDK buffer for this stream has already been sent and this stream resource can be freed.
3. StreamConnectionStaleFunc
This is for stream keep sending data out but not getting any ACKs for [some time](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/StreamDefinition.h#L51). When application see this happening, the connection to server is in bad state, something on the long network hop could be having issue. 
Suggestion here is to reset connection, new connection will be established with our backend and data in SDK buffer will be resend from where it started losing ACKs.
4. StreamLatencyPressureFunc
This is for stream getting more and more data in SDK buffer, when the duration of those frames is more than given [max latency](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/StreamDefinition.h#L37). This generally means your network is slow and could not send data out as fast as data is generated, so data started cumulating in buffer.
Suggestion here is to reset connection and see if new connection can be fast and solve this issue. If this issue continues, you might want to check your network status or hard reset the SDK by freeing the KinesisVideoClient and restart (**Note: this could cause your data loss for all data stored in the SDK buffer**).
5. DroppedFrameReportFunc
This is for dropping frame in a stream, generally it could happen when the [SDK buffer](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/StreamDefinition.h#L49) for this stream is full, but new frames keep coming, the oldest frames will be dropped. As oldest frames are dropped, some key frames could be dropped and result in INVALID_MKV_DATA error ACK in this case.
Suggestion here is to try to reset connection or just hard reset the SDK. Ideally application should not get into this case, the network issue would cause StreamConnectionStaleFunc or StreamLatencyPressureFunc before running into this.
6. Callback Frequency
These callbacks except StreamClosedFunc are triggered on each frame. So we suggest keeping the callback silent for some time to see if application's behavior (reset connection, etc.) helps recovered. 

Thanks,
-ZH

unicornss, 2018-06-27:
Additional Ref:
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-callbacks.html

ameyat2812, 2018-06-29:
Hi @zhiyua-git Thanks for the detailed explanation and sample code. I could see that the sample code has a limited lifespan for the stream. Meaning, The stream is created - non fatal errors are just reported - and stream closes on fatal error.

However I have a requirement of keeping the stream alive at all times. Let me quickly explain how I am using the SDK.

I am having multiple streams opened per Producer instance, and the I start feeding video data to any of the steams based on events of interests. So I have written a wrapper class around Kinesis Producer SDK which exposes simple pushVideoFrame API for each of the supported streams. This wrapper class maintains a cond_var for state readiness (something similar to KinesisVideoStream cond_var in SDK) and waits for stream to be ready before it calls putFrame function.

OnStreamErrorCallback, I first try resetConnection() and if that fails, I stop the stream marking my internal state for the stream as not ready. Then on receiving streamClosedCallback. I free the stream, and restart it marking my stream as ready.

I have some follow-up questions for you, I hope the explanation above will help you put these questions is perspective.

1. StreamErrorReportFunc
How can I distinguish between fatal error for which I should stop - free - restart my stream connection and non-fatal errors for which SDK can recover.

2. StreamClosedFunc 
I am seing a behaviour when streamClosed is triggered even without the application calling. It could also be some race condition between Application stopping and the callback thread. But is it safe to assume that Application can restart the stream from this callback no matter when it is triggered as I want the stream to be always running.

3. StreamConnectionStaleFunc/StreamLatencyPressureFunc/DroppedFrameReportFunc
Will add resetConnection logic here. But just not on the first callback. Maybe after errors persist for a while. Does it make sense?

Regards,
Ameya

zhiyua-git, 2018-06-30:
Hi Ameya,

These are answers to your questions:

1. The best option for the application to deal with StreamErrorReportFunc  would be to re-create the stream on any status code which is NOT one of the following:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/d1684599a141785752582c16264e3123866f3cf8/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L145-L164

2. It is OK to restart the stream as streamClosed will only be triggered when stream is stopped.

3.  I would suggest resetConnection on first callback, because these callback will be triggered only after conditions are met. For example, ACKs are missing for XX seconds and network layer sends data normally, if data cannot make it to our service through this stale connection, it means you would have XX seconds delay already, it would be great to create a new connection ASAP. Also similar [behavior](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/ConnectionStaleStateMachine.cpp) in our gstreamer plugin is suggested, so you can have multiple times of retry on resetConnection.

Thanks,
-ZH

ameyat2812, 2018-07-04:
Hi @zhiyua-git 

Thanks for the answer. The info that you provided really helped understand the callback structure. However there are some inconsistencies in what you are suggesting and what I have seen happening with the SDK after I implemented the changes.

Let me write some of the observations/queries here

1. As the SDK returns most of the errors async and putting more frames into broken stream generally makes things worse, I have a cond_var (if_stream_ready) logic around my streams. I mark the stream as ready on streamReadyCallback and mark it not ready when I get streamClosedCallback.

I tried using the cond_var that KinesisVideoStream offers but I found it never really reflect the true readiness of the stream (You might want to check if it is being properly updated in all cases.)

2. As per your advise, I have added logic to hard restart the stream on streamClosed as I expect this callback to happen only when stream is absolutely bonkers. But I have seen this getting triggered almost immediately after I make resetConnection() for any frame error such as droppedFrameCallback or connectionStaleness etc. This raises a question that is it even an absolute error state in which stop - free and restart is the only option. 

Point to note here is , we use kinesis video streams to upload live events but in bursts.  So the stream is created and could be idle for a while before it has data flowing for some time. I have observed, whenever video data is fed to an idle stream, streamConnectionStaleness callback is triggered. For which I resetConnection() and this triggers streamClosed / streamReady is quick succession.

3.  I use KinesisVideoProducer::createStreamSync function to get shared_ptr for the KinesisVideoStream object. I end up getting streamReadyCallback even before the above function returns. So if I have make any updates to my wrapper class in streamReady callback, I have to wait for createStreamSync to return.

most of the above issue are basically because of following in my opinion.
1. Lack of synchronous  errors for APIs like putFrame so that the application can maybe drop some frames and let the SDK recover.

2. stream_ready_ not reflecting the true readiness of the stream so that when the streamReady is true is almost guaranteed that the frame will be sent to kinesis.

3. Maybe a callback to inform when the stream is absolutely busted. So that it can be stopped - freed - and restarted and no callbacks are triggered on that dead stream.

4. An API to do a hard reset on the stream so that you don't really have to destroy the stream object. Again no callback triggered when this is happening

Please let me know if it makes any sense???

Regards,
Ameya

MushMal, 2018-07-05:
@ameyat2812 thanks for the details. I will try to address them one-by-one. First of all, the entire core of the SDK is asynchronous and doesn't have it's own threading (PIC). The C++ is simply a wrapper layer that makes the consumption of the PIC easier and adds the real networking capabilities. As such, it has to wrap around the callbacks and satisfy them. Some of the callbacks then get exposed to the client application that sits on top of C++. Most of the callbacks can not possibly be sync during the "play" time as they will block on networking which is a bad thing to do. Now, with that in mind, let me go over the bullet points.

1) All of the PIC callbacks are exposed through the C++ SDK layer. When the application calls stopStream() API (which is async API) it tells the PIC to send out the existing buffer however it can and notify when it finished by calling streamClosed callback. The issue you might be seeing is that it times out due to networking. You should ensure that you have a timeout of some sort - similar to what we do with createStreamSync API. We are going to add closeStreamSync API which will either successfully finish transmitting the remainder of the buffer or throw on timeout.
2) Not entirely sure what you meant. Does the teardown of the stream fail? Generally speaking, you can always free the stream but you might lose the data in the buffer. This is the reason why we have stopStream API which can be called first to try to send out the remainder of the bits (or timeout). This is done as the last ditch attempt to send the buffered data. There are errors which will mean that the remainder of the buffer will not be possible to be sent to the backend and in this case hard reset (free) needs to be called.
3) If you use createStreamSync then you don't need to listen to streamReady API. Any sync API provided in the C++ layer makes it easier to not code your own timeout routines, etc... However, if you have more advanced Application which handles things in the background while the stream is being created, you could use the async API (createStream) and trigger readiness when streamReady callback is called. We are going to add similar stopStreamSync API in the near future.

One thing that I would recommend is if you could look at the GStreamer plugin code - there are similar concepts as you described.

turb0kat, 2018-07-06:
There is a conceptual mismatch here.  The SDK is designed for a continuous live data stream in.  With a live video producer, you would never 'stop sending frames and wait for the "SDK" (most likely, the network) to recover'.  

In your case it sounds like you want to do this because you have the video stored locally and you want to make sure it all gets to cloud without dropping frames.   

Idea #1: Direct invocation of PutMedia API might be worth considering.  This will have its own challenges.

Idea #2: A synchronous wrapper around the SDK with putFrameSync and potentially even putFragmentSync APIs, which returns a hard error if the connection is down / buffer is full / etc, might be a good addition to the SDK for your use case.  This effectively converts our streaming solution back into a request/response solution from the client perspective while retaining the high-throughput streaming and buffering of the SDK.  Such a wrapper could manage connectivity for you while making sure not to lose frames unless the process crashed.  [we will consider this as an SDK feature request, also you could implement it and submit a PR]

MushMal, 2018-07-09:
Entirely forgot to mention... For the intermittent producers, it's more prudent not to pre-create a stream and wait for an event to send the data. Perhaps, a better approach would be on each event to create the stream, stream the necessary data, close and free the stream.

ameyat2812, 2018-07-10:
Thanks guys for your comments. I they clear out most of the doubts

unicornss, 2018-07-10:
Closing this for now as the question been answered. Please reopen if required.

closed 2018-07-10 by unicornss

sairoop-elear, 2020-10-13:
> Entirely forgot to mention... For the intermittent producers, it's more prudent not to pre-create a stream and wait for an event to send the data. Perhaps, a better approach would be on each event to create the stream, stream the necessary data, close and free the stream.

My application uses Kinesis C producer SDK, there are cases where I sometimes get errors while writing frames using putKinesisVideoFrame(). But we placed a cool off period of 30 seconds interval between video recordings, can I close and free the stream after I finish my recording?

MushMal, 2020-10-13:
@sairoop-elear the backend (and the SDK for that matter) has a 30 second inactivity timeout. So, if your application doesn't produce any frames within that time, the SDK will close the connection and loop back to Ready state. Next time you have a frame produced, it will issue a PutMedia call and continue. The only drawback here is that because the previous fragment wasn't properly persisted (as it's not properly closed) it will re-stream from the previous fragment and might cause a latency pressure.

Recommendation here is to issue/putFrame a special EoFR frame when your session ends so the SDK will cause the backend to finalize the current fragment so next time it won't rollback to re-stream the "unfinished" fragment.

This said, you could use an explicit intermittent producer scenario by stopSync and then resetStream API calls when next time you need to continue. You could also stopSync/free and re-create. In both cases, the issue is that a few backend calls will be made (like Describe. GetEndpoint).