awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #94: File based buffer in case of no connectivity

mishitpatel, 2018-06-27:
Hi all,
Do we have file based persistence buffering support for this SDK yet in case out internet connection outage ?  (Trying to use `kinesis-video-gst-demo`)

unicornss, 2018-06-28:
Currently its not supported. We will update with samples when its available, please stay tuned. Thanks

unicornss, 2018-06-28:
Hi @mishitpatel Could you help us understand your use case better?  

- Are you referring file based storage just to hold the video streams only during outage?
--  What is the maximum duration of the typical outage you expect to accommodate? 
 - or do you plan to send video files to KVS?

Thanks

closed 2018-08-08 by turb0kat

mishitpatel, 2018-10-21:
@unicornss 
> Are you referring file based storage just to hold the video streams only during outage?
Correct. We are planning to use it for security surveillance system. In case of internet outage SDK should be able to persist video frames on to the disk (few days of data) and when internet comes back online it should be able sync all the data back to cloud.

MushMal, 2018-10-24:
@mishitpatel thanks for the information. While this scenario is not currently supported by the SDK, we are working toward a solution that can handle it in the SDK itself by allowing the SDK to use the file-backed storage for large buffers. There are multiple issues that we will need to account for - for example, once the network is back, would it be possible to stream faster-than-real-time to catch up with the current head. The alternative is to create a separate stream that can stream up the archived data while the real-time is streaming the current stream. All these add to the complexity of the SDK as there are different applications which different desired behaviors. We would like to hear from you about your requirements to ensure we account for it.

wlinInspire, 2020-04-08:
@MushMal We have a similar use case that needs KVS to buffer large data using file storage when the network is poor. Just want to check in to see when this feature will be available. Thanks!

MushMal, 2020-04-08:
@wlinInspire what is your expectation in this case. Say you have a large content store that can store hours of frames. Once the network is back, do you expect it to pick up and stream from the tail and stream faster than realtime to catch up with the head or you are looking for a more involved scenario of a backfill where the realtime (the head) can continue streaming but there would be a separate stream streaming into the same backend stream to backfill the older frames?

We are collecting the data to understand the right path forward

wlinInspire, 2020-04-08:
@MushMal I think the former scenario is relatively simpler and can be done first, since the latter one can reuse the former one's code if possible. I think the latter scenario is ideal personally. Let me know.

MushMal, 2020-04-09:
Indeed, the implementation of a file-based hybrid content store is the simple solution. Let us think on how we can prioritize this work.