awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #95: The sample app no works

jdavidguerrero, 2018-07-06:
Hi, I'm trying to run the example app, but I do not want to play the streaming on the console, export the access-key-id and the secret-access-key, but this is the result, I appreciate your help

[INFO ][2018-07-06 04:41:39] Initializing curl.
[INFO ][2018-07-06 04:41:39] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-07-06 04:41:39] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-07-06 04:41:39] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-07-06 04:41:39] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-07-06 04:41:39] Refreshing credentials. Force refreshing: 0 Now time is: 1530852099 Expiration: 0
[INFO ][2018-07-06 04:41:39] New credentials expiration is 1530854499
[DEBUG][2018-07-06 04:41:39] createDeviceHandler invoked
[INFO ][2018-07-06 04:41:39] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-07-06 04:41:39] clientReadyHandler invoked
[DEBUG][2018-07-06 04:41:39] Awaiting for the producer to become ready...
[DEBUG][2018-07-06 04:41:39] Kinesis Video producer is Ready.
[DEBUG][2018-07-06 04:41:39] Client is ready
[INFO ][2018-07-06 04:41:39] Creating Kinesis Video Stream test
[INFO ][2018-07-06 04:41:39] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-07-06 04:41:39] describeStreamHandler invoked
[INFO ][2018-07-06 04:41:39] Refreshing credentials. Force refreshing: 0 Now time is: 1530852099 Expiration: 0
[DEBUG][2018-07-06 04:41:39] Awaiting for the stream to become ready...
[ERROR][2018-07-06 04:41:44] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5515 milliseconds
[WARN ][2018-07-06 04:41:44] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJSP6ZNIDKJGZF3MQ/20180706/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=3cacc5ff783359a473d54f490bd433bcca7706e9dfd26a14d19145cdd6ff0211
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180706T044139Z
[DEBUG][2018-07-06 04:41:44] describeStream response: 
[INFO ][2018-07-06 04:41:44] Describe stream did not find the stream test in Kinesis Video (stream will be created)
[INFO ][2018-07-06 04:41:44] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-07-06 04:41:44] describeStreamHandler invoked
[INFO ][2018-07-06 04:41:44] Refreshing credentials. Force refreshing: 0 Now time is: 1530852104 Expiration: 0
[ERROR][2018-07-06 04:41:50] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5515 milliseconds
[WARN ][2018-07-06 04:41:50] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJSP6ZNIDKJGZF3MQ/20180706/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=6ea22e81cff0ea1d6c33a305087b1ffe678634e21ee5c121c8e5b6d6cccc5e56
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180706T044144Z
[DEBUG][2018-07-06 04:41:50] describeStream response: 
[INFO ][2018-07-06 04:41:50] Describe stream did not find the stream test in Kinesis Video (stream will be created)
[INFO ][2018-07-06 04:41:50] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-07-06 04:41:50] describeStreamHandler invoked
[INFO ][2018-07-06 04:41:50] Refreshing credentials. Force refreshing: 0 Now time is: 1530852110 Expiration: 0
[ERROR][2018-07-06 04:41:56] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5514 milliseconds
[WARN ][2018-07-06 04:41:56] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJSP6ZNIDKJGZF3MQ/20180706/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=e4dad4ebaf5e71eb454316acb33901c14a2078257376c2a9ac9e3f7711294e1b
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180706T044150Z
[DEBUG][2018-07-06 04:41:56] describeStream response: 
[INFO ][2018-07-06 04:41:56] Describe stream did not find the stream test in Kinesis Video (stream will be created)
[INFO ][2018-07-06 04:41:56] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-07-06 04:41:56] describeStreamHandler invoked
[INFO ][2018-07-06 04:41:57] Refreshing credentials. Force refreshing: 0 Now time is: 1530852117 Expiration: 0
[ERROR][2018-07-06 04:42:02] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5515 milliseconds
[WARN ][2018-07-06 04:42:02] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJSP6ZNIDKJGZF3MQ/20180706/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=07274c3d2b8b0e69f3943e74f276d75d7d269a636ae1d71d0ec07270bb78d12b
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180706T044157Z
[DEBUG][2018-07-06 04:42:02] describeStream response: 
[INFO ][2018-07-06 04:42:02] Describe stream did not find the stream test in Kinesis Video (stream will be created)
[INFO ][2018-07-06 04:42:02] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-07-06 04:42:02] describeStreamHandler invoked
[INFO ][2018-07-06 04:42:04] Refreshing credentials. Force refreshing: 0 Now time is: 1530852124 Expiration: 0
[ERROR][2018-07-06 04:42:09] Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.
Abortado

unicornss, 2018-07-06:
Please check network connectivity is working and certificates are installed as per the documentation ( https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/79 ) .

> [ERROR][2018-07-06 04:42:02] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result **Timeout was reached:** Resolving timed out after 5515 milliseconds

Could you pl tell us which environment you have been running from -  Ubuntu, Raspberry PI, MacOS ?

Ga22be, 2018-07-06:
Also note proxy can affect network connectivity for the application.

jdavidguerrero, 2018-07-06:
I test in raspberry pi 3 and elementary OS loki (ubuntu 16.04)

unicornss, 2018-07-06:
@jdavidguerrero  Thanks for the details.  Please check the connectivity exists for all the network interfaces - more details can be found here :
( https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/62  )

jdavidguerrero, 2018-07-06:
already check the connections just leave the wlan and as for the certificates try with the command
curl https://kinesisvideo.us-west-2.amazonaws.com --verbose and the result is:
* Rebuilt URL to: https://kinesisvideo.us-west-2.amazonaws.com/
*   Trying 35.161.88.185...
* TCP_NODELAY set
* Connected to kinesisvideo.us-west-2.amazonaws.com (35.161.88.185) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/cert.pem
  CApath: /etc/ssl/certs
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: CN=kinesisvideo.us-west-2.amazonaws.com
*  start date: Nov 21 00:00:00 2017 GMT
*  expire date: Nov 21 12:00:00 2018 GMT
*  subjectAltName: host "kinesisvideo.us-west-2.amazonaws.com" matched cert's "kinesisvideo.us-west-2.amazonaws.com"
*  issuer: C=US; O=Amazon; OU=Server CA 1B; CN=Amazon
*  SSL certificate verify ok.
> GET / HTTP/1.1
> Host: kinesisvideo.us-west-2.amazonaws.com
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 403 Forbidden
< x-amzn-RequestId: 6b0ca4c6-8144-11e8-a1b6-951e27640d49
< Content-Length: 127
< Date: Fri, 06 Jul 2018 17:45:48 GMT
< 
<MissingAuthenticationTokenException>
  <Message>Missing Authentication Token</Message>
</MissingAuthenticationTokenException>
* Curl_http_done: called premature == 0
* Connection #0 to host kinesisvideo.us-west-2.amazonaws.com left intact

unicornss, 2018-07-10:
Hi @jdavidguerrero 
Would you please check the curl from this SDK downloads directory?
<your-sdk-download-directory>/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/bin/curl

jdavidguerrero, 2018-07-10:
hi @unicornss look the content of directory:

jdavidguerrero@CEO-GPROG:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/bin(master)$ ls
aclocal       automake       autoupdate   envsubst      gdbus-codegen  gio-querymodules        glib-mkenums            gst-discoverer-1.0  gst-typefind-1.0  libtool     msgcmp   msgfilter  msgunfmt     pcregrep         yacc
aclocal-1.15  automake-1.15  bison        flex          gettext        glib-compile-resources  gobject-query           gst-inspect-1.0     gtester           libtoolize  msgcomm  msgfmt     msguniq      pcretest         yasm
autoconf      autopoint      c_rehash     flex++        gettextize     glib-compile-schemas    gresource               gst-launch-1.0      gtester-report    m4          msgconv  msggrep    ngettext     recode-sr-latin  ytasm
autoheader    autoreconf     **curl**         gapplication  gettext.sh     glib-genmarshal         gsettings               gst-play-1.0        iconv             msgattrib   msgen    msginit    openssl      vsyasm
autom4te      autoscan       **curl-config**  gdbus         gio            glib-gettextize         gst-device-monitor-1.0  gst-stats-1.0       ifnames           msgcat      msgexec  msgmerge   pcre-config  xgettext

unicornss, 2018-07-11:
Hi @jdavidguerrero  Thanks for the folder contents.

I was referring to your earlier message:
> already check the connections just leave the wlan and as for the certificates try with the command
curl https://kinesisvideo.us-west-2.amazonaws.com --verbose and the result is:

In that case the `curl https://kinesisvideo.us-west-2.amazonaws.com` could have been using the system installed `curl` rather than the one in the `~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/bin/curl` . Wanted to check if there is any specific certificate location unique to elementary OS loki (ubuntu 16.04) causing this. I 

Just to be sure can you also please verify the network connectivity exists in the wired network as well ? If there is no connectivity to AWS through wired network then that could result in Time out if the wired network interface gets to send traffic.

turb0kat, 2018-08-08:
closing due to inactivity, feel free to re-open.

closed 2018-08-08 by turb0kat