awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #99: support existing aws credential methods

ranman, 2018-07-11:
The https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/kinesis-video-gstreamer-plugin gstreamer plugin should support reading credentials from ~/.aws/credentials with profile support and similar - like the AWS C++ sdk.

It should also take a region as a parameter instead of having to set the AWS_DEFAULT_REGION environment variable.

It looks like this might be possible if I pass in aws_region into the sink? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/gstkvssink.cpp#L302

That isn't documented anywhere other than the code though.

unicornss, 2018-07-11:
Thanks @ranman 
We will update the  [documentation](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin-parameters.html) with the `aws-region` parameter as well
`
aws-region          : AWS Region
                              flags: readable, writable
                              String. Default: "us-west-2"
`

Deshke, 2018-09-04:
why are the ec2role/iam credentials missing? Could we also get them added ?

MushMal, 2018-09-04:
@Deshke could you please elaborate what you mean by the ec2role/iam credentials missing? We recently discovered that our documentation was missing the credential integration information and we are in the process of updating the documentation.

MushMal, 2018-09-04:
@ranman this is a good suggestion to use the credentials from the standard aws file if the credentials have not been specified otherwise. Currently, there is a way to point to a file with credentials but the default AWS credentials file won't be loaded by default.

Deshke, 2018-09-04:
@MushMal sry, ec2roles are not supported anyway (why is that?)

timurb, 2019-09-05:
any resolution on this?

MushMal, 2019-09-06:
@timurb we are working on it. No ETA yet. We will update the README as soon as we implement it.

MushMal, 2020-05-04:
We are still looking into this. Not ETA yet. Just wanted to update

TheSilentHorizon, 2020-06-04:
Sorry, for pushing this. 

If i need to enter my session_token as a AWS Educate Account, how can I do it with kvssink when .aws\credentials isnt supported?

MushMal, 2020-06-04:
Can be done via env var: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L249

hassanctech, 2020-08-25:
@TheSilentHorizon Does what was suggested by @MushMal above suffice for you, or is there something more that you need?

Deshke, 2020-08-25:
@hassanctech @MushMal ec2 instance/iam roles are still not supported.

riotgibbon, 2020-10-14:
> Can be done via env var: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L249

I wish I had seen this 3 days ago, I've been bashing my head trying to get kvssink to work with Cognito Identity Pools in Python

In case anybody else is going through this:

```
   print (f"Getting an Identity for Identity Pool {IdentityPoolId}")
    response = client.get_id(
        AccountId=AccountId,
        IdentityPoolId=IdentityPoolId
        )

    IdentityId=response["IdentityId"]
    print (f"Retrieved anonymous Identity {IdentityId}")

    open_id_response = client.get_open_id_token(
        IdentityId=IdentityId
    )

    token= open_id_response["Token"]
    print(f"token: {token}")
    sts_client = boto3.client('sts')
    basicCreds=sts_client.assume_role_with_web_identity(
        RoleArn=unAuthRoleArn,
        RoleSessionName='molvuUnauth',
        WebIdentityToken=token
    )
    print (basicCreds)
    ACCESS_KEY=basicCreds["Credentials"]['AccessKeyId']
    SECRET_KEY=basicCreds["Credentials"]['SecretAccessKey']
    SESSION_TOKEN=basicCreds["Credentials"]['SessionToken']

    os.environ['AWS_ACCESS_KEY_ID'] = ACCESS_KEY 
    os.environ['AWS_SECRET_ACCESS_KEY'] = SECRET_KEY 
    os.environ['AWS_SESSION_TOKEN'] = SESSION_TOKEN 
```

then you call Gst.parse_launch , but without any key parameters - the README really needs updating to cover this. How amenable are the maintainers to users doing that?

jbaker-dstl, 2021-12-07:
Has there been any progress on this?

I'm looking to stream video into KVS using an ECS task. It works for a while, but I then get an authentication error (presumably because my credentials time out) and the only way I've been able to address it so far is to restart GStreamer with new credentials each time it errors. This isn't ideal, and results in a gap in the stream as it restarts (and sometimes, it seems to just hang after bad credentials rather than exiting).

I'm getting the credentials and running GStreamer/kvssink as follows.

```
python3 creds.py > /opt/credentials.txt
```

```
from urllib.request import urlopen
import json
import os

metadata_url = 'http://169.254.170.2' + os.getenv('AWS_CONTAINER_CREDENTIALS_RELATIVE_URI')
credentials = json.load(urlopen(metadata_url))

print('CREDENTIALS', credentials['AccessKeyId'], credentials['Expiration'], credentials['SecretAccessKey'], credentials['Token'])
```

```
gst-launch-1.0 souphttpsrc do-timestamp=true location=$MY_URL ! hlsdemux ! tsdemux ! queue ! video/x-h264 ! h264parse ! kvssink stream-name=my-stream credential-path=/opt/credentials.txt
```

It would be great if there was an easy way of using kvssink in this situation, where it would automatically get the latest credentials from the container, rather than having to run the above code in a loop!

If this is already possible somehow, please do let me know.