awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #100: ./kinesis_video_gstreamer_sample_app: No such file or directory

pravin82, 2018-07-16:
I am trying to run gstreamer sample app from Mac,but I am getting this error "./kinesis_video_gstreamer_sample_app: No such file or directory".I have executed the ./install_script inside the" kinesis-video-native-build " directory

unicornss, 2018-07-16:
Can you please send the contents of  kinesis-video-native-build directory ?

`$ ls kinesis-video-native-build `

pravin82, 2018-07-16:
here are the contents of kinesis-video-native-build

CMakeLists.txt			               java-install-script
certs				                       kvs_log_configuration
downloads			                       min-install-script
gstreamer-plugin-install-script             samples
install-script

unicornss, 2018-07-16:
Hi @pravin82  If the ./install-script is complete successfully then you will see all the demo samples are available in this folder.  Given there is no demo sample binaries in this folder suggests that either the install-script was not complete or did not run. 

> I have executed the ./install_script inside the" kinesis-video-native-build " directory

Can you provide details on the output of the ./install-script ?

pravin82, 2018-07-16:
Hi @unicornss here is details of the output of the ./install-script.

"./install-script: line 368: cd: /Users/pravinmishra/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/flex-2.6.3: No such file or directory"

I have flex 2.6.4 installed on my system.

unicornss, 2018-07-16:
Seems install-script was not completed fully before,  in order to debug your environment we need more details. 
-  Which version of Mac OS are your running?
- How are you planning to send video to Kinesis Video Streams
-  Can you elaborate on how did you download/install 2.6.4 ?  brew install? Its not clear why flex 2.6.4 installed as the install-script is looking for 2.6.3 and downloads as part of the install anyways.
- if you have downloaded and in the kinesis-video-native-build/downloads directory then the install-script still does not pick up that version as its written to download 2.6.3

```
if [ ! -f $DOWNLOADS/local/lib/libfl.2.dylib ]; then
    if [ ! -f $DOWNLOADS/local/lib/libfl.2.so ]; then
        if [ ! -f $DOWNLOADS/local/lib/libfl.so ]; then
          echo "flex not found. Installing"
          if [ ! -f $DOWNLOADS/flex-2.6.3.tar.gz ]; then
            cd $DOWNLOADS
            $SYS_CURL -L $addl_curl_args "https://github.com/westes/flex/releases/download/v2.6.3/flex-2.6.3.tar.gz" -o "flex-2.6.3.tar.xz"
          fi
          cd $DOWNLOADS
          tar -xvf flex-2.6.3.tar.xz
          cd $DOWNLOADS/flex-2.6.3
          ./configure --prefix=$DOWNLOADS/local  CFLAGS="-I$DOWNLOADS/local/include" LDFLAGS="-L$DOWNLOADS/local/lib"
          make -j $max_parallel
          make install
      fi
  fi
fi
```

Seems https://github.com/westes/flex/releases/download/v2.6.3/flex-2.6.3.tar.gz downloadable. 
Could you please add additional output you are seeing just before 
 
> "./install-script: line 368: cd: /Users/pravinmishra/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/flex-2.6.3: No such file or directory"

pravin82, 2018-07-17:
I am running on MacOS version 10.13.3 and trying to send video through webcam.
Installed flex through brew install.
I deleted "download" folder from "kinesis-video-native-build" directory and rerun the install-script.Now I am getting this as output

"checking for g++ option to produce PIC... -fno-common -DPIC
checking if g++ PIC flag -fno-common -DPIC works... yes
checking if g++ static flag -static works... no
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... darwin17.4.0 dyld
checking how to hardcode library paths into programs... immediate
configure: working c++ compiler found: yes
checking for perl... /usr/bin/perl
checking for bison... /usr/bin/bison
checking bison version 2.3 >= 2.4... configure: error: no"
I have bison 2.3 installed.

unicornss, 2018-07-17:
The brew installs  in /usr/local directory E.g. /usr/local/Cellar/bison/3.0.4_1/bin 

You need to `export PATH=/usr/local/Cellar/bison/3.0.4_1/bin:$PATH` so that brew install precedes the older version in `/usr/bin/bison`.  You can do similar for  autoconf and pkg-config also as noted in the install-script (E.g. `export PATH=/usr/local/Cellar/autoconf/2.69/bin:/usr/local/Cellar/pkg-config/0.29.2/:$PATH` ).

 Hope this helps. You can also the Mac High Sierra related posts in this section.

pravin82, 2018-07-17:
Hi @unicornss  thanks a lot for your help, it worked.But now I am getting a new
 error when I execute "./kinesis_video_gstreamer_sample_app" script.
The error log is "Failed to create Kinesis Video Stream - timed out.
libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Failed to create Kinesis Video Stream - timed out.
Abort trap: 6"

Above that there is log saying "curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Couldn't connect to server: Failed to connect to kinesisvideo.us-west-2.amazonaws.com port 443: Connection refused".

pravin82, 2018-07-18:
Thanks @unicornss  for the help.It was a proxy problem and now it's working fine.

closed 2018-07-18 by pravin82