awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #101: Install script results in make errors on macOS High Sierra

msjoshi, 2018-07-19:
I'm using this library for the first time and trying to run the install script. As advised, I pre-installed autoconf and bison before running the install-script. I get the below error when I run the install script. I have Java 8 installed and do not understand why it is looking for JDK 1.3. 

> Making all in intl-java
/bin/sh ../javacomp.sh -d . ./gnu/gettext/GettextResource.java
cd . && test ! -d /usr/lib/jdk1.3.1 || env PATH=/usr/lib/jdk1.3.1/bin:$PATH javadoc -d javadoc2 gnu.gettext gnu/gettext/*.java
warning: [options] bootstrap class path not set in conjunction with -source 1.3
error: Source option 1.3 is no longer supported. Use 1.6 or later.
error: Target option 1.1 is no longer supported. Use 1.6 or later.
make[4]: *** [gnu/gettext/GettextResource.class] Error 2
make[3]: *** [all-recursive] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

I tried running the min-install-script and it runs fine. Any thoughts where things could be wrong?

unicornss, 2018-07-25:
@msjoshi  We can't recreate this in Mac High Sierra locally ... Could you please try removing the downloads directory in the SDK  ( within the ./kinesis-video-native-build )  directory and run install-script again.

msjoshi, 2018-08-02:
I deleted the downloads directory and tried running the install script. Same error below. 

> libtool: compile:  gcc -c -DLOCALEDIR=\"/Users/mjoshi/projects/scratch/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/share/locale\" -DLOCALE_ALIAS_PATH=\"/Users/mjoshi/projects/scratch/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/share/locale\" -DLIBDIR=\"/Users/mjoshi/projects/scratch/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib\" -DBUILDING_LIBINTL -DBUILDING_DLL -DIN_LIBINTL -DENABLE_RELOCATABLE=1 -DIN_LIBRARY -DINSTALLDIR=\"/Users/mjoshi/projects/scratch/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib\" -DNO_XMALLOC -Dset_relocation_prefix=libintl_set_relocation_prefix -Drelocate=libintl_relocate -DDEPENDS_ON_LIBICONV=1 -DHAVE_CONFIG_H -I. -I. -I.. -g -O2 -fvisibility=hidden ./intl-compat.c -o intl-compat.o >/dev/null 2>&1
/bin/sh ../libtool  --tag=CC --mode=link \
	  gcc  -g -O2 -fvisibility=hidden   -o libintl.la \
	  bindtextdom.lo dcgettext.lo dgettext.lo gettext.lo finddomain.lo hash-string.lo loadmsgcat.lo localealias.lo textdomain.lo l10nflist.lo explodename.lo dcigettext.lo dcngettext.lo dngettext.lo ngettext.lo plural.lo plural-exp.lo localcharset.lo threadlib.lo lock.lo relocatable.lo langprefs.lo localename.lo log.lo printf.lo setlocale.lo version.lo xsize.lo osdep.lo intl-compat.lo -liconv -Wl,-framework -Wl,CoreFoundation   -lc \
	   \
	  -version-info 10:4:1 \
	  -rpath /Users/mjoshi/projects/scratch/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib \
	  -no-undefined
libtool: link: gcc -dynamiclib  -o .libs/libintl.9.dylib  .libs/bindtextdom.o .libs/dcgettext.o .libs/dgettext.o .libs/gettext.o .libs/finddomain.o .libs/hash-string.o .libs/loadmsgcat.o .libs/localealias.o .libs/textdomain.o .libs/l10nflist.o .libs/explodename.o .libs/dcigettext.o .libs/dcngettext.o .libs/dngettext.o .libs/ngettext.o .libs/plural.o .libs/plural-exp.o .libs/localcharset.o .libs/threadlib.o .libs/lock.o .libs/relocatable.o .libs/langprefs.o .libs/localename.o .libs/log.o .libs/printf.o .libs/setlocale.o .libs/version.o .libs/xsize.o .libs/osdep.o .libs/intl-compat.o   -liconv -lc  -g -O2 -Wl,-framework -Wl,CoreFoundation   -install_name  /Users/mjoshi/projects/scratch/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libintl.9.dylib -compatibility_version 11 -current_version 11.4 -Wl,-single_module
libtool: link: (cd ".libs" && rm -f "libintl.dylib" && ln -s "libintl.9.dylib" "libintl.dylib")
libtool: link: ar cru .libs/libintl.a  bindtextdom.o dcgettext.o dgettext.o gettext.o finddomain.o hash-string.o loadmsgcat.o localealias.o textdomain.o l10nflist.o explodename.o dcigettext.o dcngettext.o dngettext.o ngettext.o plural.o plural-exp.o localcharset.o threadlib.o lock.o relocatable.o langprefs.o localename.o log.o printf.o setlocale.o version.o xsize.o osdep.o intl-compat.o
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(threadlib.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(printf.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(xsize.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(osdep.o) has no symbols
libtool: link: ranlib .libs/libintl.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(threadlib.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(printf.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(xsize.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libintl.a(osdep.o) has no symbols
libtool: link: ( cd ".libs" && rm -f "libintl.la" && ln -s "../libintl.la" "libintl.la" )
Making all in intl-java
/bin/sh ../javacomp.sh -d . ./gnu/gettext/GettextResource.java
cd . && test ! -d /usr/lib/jdk1.3.1 || env PATH=/usr/lib/jdk1.3.1/bin:$PATH javadoc -d javadoc2 gnu.gettext gnu/gettext/*.java
warning: [options] bootstrap class path not set in conjunction with -source 1.3
error: Source option 1.3 is no longer supported. Use 1.6 or later.
error: Target option 1.1 is no longer supported. Use 1.6 or later.
make[4]: *** [gnu/gettext/GettextResource.class] Error 2
make[3]: *** [all-recursive] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

zhiyua-git, 2018-08-02:
Hi @msjoshi,

Can you run “/usr/libexec/java_home -V” and “java -version” and paste the result? 

From the log you are likely having java1.3 also installed on your Mac and the build is choosing it.

Thanks,
-ZH

msjoshi, 2018-08-02:
@zhiyua-git see below

> mjoshi@Mayurs-MacBook-Pro kinesis-video-native-build (master)*$ /usr/libexec/java_home -V
Matching Java Virtual Machines (2):
    9.0.1, x86_64:	"Java SE 9.0.1"	/Library/Java/JavaVirtualMachines/jdk-9.0.1.jdk/Contents/Home
    1.8.0_162, x86_64:	"Java SE 8"	/Library/Java/JavaVirtualMachines/jdk1.8.0_162.jdk/Contents/Home

> /Library/Java/JavaVirtualMachines/jdk-9.0.1.jdk/Contents/Home
mjoshi@Mayurs-MacBook-Pro kinesis-video-native-build (master)*$ java -version
java version "1.8.0_162"
Java(TM) SE Runtime Environment (build 1.8.0_162-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.162-b12, mixed mode)

zhiyua-git, 2018-08-02:
Hi @msjoshi,

If you are not using java 1.3 at all, can you try to remove /usr/lib/jdk1.3.1 and run the install-script again? Hope that can solve your issue.

Meanwhile, can you send us the full log generated by running install-script? You can send it to kinesis-video-support@amazon.com. 

Thanks,
-ZH

closed 2018-08-08 by turb0kat

TodorBalabanov, 2019-04-29:
I have the same error when I am trying to build LibreOffice on Mac OS:

`Making all in intl-java

/bin/sh ../javacomp.sh -d . ./gnu/gettext/GettextResource.java

warning: [options] bootstrap class path not set in conjunction with -source 1.3

error: Source option 1.3 is no longer supported. Use 6 or later.

error: Target option 1.1 is no longer supported. Use 1.6 or later.

make[4]: *** [gnu/gettext/GettextResource.class] Error 2

make[3]: *** [all-recursive] Error 1

make[2]: *** [all] Error 2

make[1]: *** [all-recursive] Error 1

make: *** [all] Error 2

error building gettext`