awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #102: Setting the IP Camera as a Media Source.

pravin82, 2018-07-19:
I want to send video to AWS server from IP Camera, but I am not able to understand that how to set an IP camera as a Media Source?

chehefen, 2018-07-19:
are you able to find the ip address of the camera? if so, you can use the kinesis_video_gstreamer_sample_rtsp_app.

pravin82, 2018-07-20:
Hi @chehefen  I have tried to run the kinesis_video_gstreamer_sample_rtsp_app by using IP address of camera, but got the following error

"appsink configured
[DEBUG][2018-07-20 17:20:09] Pipeline built, elements *NOT* yet linked
[DEBUG][2018-07-20 17:20:09] Pipeline playing
Error received from element source: Could not read from resource.
Debugging information: gstrtspsrc.c(5464): gst_rtspsrc_try_send (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Could not receive message. (Generic error)"
Any help will be great

pravin82, 2018-07-21:
I am using a Chineses camera of company V380 model no:IPC-T9810-Q5(T). 
I tried using different rtsp url but sometime I get the error "Error received from element source: Could not open resource for reading and writing." and sometime I get the error "Could not read from resource."
Different urls that I have tried are
rtsp://192.168.2.13/live/ch00_1
rtsp://username:password@ipnumber/live/ch00_1

pravin82, 2018-07-21:
I think the error is due to the fact that my camera does not have port 554 open and probably it does not support rtsp.Camera have port 5050 open and it supports MMCC.
is there any way that I can stream video to AWS KInesis Video by using MMCC.

unicornss, 2018-07-25:
@pravin82  How are you viewing the video stream from the camera now?
 
Not sure about MMCC but it's very straightforward to create the pipeline for RTSP (and RTMP) streaming and send to KVS. You can check if this camera supports RTSP protocol in the manufacturer's support or in the URL mentioned by @chehefen  in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/104#issuecomment-407478332.

closed 2018-08-08 by turb0kat