awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #104: rtsp url of Sricam SP012 as the camera media source

anmol1ratn, 2018-07-24:
I'm have Sricam SP012 IP camera which I want to use as the camera source. I'm using Ubuntu 18.04.
But I don't know how to find rtsp url of the camera. It's one of the tried setup as mentioned in Kinesis-Video Developer Guide. Can you help me with this?
@chehefen 
@MushMal

chehefen, 2018-07-24:
can you try rtsp://admin:password@ip/onvif1 ? If that does not work. Can you try the urls listed here? https://www.ispyconnect.com/man.aspx?n=Sricam

anmol1ratn, 2018-07-29:
@chehefen 
Sorry for replying late. I tried the rtsp url which you mentioned and I got the following error.

```
[DEBUG][2018-07-29 15:37:41] Kinesis Video stream VideoStream is Ready.
[DEBUG][2018-07-29 15:37:41] Stream is ready
[DEBUG][2018-07-29 15:37:41] Pipeline OK
[DEBUG][2018-07-29 15:37:41] rtsp://admin:888888@ip/onvif1
[DEBUG][2018-07-29 15:37:41] appsink configured
[DEBUG][2018-07-29 15:37:41] Pipeline built, elements *NOT* yet linked
[DEBUG][2018-07-29 15:37:41] Pipeline playing
Error received from element source: Could not open resource for reading and writing.
Debugging information: gstrtspsrc.c(6775): gst_rtspsrc_retrieve_sdp (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Failed to connect. (Generic error)
[INFO ][2018-07-29 15:37:41] 
freeKinesisVideoClient(): Freeing Kinesis Video Client
```

Do I also have to configure the camera in any way? Is there any guide to achieve this? Coz it's one of the tried setups as mentioned in dev guide.

chehefen, 2018-07-30:
do you know what the ip address of your camera is? For example, if the ip address is 192.168.0.1, then the rtsp url could be rtsp://admin:password@192.168.0.1/onvif1.

anmol1ratn, 2018-07-31:
Okay, I got it working. Thanks for the help. :)

closed 2018-07-31 by anmol1ratn