awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #107: Amazon Linux ECR image is not available anymore

AndrewChubatiuk, 2018-08-01:
Cannot download image amazon linux from ECR
```
$(aws ecr get-login --no-include-email --region us-west-2 --registry-ids 546150905175)
docker pull 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest

Error response from daemon: pull access denied for 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux, repository does not exist or may require 'docker login'
```

but works as expected for raspberry-pi
```
$(aws ecr get-login --no-include-email --region us-west-2 --registry-ids 546150905175)
docker pull 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest

latest: Pulling from kinesis-video-producer-sdk-cpp-raspberry-pi
6e71256bf6a8: Pulling fs layer 
686910442b38: Pulling fs layer 
363144da4f64: Pulling fs layer 
d0d5fe8fb182: Waiting 
97ebef7d305c: Waiting 
622fccb0dd60: Waiting 
51cb35310d3d: Waiting 
8064f136628b: Waiting 
7eaa7272d90f: Waiting 
0c487ffa1ca7: Waiting 
2845601b4fe3: Waiting 
830fab17e676: Waiting 
2a646330310e: Waiting 
46fb8a41365c: Waiting 
5ca34234dc75: Waiting 
cdc78c0dac10: Waiting 
cb886c2dccd6: Waiting 
632900ce2454: Waiting 
b75514a22787: Waiting 
fb287cf27858: Waiting 
```

zhiyua-git, 2018-08-01:
Hi @AndrewChubatiuk,

I just confirmed it is working by running docker pull.
```
docker pull 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest
latest: Pulling from kinesis-video-producer-sdk-cpp-amazon-linux
c8c98d2e5e16: Downloading [=========>                                         ]  11.33MB/60.24MB
7dc6c939a9e7: Downloading [===>                                               ]  13.49MB/185.3MB
ed2a500b3a25: Downloading [======================>                            ]  11.98MB/26.83MB
dc6377b3febc: Waiting 
f187db20fe32: Waiting 
81729481a43f: Waiting 
17d3e7edacb9: Waiting 
```

Can you take another try?

Thanks,
-ZH

AndrewChubatiuk, 2018-08-01:
Now it works, thanks

zhiyua-git, 2018-08-01:
Great to hear it works! I will close this issue for now, if you cannot get access to the docker image again, feel free to reopen this.

closed 2018-08-01 by zhiyua-git