awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #108: Convert get_time to strptime in KvsSinkRotatingCredentialProvider

nickrobinson, 2018-08-03:
When trying to cross-compile the gstreamer plugin for an armv7hf platform I ran into a compilation error:

```
error: 'get_time' is not a member of 'std'
         if (ss >> std::get_time(&timeinfo, "%Y-%m-%dT%H:%M:%SZ"))
```

From what I can tell you must be using GCC > 5.0 in order for get_time to be found during compilation according to https://gcc.gnu.org/bugzilla/show_bug.cgi?id=54354#c9 and https://stackoverflow.com/questions/30043293/c-error-function-get-time-could-not-be-resolved-ubuntu-g.

Could you instead use strptime instead with something like:

```
if (strptime(expiration_string.c_str(), "%Y-%m-%dT%H:%M:%SZ", &timeinfo) != NULL) {
        std::time_t tt = std::mktime(&timeinfo);
        system_clock::time_point tp = system_clock::from_time_t (tt);
        expiration = std::chrono::duration_cast<std::chrono::seconds>(tp.time_since_epoch());
  } else {
         LOG_ERROR("Failed to parse AWS_TOKEN_EXPIRATION.");
         goto CleanUp;
    }
```

This would make it easier for those on older versions of GCC to to be able to compile the Gstreamer plugin

chehefen, 2018-08-03:
Unfortunately, the plugin is basically a wrapper of our producer library which requires C++14 features. I think even if you get through this error you will run into trouble when linking with producer.

nickrobinson, 2018-08-07:
I made this change and built locally. I have deployed a new version of the kvssink plugin with this change and it is working successfully on my dev cameras. This doesn' t have as much to do with C++14 features as it does with an underlying lack of support for std::get_time in all GCC versions < 4.9. As far as I can tell this one instance of get_time is the only thing that will block customers using versions of GCC < 4.9 from being able to compile the kvssink plugin. 

This is a particular problem for customers who are targeting embedded platforms and have vendor supplied toolchains that may not have up to date versions of GCC.

chehefen, 2018-08-07:
Can you tell us how did you cross compile the producer library which is an dependency of the gstreamer plugin? Can you describe the process of how you cross-compile the sdk? Thanks.

nickrobinson, 2018-08-08:
I use CodeBuild with the following [buildspec file](https://github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/buildspec.yml). I am using a [dockcross](https://hub.docker.com/r/dockcross/linux-armv7/) docker image as my build container to pick up the ARMv7 toolchain in order to cross-compile the producer library.

I also had to slightly modify the install-script in order to cross-compile for my platform: https://github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/install-script

MushMal, 2018-08-09:
@nickrobinson It seems that the issue is with the particular toolchain semi-supporting C++14. I am trying to understand this issue better. The C++ SDK uses C++14 features so the compiler should support it. It seems that GCC 4.9 does have a support for C++14 as you can compile the C++ SDK codebase and it fails only on this issue while compiling the GST plugin.

std::get_time is supported since C++11 (according to https://en.cppreference.com/w/cpp/io/manip/get_time) but this is not the "main" compiler feature but rather the CRT which for gcc is implemented in libstdc++. This particular feature is added in GCC 5.1 it seems.

As you've mentioned, we could potentially use strptime but the problem would then be the portability - for example, there is no equivalent of strptime on Windows platform (there is, however, an open source alternative).

There are few approaches that need to be looked at, including implementing the functionality directly or trying to pull down or build libstdc++ which includes the std::get_time and getting the compiler to use this particular CRT.

nickrobinson, 2018-08-09:
Understood. The portability issue makes sense. I had not thought about individuals using this application on Windows platforms. Could we use [predefined compiler](https://sourceforge.net/p/predef/wiki/Compilers/) macros in order to detect when users are going to run into this issue? I know this involves creating two different code paths here which makes this code a bit more difficult to maintain, but it is a matter of time until another customer runs into this same issue.

For instance:
```
#if defined(__GNUC__) && (__GNUC___ < 5)
  if (strptime(expiration_string.c_str(), "%Y-%m-%dT%H:%M:%SZ", &timeinfo) != NULL) {
    std::time_t tt = std::mktime(&timeinfo);
    system_clock::time_point tp = system_clock::from_time_t(tt);
    expiration = std::chrono::duration_cast<std::chrono::seconds>(tp.time_since_epoch());
  } else {
    LOG_ERROR("Failed to parse AWS_TOKEN_EXPIRATION.");
    goto CleanUp;
  }
#else
  std::tm timeinfo = std::tm();
  std::istringstream ss(expiration_string);
  if (ss >> std::get_time(&timeinfo, "%Y-%m-%dT%H:%M:%SZ")) {
    std::time_t tt = std::mktime(&timeinfo);
    system_clock::time_point tp = system_clock::from_time_t (tt);
    expiration = std::chrono::duration_cast<std::chrono::seconds>(tp.time_since_epoch());
  } else {
    LOG_ERROR("Failed to parse AWS_TOKEN_EXPIRATION.");
    goto CleanUp;
  }
#else
```

MushMal, 2018-08-09:
Thanks @nickrobinson , perhaps you are right, we could leave out the odd case of GCC 4.9 on Windows for a time being. I would also rather #ifdef the given functionality that's needed (std::get_time and strptime) behind a definition and then use that functionality streamlined in the code. Will update asap!

MushMal, 2018-08-31:
@nickrobinson, the latest release should have adequate coverage for the GCC 4.9 standard C++ libraries missing support for C++ 14 constructs.

Marking as assumed answered.

Thanks again for bringing it up to our attention.

unicornss, 2018-09-12:
@nickrobinson  The latest release addresses the issue. Closing this issue, for now, please reopen if required.  Thanks

closed 2018-09-12 by unicornss

texano00, 2018-11-01:
Hi @nickrobinson ,
i've tried to build your repository through CodeBuild as you said but i receive this error
`/bin/sh: 1: arm-linux-gnueabihf-gcc-4.9: not found`.

Full logs:

```[Container] 2018/11/01 09:13:16 Waiting for agent ping
[Container] 2018/11/01 09:13:19 Waiting for DOWNLOAD_SOURCE
[Container] 2018/11/01 09:13:21 Phase is DOWNLOAD_SOURCE
[Container] 2018/11/01 09:13:21 CODEBUILD_SRC_DIR=/codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp
[Container] 2018/11/01 09:13:21 YAML location is /codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/buildspec.yml
[Container] 2018/11/01 09:13:21 Processing environment variables
[Container] 2018/11/01 09:13:21 Moving to directory /codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp
[Container] 2018/11/01 09:13:21 Registering with agent
[Container] 2018/11/01 09:13:21 Phases found in YAML: 1
[Container] 2018/11/01 09:13:21 BUILD: 5 commands
[Container] 2018/11/01 09:13:21 Phase complete: DOWNLOAD_SOURCE Success: true
[Container] 2018/11/01 09:13:21 Phase context status code: Message:
[Container] 2018/11/01 09:13:21 Entering phase INSTALL
[Container] 2018/11/01 09:13:21 Phase complete: INSTALL Success: true
[Container] 2018/11/01 09:13:21 Phase context status code: Message:
[Container] 2018/11/01 09:13:21 Entering phase PRE_BUILD
[Container] 2018/11/01 09:13:21 Phase complete: PRE_BUILD Success: true
[Container] 2018/11/01 09:13:21 Phase context status code: Message:
[Container] 2018/11/01 09:13:21 Entering phase BUILD
[Container] 2018/11/01 09:13:21 Running command cd $CODEBUILD_SRC_DIR/kinesis-video-native-build
 [Container] 2018/11/01 09:13:21 Running command pwd
/codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
 [Container] 2018/11/01 09:13:21 Running command ls
CMakeLists.txt
gstreamer-plugin-install-script
install-script
java-install-script
kvs_log_configuration
min-install-script
samples
 [Container] 2018/11/01 09:13:21 Running command ./install-script -a
 ===========================Starting to build ===========================
 Kinesis Video Stream Producer SDK and Demo Application
 ========================================================================
 Following open source libraries and tools will be downloaded and built
 before building the producer sdk.
 ========================================================================
 # ------------------------- BUILD-TOOLS -------------------------------#
 autoconf automake
 bison curl
 flex gettext
 googletest iconv
 libffi libpcre
 libtools log4cplus
 m4 openssl
 zlib
 # ------------------------- GSTREAMER - for demo application --------#
 GStreamer, GStreamer plugins, yasm, gst-libav and x264.
 ------------------------------------------------------------------------
 *** NOTE *** for MacOS High Sierra: Use
 brew install autoconf
 brew install bison before running this script
 ------------------------------------------------------------------------
 ========================================================================
 This install script downloads and builds the dependent libraries from
 source in the ./kinesis-video-native-build/downloads directory only.
 ========================================================================
 Starting auto-build
Include building GStreamer extra plugins= Y
Additional curl options=
cmake version 3.11.0 CMake suite maintained and supported by Kitware (kitware.com/cmake).
cert.pem file does not exist in the /codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build directory
Downloading the https://www.amazontrust.com/repository/SFSRootCAG2.pem to /codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/certs/cert.pem
 % Total % Received % Xferd Average Speed Time Time Time Current
[... 2663 lines cut ...]
make[1]: Entering directory '/codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/openssl-1.1.0f'
( trap "rm -f crypto/aes/aes-armv4.s.*" INT 0; \
 CC="arm-linux-gnueabihf-gcc-4.9" /usr/bin/perl crypto/aes/asm/aes-armv4.pl linux32 crypto/aes/aes-armv4.s.S; \
 arm-linux-gnueabihf-gcc-4.9 -I. -Icrypto/include -Iinclude -Icrypto -DDSO_DLFCN -DHAVE_DLFCN_H -DNDEBUG -DOPENSSL_THREADS -DOPENSSL_NO_STATIC_ENGINE -DOPENSSL_PIC -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DAES_ASM -DBSAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DPOLY1305_ASM -DOPENSSLDIR="\"/codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/\"" -DENGINESDIR="\"/codebuild/output/src030449628/src/github.com/nickrobinson/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/engines-1.1\"" -Wall -O3 -pthread -E crypto/aes/aes-armv4.s.S | \
 /usr/bin/perl -ne '/^#(line)?\s*[0-9]+/ or print' > crypto/aes/aes-armv4.s.i && \
 mv -f crypto/aes/aes-armv4.s.i crypto/aes/aes-armv4.s )
[... cut ...]
[Container] 2018/11/01 09:13:23 Phase context status code: COMMAND_EXECUTION_ERROR Message: Error while executing command: ./install-script -a. Reason: exit status 2
[Container] 2018/11/01 09:13:24 Entering phase POST_BUILD
[Container] 2018/11/01 09:13:24 Phase complete: POST_BUILD Success: true
[Container] 2018/11/01 09:13:24 Phase context status code: Message:
[Container] 2018/11/01 09:13:24 Phase complete: UPLOAD_ARTIFACTS Success: true
[Container] 2018/11/01 09:13:24 Phase context status code: Message:```