awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #109: RTSP demo app failing with error InvalidCodecPrivateDataException

ankit3600, 2018-08-16:
I am getting the below error in terminal:

>[ERROR][2018-08-16 18:26:27] 
on_new_sample(): kinesis_video_flags=0 pts=957358149 dts=957358149

[fullErrorLog.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2294888/fullErrorLog.txt)

On the aws console, I am getting the below error:

>Unexpected error
Exception: InvalidCodecPrivateDataException
Status code: 400
Request ID: c117a413-5b4f-a877-92f8-6039ce303a40
This may be an intermittent issue. Retry the same action, or try refreshing your browser tab.

[image]

These are my system specs:

>cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

>uname -r
4.15.0-32-generic

>dpkg --get-selections | grep -i gst
gir1.2-gst-plugins-base-1.0			install
gir1.2-gstreamer-1.0				install
gstreamer1.0-alsa:amd64				install
gstreamer1.0-clutter-3.0:amd64			install
gstreamer1.0-gl:amd64				install
gstreamer1.0-packagekit				install
gstreamer1.0-plugins-base:amd64			install
gstreamer1.0-plugins-base-apps			install
gstreamer1.0-plugins-good:amd64			install
gstreamer1.0-pulseaudio:amd64			install
gstreamer1.0-tools				install
gstreamer1.0-x:amd64				install
libclutter-gst-3.0-0:amd64			install
libgstreamer-gl1.0-0:amd64			install
libgstreamer-plugins-base1.0-0:amd64		install
libgstreamer-plugins-good1.0-0:amd64		install
libgstreamer1.0-0:amd64				install
libreoffice-avmedia-backend-gstreamer		install

ankit3600, 2018-08-20:
This is the link on amazon:
https://www.amazon.in/ProElite-IP01A-Wireless-Security-supports/dp/B079JL696S

This is the link on camera's website:
http://www.proelite.co.in/ProElite/ProElite-IP01AX-2MP-Full-HD--1920x1080--WiFi-Wireless-IP-Security-Camera-CCTV--Watch-LIVE-Demo---supports-upto-128-GB-SD-card---Dual-Antenna---Black-id-3564689.html

I am able to play the camera successfully in vlc media player by giving the rtsp url (rtsp://192.168.0.2/11). In vlc media player configuration, it shows that the video is H264 encoded.

unicornss, 2018-08-20:
thanks @ankit3600 

Could you please send the logs with the debug enabled GST_DEBUG=5? This should help us to understand what source format is recognized from the camera by the GStreamer pipeline..

**Example:**
```
-ThinkPad-X1-Yoga-1st:~/work/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$GST_DEBUG=5 ./kinesis_video_gstreamer_sample_rtsp_app 
 ```

ankit3600, 2018-08-21:
ankit@ankit-ThinkPad-X1-Yoga-1st:~/work/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ GST_DEBUG=5 ./kinesis_video_gstreamer_sample_rtsp_app "rtsp://192.168.0.5/11" "KinesisTestStreamAnkit" |& tee ~/out.txt

Please find the output file (out.txt) in the below link:
https://drive.google.com/file/d/1lX8_lNsw8ZYNfHfWn6_JoktTvb7gOrnq/view?usp=sharing

FYI screenshot of codec info in vlc media player
[image]

unicornss, 2018-08-23:
Hi @ankit3600  

I do see that the video stream has code private data.  Could it be possible that the video source settings were changed at the time of streaming? Can you create a new stream and check if that helps...

> 0x7fb8a000e6d0, time 99:99:99.999999999, seq-num 261, GstEventCaps, caps=(GstCaps)"video/x-h264\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au\,\ codec_data\=\(buffer\)014d002affe1000b674d002a95a81e0089f95001000468ee3c80\,\ level\=\(string\)4.2\,\ profile\=\(string\)main";

ankit3600, 2018-08-23:
I have tested it many times (doing the testing for last 2 weeks).
I have not changed anything in video source settings (don't even know how to do that).
Everything is at its default state (camera, vlc, kinesis demo app and any settings that it may have).

This isssue is always present.

unicornss, 2018-08-24:
Hi @ankit3600, Thanks for the logs.

1.  I extracted the following codec_data from your logs and noticing that it does change in the middle of stream.

```
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80
 codec_data=(buffer)014d002affe1000b674d002a95a81e0089f95001000468ee3c80
 codec_data=(buffer)014d0029ffe10015474d002999a01e0089f96100000303e800009c408401000448ee3c80

```
This is not typically expected and could cause the playback to display codec data change the error message.

2. The second aspect in your camera codec data is that the codec_data 

> 014d0029ffe10015**47**4d002999a01e0089f96100000303e800009c408401000448ee3c80  
 causes fail in validation check both in ingest side and playback side, where as the codec_data

> 014d002affe1000b**67**4d002a95a81e0089f95001000468ee3c80 should work fine.

Though revising the line CHK(read == SPS_NALU_67 || read == SPS_NALU_27, STATUS_MKV_INVALID_H264_H265_SPS_NALU);  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/74cc57332d089e6ac8b932b3cd5d6c6b59c03667/kinesis-video-pic/src/mkvgen/src/SpsParser.cpp#L243 

to  CHK(read == SPS_NALU_67 || read == SPS_NALU_27 || read == 0x47, STATUS_MKV_INVALID_H264_H265_SPS_NALU); should ingest the video for that frame it may still fail validation on the playback.

So, I would suggest trying the h264 profile in the camera as **baseline** profile instead of main profile 

> (profile=(string)main" )

using the camera configuration/setup instructions provided by the camera manufacturer. Let us know if helps to resolve codec_data change issue.  Unfortunately, we don't have this particular model to recreate the issue locally. 

Also, could you tell us more about how are you planning to use the Kinesis Video Streams?
 
Thanks

unicornss, 2018-10-03:
Some IP cameras provide a web interface to configure the settings.  your camera  h264 encoder options generate video stream in the 
`(profile=(string)main" )`.

I was referring to the possibility of changing the configuration (through the camera web interface if available) to a baseline profile and see if that helps in the _codec data change_ during the video-streaming.

unicornss, 2018-10-04:
Closing the issue as it has been answered. Please reopen or create a new issue with details on the support required. Thanks.

closed 2018-10-04 by unicornss

zebpaypeswani, 2019-02-25:
@unicornss I am getting same error as above I am simply trying to put an locally stored mkv file 

```
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181432392682062618274554370250778456', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181432392682062618274554370250778456', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181432392682062618274554370250778456', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181437344442219759795658471381552466', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181437344442219759795658471381552466', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181437344442219759795658471381552466', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181442296202376901316762628670330617', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181442296202376901316762628670330617', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181442296202376901316762628670330617', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181447247962534042837866811084792255', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181447247962534042837866811084792255', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181447247962534042837866811084792255', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181452199722691184358970950757361025', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181452199722691184358970950757361025', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181452199722691184358970950757361025', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181457151482848325880075057651118352', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181457151482848325880075057651118352', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181457151482848325880075057651118352', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181462103243005467401179214107806713', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181462103243005467401179214107806713', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181462103243005467401179214107806713', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181467055003162608922283293743845045', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181467055003162608922283293743845045', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181467055003162608922283293743845045', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181472006763319750443387341676117927', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181472006763319750443387341676117927', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181472006763319750443387341676117927', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181476958523476891964491557115294611', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181476958523476891964491557115294611', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181476958523476891964491557115294611', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181481910283634033485595743756007180', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181481910283634033485595743756007180', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181481910283634033485595743756007180', errorCode=null, errorId=null}
onComplete
[... cut ...]
onAckEvent AckEvent{ackEventType=RECEIVED, fragmentTimecode=0, fragmentNumber='91343852333181600752527405429992096452730593687', errorCode=null, errorId=null}
onAckEvent AckEvent{ackEventType=PERSISTED, fragmentTimecode=0, fragmentNumber='91343852333181600752527405429992096452730593687', errorCode=null, errorId=null}
onComplete
onAckEvent AckEvent{ackEventType=BUFFERING, fragmentTimecode=0, fragmentNumber='91343852333181605704287562571513200555707130611', errorCode=null, errorId=null}
```

But at aws console video is not showing

MushMal, 2019-02-25:
@zebpaypeswani your stream is being accepted, indexed and persisted indicating that you had a valid MKV format being uploaded. However, in order for the console to playback the stream, it has to be a valid "video/h264" type which is in AvCC format instead of Annex-B or raw. It should also have a valid Codec Private Data (which is SPS/PPS for h264). 

Could you validate it?

An alternative is to use the C++ SDK sample application to upload the existing file.

zebpaypeswani, 2019-02-27:
@MushMal I recorded a video from my mobile and converted it using `ffmpeg -i input.mp4 -b:v 10M -minrate 10M -maxrate 10M -bufsize 10M -bf 0 input.mkv`  is there anything else I can do to get it in AvCC format?
here is some details I get from `ffprobe` 
https://pastebin.com/TWbQDMPi

unicornss, 2019-02-27:
Hi @zebpaypeswani  

As @MushMal  has mentioned earlier, the stream is getting ingested without any issues. 

Could you please create a new stream and try sending ? Also,  send us the details from ffprobe on the mp4 file to kinesis-video-support@amazon.com and screenshot of the error you are seeing in the console.

Thanks

zebpaypeswani, 2019-03-01:
@unicornss I am doing face matching the photos are indexed in s3 bucket in a collection. If video is getting ingested how can I get data from kineses data stream which will tell face is matched 
I see here an example https://docs.amazonaws.cn/en_us/streams/latest/dev/kcl2-standard-consumer-java-example.html but I am not able to figure out how to get the data.

unicornss, 2019-03-01:
Hi @zebpaypeswani ,

Please refer Kinesis Video Streams Parser Library examples for additional details on how to perform further processing, recognition integration and analytics.

- https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/test/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoRekognitionIntegrationExampleTest.java
- 
- https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/main/java/com/amazonaws/kinesisvideo/parser/examples/BoundingBoxImagePanel.java

Thanks

qqia, 2019-03-01:
Hi @zebpaypeswani ,

Please make sure you are using latest code running below example

*  https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/test/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoRekognitionIntegrationExampleTest.java

We have just published a fix for a bug.

zebpaypeswani, 2019-03-04:
@qqia @unicornss I ran the example but getting below errors

```
2019-03-04 13:14:16 WARN  BasicProfileConfigLoader:Your profile name includes a 'profile ' prefix. This is considered part of the profile name in the Java SDK, so you will need to include this prefix in your profile name when you reference this profile from your Java code.
s2019-03-04 13:14:16 WARN  BasicProfileConfigLoader:Your profile name includes a 'profile ' prefix. This is considered part of the profile name in the Java SDK, so you will need to include this prefix in your profile name when you reference this profile from your Java code.
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Arn : arn:aws:rekognition:us-west-2:655136467581:streamprocessor/FaceMatching
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Arn : arn:aws:rekognition:us-west-2:655136467581:streamprocessor/FaceMatching
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Input kinesisVideo stream : arn:aws:kinesisvideo:us-west-2:655136467581:stream/FaceMatching/1550831945886 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Input kinesisVideo stream : arn:aws:kinesisvideo:us-west-2:655136467581:stream/FaceMatching/1550831945886 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Output kinesisData stream arn:aws:kinesis:us-west-2:655136467581:stream/FaceStream 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Output kinesisData stream arn:aws:kinesis:us-west-2:655136467581:stream/FaceStream 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:RoleArn arn:aws:iam::655136467581:role/face_matching 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:RoleArn arn:aws:iam::655136467581:role/face_matching 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:CollectionId FaceCollection 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:CollectionId FaceCollection 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Status FAILED 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Status FAILED 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Status message The requested stream is not found or not active. 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Status message The requested stream is not found or not active. 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Creation timestamp Fri Feb 22 19:12:27 IST 2019 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Creation timestamp Fri Feb 22 19:12:27 IST 2019 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Last update timestamp Mon Mar 04 13:10:08 IST 2019 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:Last update timestamp Mon Mar 04 13:10:08 IST 2019 
2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:SdkResponseMetadata : {AWS_REQUEST_ID=510a8e91-3e51-11e9-b3f0-59c6199b1664} 
s2019-03-04 13:14:18 INFO  RekognitionStreamProcessor:SdkResponseMetadata : {AWS_REQUEST_ID=510a8e91-3e51-11e9-b3f0-59c6199b1664} 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Arn : arn:aws:rekognition:us-west-2:655136467581:streamprocessor/FaceMatching
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Arn : arn:aws:rekognition:us-west-2:655136467581:streamprocessor/FaceMatching
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Input kinesisVideo stream : arn:aws:kinesisvideo:us-west-2:655136467581:stream/FaceMatching/1550831945886 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Input kinesisVideo stream : arn:aws:kinesisvideo:us-west-2:655136467581:stream/FaceMatching/1550831945886 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Output kinesisData stream arn:aws:kinesis:us-west-2:655136467581:stream/FaceStream 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Output kinesisData stream arn:aws:kinesis:us-west-2:655136467581:stream/FaceStream 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:RoleArn arn:aws:iam::655136467581:role/face_matching 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:RoleArn arn:aws:iam::655136467581:role/face_matching 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:CollectionId FaceCollection 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:CollectionId FaceCollection 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Status STARTING 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Status STARTING 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Status message null 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Status message null 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Creation timestamp Fri Feb 22 19:12:27 IST 2019 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Creation timestamp Fri Feb 22 19:12:27 IST 2019 
2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Last update timestamp Mon Mar 04 13:14:18 IST 2019 
s2019-03-04 13:14:20 INFO  RekognitionStreamProcessor:Last update timestamp Mon Mar 04 13:14:18 IST 2019 
2019-03-04 13:14:20 INFO  LeaseCoordinator:With failover time 10000 ms and epsilon 25 ms, LeaseCoordinator will renew leases every 3308 ms, takeleases every 20050 ms, process maximum of 2147483647 leases and steal 1 lease(s) at a time.
s2019-03-04 13:14:20 INFO  LeaseCoordinator:With failover time 10000 ms and epsilon 25 ms, LeaseCoordinator will renew leases every 3308 ms, takeleases every 20050 ms, process maximum of 2147483647 leases and steal 1 lease(s) at a time.
[... cut ...]
s2019-03-04 13:15:21 INFO  Worker:Current stream shard assignments: shardId-000000000009, shardId-000000000008, shardId-000000000019, shardId-000000000007, shardId-000000000018, shardId-000000000006, shardId-000000000017, shardId-000000000005, shardId-000000000016, shardId-000000000004, shardId-000000000015, shardId-000000000003, shardId-000000000014, shardId-000000000002, shardId-000000000013, shardId-000000000001, shardId-000000000012, shardId-000000000000, shardId-000000000011, shardId-000000000010
2019-03-04 13:15:21 INFO  Worker:Sleeping ...
s2019-03-04 13:15:21 INFO  Worker:Sleeping ...
```

zebpaypeswani, 2019-03-04:
@qqia @unicornss I am getting this error when running example 
https://pastebin.com/jisMD3pt

qqia, 2019-03-04:
hi @zebpaypeswani ,

I saw quite some exceptions like

```
 Unable to execute HTTP request: dynamodb.us-west-2.amazonaws.com
```

```
Unable to execute HTTP request: monitoring.us-west-2.amazonaws.com
```

Can you check if its DNS issue or something?