awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #112: Error running GStreamer plugin in Docker Container

RafaelRViana, 2018-08-31:
I'm running Docker 17.06 on Mac OSX High Sierra. 

I followed this guide: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker
To use GStreamer Plugin in Docker.

However, when I ran:
`gst-launch-1.0 rtspsrc location="rtsp url” short-header=TRUE !rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=“Test” storage-size=512
`
as described in: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-launch

I get this error:
[DEBUG][2018-08-31 21:09:27] Awaiting for the stream to become ready...
[INFO ][2018-08-31 21:09:27] Refreshing credentials. Force refreshing: 0 Now time is: 1535749767 Expiration: 0
[WARN ][2018-08-31 21:09:28] HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=/20180831/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=a50c7cccd637fca1d8f42426a7c15b2501dbb9583aa52dc04efc8f2003da33bf
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180831T210927Z
**[DEBUG][2018-08-31 21:09:28] describeStream response: {"message":"The security token included in the request is invalid."}**
**[INFO ][2018-08-31 21:09:28] Describe stream did not find the stream Test in Kinesis Video (stream will be created)**
[INFO ][2018-08-31 21:09:28] 
describeStreamResultEvent(): Describe stream result event.
[ERROR][2018-08-31 21:09:28] Submitting event result for stream: 10404414 failed with: 1375731728
[ERROR][2018-08-31 21:09:28] Reporting stream error. Errored timecode: 0 Status: 0x52000010
[ERROR][2018-08-31 21:09:57] Failed to create Kinesis Video Stream - timed out.
[INFO ][2018-08-31 21:09:57] Freeing Kinesis Video Stream Test
[INFO ][2018-08-31 21:09:57] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
ERROR: Pipeline doesn't want to pause.
**ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: Could not initialize supporting library.**

zhiyua-git, 2018-08-31:
Hi @RafaelRViana,

Can you try this? You need to provide accessKey/secretKey to send the video stream.

$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="plugin" storage-size=512 **access-**="YourAccessKey" secret-key="YourSecretKey"**

We will update the document soon, sorry for the missing instructions for Mac and Raspberry PI.

For more details, you can also read this [link](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#running-examples).

Thanks,
-ZH

RafaelRViana, 2018-09-01:
Thanks. It worked.

However, I'm with error below (RTSP URL is right)

```
[[DEBUG][2018-09-01 04:26:48] streamReadyHandler invoked
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://user:password@url:port/cam/realmonitor?channel=1&subtype=1
**ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not open resource for reading and writing.**
Additional debug info:
gstrtspsrc.c(6775): gst_rtspsrc_retrieve_sdp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Failed to connect. (Generic error)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...](url)
```

nikparmar, 2018-09-03:
@RafaelRViana this error is mostly because of wrong credentials for RTSP url

RafaelRViana, 2018-09-03:
I think I discovered error:

I'm testing with a Dahua PTZ the RTSP url is as follow:
rtsp://user:pass@ip:port/cam/realmonitor?**channel=1&subtype=1**

The command I used:
`gst-launch-1.0 rtspsrc location=rtsp://user:pass@ip:port/cam/realmonitor?channel=1&subtype=1 short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=Test storage-size=512 access-key="MY_ACCESS_KEY" secret-key="MY_SECRET_KEY"`

The output log:
```
Progresso: (open) Opening Stream
Progresso: (connect) **Connecting to rtsp://user:pass@IP:port/cam/realmonitor?channel=1**
Progresso: (open) Retrieving server options
Progresso: (open) Retrieving media info
ERRO: do elemento /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Not found
Informação adicional de depuração:
gstrtspsrc.c(5576): gst_rtspsrc_send (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Not Found (404)
ERRO: a fila de processamento não quer preparar.
```

Maybe there is some part in code escaping the args after &

If i try to open in VLC:
rtsp://user:pass@IP:port/cam/realmonitor?channel=1 (VLC Crash)
rtsp://user:pass@IP:port/cam/realmonitor?channel=1&subtype=1 (VLC Works)

As code is trying to open rtsp as rtsp://user:pass@IP:port/cam/realmonitor?channel=1 it get's that error: **ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not open resource for reading and writing.**

RafaelRViana, 2018-09-03:
I tested with an Avigilon camera that has the following RTSP URL: "rtsp://user:pass@ip/defaultPrimary?streamType=u"

It worked with gst-launch1.0 and kinesis_video_gstreamer_sample_rtsp_app.

With Dahua PTZ with multiple args (&). 
In gst-launch1.0 happens error above: /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Not found

In kinesis_video_gstreamer_sample_rtsp_app I get the error:  "Usage: AWS_ACCESS_KEY_ID=SAMPLEKEY AWS_SECRET_ACCESS_KEY=SAMPLESECRET ./kinesis_video_gstreamer_sample_app rtsp-url my-stream-name"
As descibred in Line 358 - https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp

RafaelRViana, 2018-09-03:
That's it.

I discovered an alternative RTSP URL for Dahua`s camera: `rtsp://user:pass@ip:port/live`
It's working in my Mac and in Docker.

It would be good to support RTSP with & args, since I can choose the stream used for streaming (primary or secondary).

chehefen, 2018-09-04:
Hi,
Can you try put the url in double quotes and see what happens? We tried a rtsp url like "rtsp://user:pass@url/axis-media/media.amp?videocodec=h264&resolution=640x480" using gst-launch-1.0 and it worked. The error message indicates that the camera feed could not be opened at all. Could you try `gst-launch-1.0 -v playbin uri="rtsp://user:pass@ip:port/cam/realmonitor?channel=1&subtype=1"`and see if that works?

RafaelRViana, 2018-09-06:
`gst-launch-1.0 -v playbin uri="rtsp://user:pass@ip:port/cam/realmonitor?channel=1&subtype=1"`
works.

---

`gst-launch-1.0 rtspsrc location=“rtsp://user:pass@ip:port/cam/realmonitor?channel=1&subtype=1” short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=Test storage-size=512 access-key="" secret-key=""`

Doesn't work with following error: gstrtspsrc.c(6766): gst_rtspsrc_retrieve_sdp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
No valid RTSP URL was provided 

It doesn't allow location with quotes.

zhiyua-git, 2018-09-06:
Hi @RafaelRViana ,

The **“** you used in `gst-launch-1.0 rtspsrc location=“rtsp://user:pass@ip:port/cam/realmonitor?channel=1&subtype=1” short-header=TRUE ...` is different from **"** in `gst-launch-1.0 -v playbin uri="rtsp://user:pass@ip:port/cam/realmonitor?channel=1&subtype=1"`. Is it the reason for no valid RTSP URL?

-ZH

unicornss, 2018-10-04:
Closing the issue as it has been answered. Please reopen or create a new issue with details on the support required. Thanks.

closed 2018-10-04 by unicornss