awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #113: Error installing kvssink locally on MacOSX

RafaelRViana, 2018-09-03:
Hi,

As described in other issue created on Friday. I'm with problems to run GStreamer plugin on Docker. So, I'm testing to run it locally in my MacOSX. But, I couldn't get it working.

1) As described in: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md", the install script is supposed to install kvssink.

"Note that the install-script also builds the Kinesis Video Streams producer SDK as a GStreamer plugin (kvssink)."

I have successfully installed it as log below:
```
Include building GStreamer extra plugins= Y
Additional curl options=
cmake version 3.9.1 CMake suite maintained and supported by Kitware (kitware.com/cmake).
Checking openssl at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libssl.a
Checking curl at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcurl.a
Checking log4cplus at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus.dylib/.so
Checking gettext at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgettextlib.dylib/.so
Checking libffi at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libffi.a
Checking pcre at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libpcre.dylib/.so
Checking zlib at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libz.dylib/.so
Checking iconv at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libiconv.dylib/.so
Checking glib at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libglib-2.0.dylib/.so
Checking googletest at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgtest.a
Checking gstreamer at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstreamer-1.0.dylib/.so
Checking gst-plugin-base at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstvideo-1.0.dylib/.so
Checking gst-plugin-good at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstavi.la
Checking gst-plugin-bad at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstbadvideo-1.0.dylib/.so
Checking for yasm
Checking gst-libav in /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/
Checking x264 at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libx264.dylib/.so
Checking gst-plugin-ugly at /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstx264.dylib/.so
-- System name is Darwin
-- Kinesis Video PIC path is ../kinesis-video-pic
-- Kinesis Video Producer path is ../kinesis-video-producer
-- Kinesis Video JNI path is ../kinesis-video-producer-jni
-- Kinesis Video Gstreamer demp app path is ../kinesis-video-gst-demo
-- Kinesis Video Open Source path is /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
[100%] Built target producer
[ 56%] Built target producer
[ 58%] Linking CXX executable start
[100%] Built target start
[ 97%] Built target producer
[ 97%] Linking CXX executable kinesis_video_gstreamer_sample_app
[100%] Built target kinesis_video_gstreamer_sample_app
[ 95%] Built target producer
[ 97%] Linking CXX executable kinesis_video_gstreamer_sample_rtsp_app
[100%] Built target kinesis_video_gstreamer_sample_rtsp_app
you may want to add the following environment variables
in /Users/rafaelviana/.bashrc or /Users/rafaelviana/.bash_profile or /Users/rafaelviana/.zshrc
**********************************************************
Success!!!
**********************************************************
```

And also installed kinesis-video-gstreamer-plugin script

```
-- KINESIS_VIDEO_OPEN_SOURCE_SRC: ..//kinesis-video-native-build/downloads
-- System name is Darwin
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rafaelviana/Documents/workspace-citisee/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin
[100%] Built target gstkvssink
[100%] Built target kvs_producer_plugin_demo
[100%] Built target kvs_producer_plugin_rtsp_demo
[100%] Built target gstkvsplugintest
you may want to add the following environment variables
in /Users/rafaelviana/.bashrc or /Users/rafaelviana/.bash_profile or /Users/rafaelviana/.zshrc
**********************************************************
Successful build of kinesis-video-gstreamer-plugin !!!
**********************************************************
```

However, the plugin is not installed.

```
gst-inspect-1.0 kvssink
Nenhum elemento ou plug-in "kvssink" (Translation: None element or plugin "kvssink")
```

What am i missing?

zhiyua-git, 2018-09-03:
Hi @RafaelRViana,

You need to set the environment variable before using kvssink as mentioned [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#setting-the-environment-variables-for-library-path).

Let me know if this works for you.

-ZH

RafaelRViana, 2018-09-03:
Thanks!

Sorry for my mistake.

closed 2018-09-03 by RafaelRViana