awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #116: when i test the camera with raspistill -o image.jpg,show error below

jackylee1, 2018-09-07:
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

but my camera is connected rightly.
what's the reason

MushMal, 2018-09-07:
@jackylee1 there are number of issues that can cause this starting from the non-properly seated cable/module to the kernel and firmware update. The error you are seeing is from the kernel mode driver being unable to start the camera. 
There are some instructions that you can follow in the following thread: https://www.raspberrypi.org/forums/viewtopic.php?t=39996

Please let us know if this helps.

zhiyua-git, 2018-09-07:
Hi @jackylee1 , besides what @MushMal  mentioned above, can you also confirm if you enabled you camera beforehand? (See **Enabling the camera** in https://www.raspberrypi.org/documentation/usage/camera/README.md)

unicornss, 2018-09-08:
Please check if the following troubleshooting instruction helps you to resolve the issue.

> https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#raspberry-pi-failure-to-load-the-camera-device

MushMal, 2018-09-12:
@jackylee1 were you able to make any progress with the camera?

jackylee1, 2018-09-13:
no.

unicornss, 2018-09-17:
As @MushMal  has mentioned, this issue seems to be in the Pi setup.
 Adding a [reference](https://raspberrypi.stackexchange.com/questions/37683/raspberry-pi-camera-rev-1-3-is-not-detected) to the right forum and closing this.

Please reopen or create a new issue if any Kinesis Video Streams related support is required.
Thanks

closed 2018-09-17 by unicornss

hetthummar, 2019-06-02:
There are two camera connecter in raspberry pi ..
make sure you connect camera to right connecter....

KevinBarnhill, 2019-12-12:
Check the little connector on the camera board between the camera and the board.  I popped that back in and everything worked.

chrishobcroft, 2020-02-18:
@jackylee1 you probably put the cables in the wrong way round.

Try turning them both in the slots.

shilan, 2020-08-26:
I have a similar problem. I installed fswebcam https://www.raspberrypi.org/documentation/usage/webcams/ and it just works fine. However when I try raspistill -o -v test.jpg I get
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

and I have no idea how to fix it!