awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #117: is the name of the kinesis video stream can be any name?what's wrong with my error

jackylee1, 2018-09-07:
./kinesis_video_gstreamer_sample_app myvideotest
[INFO ][2018-09-07 13:07:20] Initializing curl.
[INFO ][2018-09-07 13:07:20] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-09-07 13:07:20] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-09-07 13:07:20] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-09-07 13:07:20] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-09-07 13:07:20] Refreshing credentials. Force refreshing: 0 Now time is: 1536325640 Expiration: 0
[INFO ][2018-09-07 13:07:20] New credentials expiration is 1536328040
[DEBUG][2018-09-07 13:07:20] createDeviceHandler invoked
[INFO ][2018-09-07 13:07:20] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-09-07 13:07:20] clientReadyHandler invoked
[DEBUG][2018-09-07 13:07:20] Awaiting for the producer to become ready...
[DEBUG][2018-09-07 13:07:20] Kinesis Video producer is Ready.
[DEBUG][2018-09-07 13:07:20] Client is ready
[INFO ][2018-09-07 13:07:20] Creating Kinesis Video Stream myvideotest
[INFO ][2018-09-07 13:07:20] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-09-07 13:07:20] describeStreamHandler invoked
[INFO ][2018-09-07 13:07:20] Refreshing credentials. Force refreshing: 0 Now time is: 1536325640 Expiration: 0
[DEBUG][2018-09-07 13:07:20] Awaiting for the stream to become ready...
[ERROR][2018-09-07 13:07:25] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5515 milliseconds
[WARN ][2018-09-07 13:07:25] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20180907/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=0fce3254a9adc6491e84c6601be86e35980fde2eaade6c4e4939561d1d955e5d
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS-PRODUCER/1.5.0 GCC/6.3.0 Linux/4.14.68-v7+ armv7l
    X-Amz-Date: 20180907T130720Z
[DEBUG][2018-09-07 13:07:25] describeStream response: 
[INFO ][2018-09-07 13:07:25] Describe stream did not find the stream myvideotest in Kinesis Video (stream will be created)
[INFO ][2018-09-07 13:07:25] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-09-07 13:07:25] describeStreamHandler invoked
[INFO ][2018-09-07 13:07:25] Refreshing credentials. Force refreshing: 0 Now time is: 1536325645 Expiration: 0
[ERROR][2018-09-07 13:07:31] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5514 milliseconds
[WARN ][2018-09-07 13:07:31] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20180907/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=b89b03ed42e5d5b2543a938ebee53d597e23c5b1d25db51817dcaeaea7325dab
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS-PRODUCER/1.5.0 GCC/6.3.0 Linux/4.14.68-v7+ armv7l
    X-Amz-Date: 20180907T130725Z
[DEBUG][2018-09-07 13:07:31] describeStream response: 
[INFO ][2018-09-07 13:07:31] Describe stream did not find the stream myvideotest in Kinesis Video (stream will be created)
[INFO ][2018-09-07 13:07:31] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-09-07 13:07:31] describeStreamHandler invoked
[INFO ][2018-09-07 13:07:31] Refreshing credentials. Force refreshing: 0 Now time is: 1536325651 Expiration: 0
[ERROR][2018-09-07 13:07:37] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5514 milliseconds
[WARN ][2018-09-07 13:07:37] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20180907/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=3913f94bf1902953d6e0256634f981a4e15d0902fee17317370ef461ab4323aa
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS-PRODUCER/1.5.0 GCC/6.3.0 Linux/4.14.68-v7+ armv7l
    X-Amz-Date: 20180907T130731Z
[DEBUG][2018-09-07 13:07:37] describeStream response: 
[INFO ][2018-09-07 13:07:37] Describe stream did not find the stream myvideotest in Kinesis Video (stream will be created)
[INFO ][2018-09-07 13:07:37] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-09-07 13:07:37] describeStreamHandler invoked
[INFO ][2018-09-07 13:07:38] Refreshing credentials. Force refreshing: 0 Now time is: 1536325658 Expiration: 0
[ERROR][2018-09-07 13:07:43] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5515 milliseconds
[WARN ][2018-09-07 13:07:43] HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20180907/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=0b35494d5a00c1a80851efb9989a6257aa5013cbe0286409ee5822a5eebeb2af
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS-PRODUCER/1.5.0 GCC/6.3.0 Linux/4.14.68-v7+ armv7l
    X-Amz-Date: 20180907T130738Z
[DEBUG][2018-09-07 13:07:43] describeStream response: 
[INFO ][2018-09-07 13:07:43] Describe stream did not find the stream myvideotest in Kinesis Video (stream will be created)
[INFO ][2018-09-07 13:07:43] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-09-07 13:07:43] describeStreamHandler invoked
[INFO ][2018-09-07 13:07:45] Refreshing credentials. Force refreshing: 0 Now time is: 1536325665 Expiration: 0
[INFO ][2018-09-07 13:07:50] Freeing Kinesis Video Stream myvideotest
[INFO ][2018-09-07 13:07:50] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[ERROR][2018-09-07 13:07:50] Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.

MushMal, 2018-09-07:
@jackylee1 it seems that you have a connection problem as your calls timeout. Please ensure you have proper internet connection.

jackylee1, 2018-09-08:
@MushMal  have tested it with proper connection.the same error.
i wonder if it has anything to do with my other issue
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/116

MushMal, 2018-09-08:
They are quite unrelated. This particular issue is about the networking call timing out trying to reach out to the service, whereas the issue described in #116 is about the device not "seeing" the camera attached.

I would ask you to provide more information regarding both of the issues. In this case, in particular, please include the steps you've verified the connection. Perhaps, an interesting test would be to simply run a shell command on the RPi and try to curl to see if you get any response from the service.

jackylee1, 2018-09-08:
thanks!
i have followed steps here
https://docs.aws.amazon.com/zh_cn/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html#producersdk-cpp-rpi-camera
and the connection is ok.i don't quite get what do you mean

unicornss, 2018-09-08:
> [ERROR][2018-09-07 13:07:37] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5514 milliseconds

Could you please check if both network interfaces (wireless - wlan0,  wired-eth0)?

Sometimes, the wired (eth0) network interface may not have the route to connect to the internet ( and the demo sample uses the wired network by default ).  Ref:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/62

unicornss, 2018-10-04:
Closing the issue as it has been answered. Please reopen or create a new issue with details on the support required. Thanks.

closed 2018-10-04 by unicornss