awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #119: Install-Script is not downloading the sample GStreamer application.

cfsabc, 2018-09-10:
I have run the install-script multiple times and cannot get it to download the sample GStreamer application. 

This is on mac.

Thank you

MushMal, 2018-09-10:
Could you do 'ls -la' in the directory and see whether the applications were generated successfully?

It's likely that your install-script has failed. Could you please include the section of the log that contains the error.

cfsabc, 2018-09-10:
Sure, here are the the results:

[image]

After running the script multiple times, it usually ends with this:

[image]

I have double checked to make sure bison is installed.

Thanks you

cfsabc, 2018-09-10:
Shoot, the second photo is supposed to be:

[image]

MushMal, 2018-09-10:
Seems related to: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/100#issuecomment-405628765

Could you please configure the path

cfsabc, 2018-09-10:
Hello @MushMal,

I found the problem,  in my case brew wasn't creating the symlinks.

I followed the following steps as noted here: 

https://stackoverflow.com/a/29053701

Thank you!