awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #121: on mac,still can not found gstreamer app

jackylee1, 2018-09-12:
CMakeLists.txt			min-install-script
certs				msys2-install-script
downloads			projectFiles
gstreamer-plugin-install-script	samples
install-script			vs-buildtools-install.bat
install-script-ami		windows-install.bat
java-install-script		windows-java-install.bat
kvs_log_configuration
by the way ,how do i set the environment variables.for bison 2.4,the version is 2.3,why the path is PATH=/usr/local/Cellar/bison/3.0.4_1/bin/:$PATH?
and how to set pkgconfig?thanks.and when i append the cert by  running sudo cat SFSRootCAG2.pem >> /etc/ssl/cert.pem.it shows  error like permission denied ..
during the install ,it show no errors

unicornss, 2018-09-12:
Can you tell us if you have installed bison using `brew install` . Once you have installed the bison using brew then you can check the directory below to see the version.
`ls /usr/local/Cellar/bison` If the installed version of bison is 3.0.4_1 then you can export the PATH in the environment as below.

`export PATH=/usr/local/Cellar/bison/3.0.4_1/bin/:$PATH`

You may have to change the path according to your local bison installed version. Hope this helps.

Regarding the cert.pem install, you need administrator privileges to update cert.pem

unicornss, 2018-10-03:
Closing the issue as it has been answered. Please reopen or create a new issue with details on the support required. Thanks.

closed 2018-10-04 by unicornss