awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #122: the same error.why is that

jackylee1, 2018-09-12:
[INFO ][2018-09-12 15:38:28] Initializing curl.
[INFO ][2018-09-12 15:38:28] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-09-12 15:38:28] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-09-12 15:38:28] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-09-12 15:38:28] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-09-12 15:38:28] Refreshing credentials. Force refreshing: 0 Now time is: 1536766708 Expiration: 0
[INFO ][2018-09-12 15:38:28] New credentials expiration is 1536769108
[DEBUG][2018-09-12 15:38:28] createDeviceHandler invoked
[INFO ][2018-09-12 15:38:28] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-09-12 15:38:28] clientReadyHandler invoked
[DEBUG][2018-09-12 15:38:28] Awaiting for the producer to become ready...
[DEBUG][2018-09-12 15:38:28] Kinesis Video producer is Ready.
[DEBUG][2018-09-12 15:38:28] Client is ready
[INFO ][2018-09-12 15:38:28] Creating Kinesis Video Stream LiveRekognitionVideoAnalysisBlog
[INFO ][2018-09-12 15:38:28] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-09-12 15:38:28] describeStreamHandler invoked
[INFO ][2018-09-12 15:38:28] Refreshing credentials. Force refreshing: 0 Now time is: 1536766708 Expiration: 0
[DEBUG][2018-09-12 15:38:28] Awaiting for the stream to become ready...
[WARN ][2018-09-12 15:38:29] HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20180912/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=791861f5a3c169ebec382be848087573c6673adddd1c62427a37e3f70d843b86
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.5.0 Clang/9.1.0 Darwin/17.5.0 x86_64
    X-Amz-Date: 20180912T153828Z
[DEBUG][2018-09-12 15:38:29] describeStream response: {"message":"The security token included in the request is invalid."}
[INFO ][2018-09-12 15:38:29] Describe stream did not find the stream LiveRekognitionVideoAnalysisBlog in Kinesis Video (stream will be created)
[INFO ][2018-09-12 15:38:29] 
describeStreamResultEvent(): Describe stream result event.
[ERROR][2018-09-12 15:38:29] Submitting event result for stream: 140369908386926 failed with: 1375731728
[ERROR][2018-09-12 15:38:29] Reporting stream error. Errored timecode: 0 Status: 1375731728
[INFO ][2018-09-12 15:38:58] Freeing Kinesis Video Stream LiveRekognitionVideoAnalysisBlog
[INFO ][2018-09-12 15:38:58] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[ERROR][2018-09-12 15:38:58] Failed to create Kinesis Video Stream - timed out.
libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Failed to create Kinesis Video Stream - timed out.
Abort trap: 6

unicornss, 2018-09-12:
`[WARN ][2018-09-12 15:38:29] HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}`

Please make sure you are providing the ACCESS KEY and SECRET KEY as per the readme instructions.

jackylee1, 2018-09-13:
yes.

closed 2018-10-04 by unicornss