awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #123: How to add video option(eg. raspistill option -vf)

JeongTaekLim, 2018-09-13:
By this awesome project,
I've succeeded to send my raspberry pi's video into AWS kinesis successfully.
Really thanks to contributors!

And now I want to use some video option (eg. **-vf option** ) in raspistill command.
Unfortunately, there is no information about this specific use case in stackoverflow and other information site.

Can anybody give me some advice for how to use those option?

OS: Raspberry Raspbian
Camera: Raspberry pi V2
Install guide 1: [install-instructions-linux.md](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md)
Install guide 2: [Using the C++ Producer SDK on Raspberry Pi](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html)

unicornss, 2018-09-13:
Hi @JeongTaekLim 

Glad to hear that you were able to send video from Raspberry Pi to Kinesis Video Streams successfully.

Regarding the `raspistill -vf ` (vertical flip?), are you referring to sending image sequence as video to Kinesis Video Streams? 

Thanks

JeongTaekLim, 2018-09-14:
Hello @unicornss ! Thank you for your fast response!
First, sorry for insufficient description about my AWS usage state.
Yes, I'm using Kinesis Video Streams and I'm sending image sequence as video to Kinesis Video Stream.

For more details, I copied and pasted screenshot of my Kinesis Video Stream.

[image]

As you can see, our camera is physically installed contrary.

So I want to transform video into right angle by modifying or doing something in software level - rotate or vertically flip.

unicornss, 2018-09-14:
Hi @JeongTaekLim 

Please check the details for using Kinesis Video Producer SDK as a GStreamer element (kvssink) with gst-launch-1.0 [here]( https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#running-the-gst-launch-10-command-to-start-streaming-from-camera-source-in-raspberry-pi).

`$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=5120000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=YourStreamName frame-timestamp=dts-only access-key=YourAccessKey secret-key=YourSecretKey`

You can add video flip (`! videoflip method=clockwise ! `) within the pipeline as below:

`$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoflip method=clockwise ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=5120000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=YourStreamName frame-timestamp=dts-only access-key=YourAccessKey secret-key=YourSecretKey
`

chehefen, 2018-09-14:
Can you try apply this patch to the latest sdk repo and run the kinesis_video_gstreamer_sample_app? I added code to use the videoflip gstreamer element.
[flip_patch.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2384444/flip_patch.txt)

JeongTaekLim, 2018-09-17:
@unicornss Thank you for detail explanation!
I've succeeded with your solution!
It works very well thanks!

@chehefen 
Also thank you for providing code snippet for me, and I'll try it and give you feedback about it.

Btw, @unicornss 's answer works very well and I've also found that there are more options for video

Ps. Who want to user `! videoflip method=clockwise !`, please note that this command should be inserted into head of command line(before `! kvssink ~~ `)