awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #124: Windows docker container git not found

jonathanmesser, 2018-09-13:
When trying to set up the Windows docker test container, it fails to build the image due to git not being installed.  I managed to get past this by editing the docker file to use chocolatey to install git - changed the Docker file from:

`FROM microsoft/windowsservercore
WORKDIR /opt/`

to:

`FROM microsoft/windowsservercore
RUN @powershell iex ((new-object net.webclient).DownloadString('https://chocolatey.org/install.ps1'))   
RUN cinst -y git
WORKDIR /opt/`

after re-running the build with this change git is installed and it clones the repository successfully

MushMal, 2018-09-13:
@jonathanmesser thanks a lot for the report. We will look into it ASAP.

zhiyua-git, 2018-09-13:
The fix is in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/commit/0296a2c8988a74f618dbaa4feecd7a8f52b36804. Thanks for reporting!

closed 2018-09-13 by zhiyua-git

collinforrester, 2018-10-31:
@jonathanmesser were you able to get the windows docker container setup? any issues with log4cplus lib?

jonathanmesser, 2018-10-31:
> @jonathanmesser were you able to get the windows docker container setup? any issues with log4cplus lib?

I didn't have any issues with the setup after getting git installed via this change. I've not used this since then so unfortunately I'm not up to date