awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #125: Stream not starting on Mac air

sumopal23, 2018-09-13:
Hi, I'm trying to run the demo on my macbook air but the stream doesn't start after i enter the command:
./kinesis_video_gstreamer_sample_app LiveRekognitionVideoAnalysisBlog
It gives the following output:
./kinesis_video_gstreamer_sample_app LiveRekognitionVideoAnalysisBlog
[INFO ][2018-09-13 16:50:17] Initializing curl.
[INFO ][2018-09-13 16:50:17] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-09-13 16:50:17] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-09-13 16:50:17] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-09-13 16:50:17] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-09-13 16:50:17] Refreshing credentials. Force refreshing: 0 Now time is: 1536857417 Expiration: 0
[INFO ][2018-09-13 16:50:17] New credentials expiration is 1536859817
[DEBUG][2018-09-13 16:50:17] createDeviceHandler invoked
[INFO ][2018-09-13 16:50:17] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-09-13 16:50:17] clientReadyHandler invoked
[DEBUG][2018-09-13 16:50:17] Awaiting for the producer to become ready...
[DEBUG][2018-09-13 16:50:17] Kinesis Video producer is Ready.
[DEBUG][2018-09-13 16:50:17] Client is ready
[INFO ][2018-09-13 16:50:17] Creating Kinesis Video Stream LiveRekognitionVideoAnalysisBlog
[INFO ][2018-09-13 16:50:17] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-09-13 16:50:17] describeStreamHandler invoked
[DEBUG][2018-09-13 16:50:17] Awaiting for the stream to become ready...
[INFO ][2018-09-13 16:50:17] Refreshing credentials. Force refreshing: 0 Now time is: 1536857417 Expiration: 0
[DEBUG][2018-09-13 16:50:18] describeStream response: {"StreamInfo":{"CreationTime":1.536775979379E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:534145513179:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:534145513179:stream/LiveRekognitionVideoAnalysisBlog/1536775979379","StreamName":"LiveRekognitionVideoAnalysisBlog","Version":"E8V6WnqZc51hgRGQ8Aeh"}}
[INFO ][2018-09-13 16:50:18] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:534145513179:stream/LiveRekognitionVideoAnalysisBlog/1536775979379
[INFO ][2018-09-13 16:50:18] stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:534145513179:stream/LiveRekognitionVideoAnalysisBlog/1536775979379
[INFO ][2018-09-13 16:50:18] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-09-13 16:50:18] streamingEndpointHandler invoked
[INFO ][2018-09-13 16:50:18] Refreshing credentials. Force refreshing: 0 Now time is: 1536857418 Expiration: 0
[DEBUG][2018-09-13 16:50:18] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2018-09-13 16:50:18] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2018-09-13 16:50:18] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2018-09-13 16:50:18] streamingTokenHandler invoked
[INFO ][2018-09-13 16:50:18] Refreshing credentials. Force refreshing: 1 Now time is: 1536857418 Expiration: 1536859817
[INFO ][2018-09-13 16:50:18] New credentials expiration is 1536859818
[INFO ][2018-09-13 16:50:18] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2018-09-13 16:50:18] streamReadyHandler invoked
[DEBUG][2018-09-13 16:50:18] Kinesis Video stream LiveRekognitionVideoAnalysisBlog is Ready.
[DEBUG][2018-09-13 16:50:18] Stream is ready
Default list of resolutions (1920x1080, 1280x720, 640x480) are not supported by video source
[INFO ][2018-09-13 16:50:19] Freeing Kinesis Video Stream LiveRekognitionVideoAnalysisBlog
[INFO ][2018-09-13 16:50:19] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[INFO ][2018-09-13 16:50:19] Freeing Kinesis Video Stream LiveRekognitionVideoAnalysisBlog
[INFO ][2018-09-13 16:50:19] 
freeKinesisVideoClient(): Freeing Kinesis Video Client
[INFO ][2018-09-13 16:50:19] 
commonHeapDebugCheckAllocator(): Heap is initialized
[INFO ][2018-09-13 16:50:19] 
commonHeapDebugCheckAllocator(): Heap limit: 				536870912
[INFO ][2018-09-13 16:50:19] 
commonHeapDebugCheckAllocator(): Heap size: 				0
[INFO ][2018-09-13 16:50:19] 
commonHeapDebugCheckAllocator(): Number of allocations: 				0
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				0x0
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x10e6b4000
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): Block:	0x10e6b4000		size:	536870872
[INFO ][2018-09-13 16:50:19] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-13 16:50:19] 
heapRelease(): Freeing native heap.

How to fix this?

MushMal, 2018-09-13:
@sumopal23 the root cause for the issue is the failure to negotiate the correct resolution vs frame rate with the supported encoder.

`<Default list of resolutions (1920x1080, 1280x720, 640x480) are not supported by video source>`

One of the solutions would be to lower the frame rate by specifying `<-f 20>` which will try to create the media pipeline with 20 fps instead of the default 30.

MushMal, 2018-09-13:
Linking an already existing issue https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/110

sumopal23, 2018-09-13:
Hi @MushMal,
Thanks for the response it worked perfectly now!

closed 2018-09-13 by sumopal23

MushMal, 2018-09-13:
Glad to hear it. We will enhance the sample to be able to handle this case