awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #127: Can't start kvs stream. Codec private data was not found in GST_CAPS event

jonathanmesser, 2018-09-14:
I've built the C++ producer on Windows and can get a video stream into Kinesis successfully. Now I'm trying to add a mux layer for audio on top of the video, and am getting this error. I can produce the file locally to file system and it contains audio and video, the error comes when trying to send out to the kvssink. I've tried using matroskamux and mpegtsmux both which give the same error. This works:

`gst-launch-1.0 -v ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! h264parse ! queue ! mpegtsmux0. autoaudiosrc ! queue ! avenc_aac ! aacparse ! mpegtsmux ! filesink location=output.mp4`

whereas this doesn't:
`gst-launch-1.0 -v ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! h264parse ! queue ! mpegtsmux0. autoaudiosrc ! queue ! avenc_aac ! aacparse ! mpegtsmux ! kvssink stream-name="test-stream" access-key=****** secret-key=******`

[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2384602/log.txt)

MushMal, 2018-09-17:
@jonathanmesser, thanks for the inquiry. Currently, we only support elementary stream for h264 out-of-the-box in the GStreamer plugin. We are working on making a generic support for multi-pad which will handle audio as well. 

In this case, the plugin gets confused as it can't extract and process Codec Private Data which in case of H264 is SPS and PPS.

jonathanmesser, 2018-09-18:
Thanks @MushMal. What would be the recommended route to get a stream which either includes audio, or perhaps audio only?

MushMal, 2018-09-18:
@jonathanmesser thanks. While we are working on to provide out-of-the-box support for multi-stream/multi-track solution for the GStreamer plugin, in the interim, you could use an example of the sample application where you could construct two separate GStreamer pipelines (audio and video) and in the app sinks create two separate KVS streams. Send the audio and video in their respective streams - similar to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_multistream_app.cpp

We hope to provide multi-stream solution very soon.

closed 2018-11-15 by unicornss