awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #128: product h264 data,video preview error,"Missing codec private data error"

xli022, 2018-09-17:
I use kinesis-video-producer c++ sdk, produce h264 es data, produce success, but  I donot display stream.
procude success print info:
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7

video preview error info:
Missing codec private data error
The fragment did not contain any codec private data. Ensure that the producer is generating valid H.264 codec private data.
RetryHide details

Fragment number: 91343852333181482019222357486862988630708668275

Producer timestamp: January 1, 1970 8:02 AM CST

Server timestamp: September 17, 2018 1:59 PM CST

Size: 206.2 KB

please help me,thank you!

MushMal, 2018-09-17:
@xli022 could you please include a little more information about the producer side. Are you using the sample application to stream video? If so, could I ask you to include the initial part of the log to make sure we have valid CPD.

Could you also include the account id, the region, the stream name and any other useful information we can help to debug.

xli022, 2018-09-18:
I use kinesis-video-producer/tst/ProducerApiTest.cpp sample to stream video.
I send h264 es data to stream video. 
account id is 274408052@qq.com,region: us-east-1,streamname: examplestream4。 
log info below:
INFO - Initializing curl.
2018-09-18 09:38:16 [140277897246528] INFO - 
createKinesisVideoClient(): Creating Kinesis Video Client
2018-09-18 09:38:16 [140277897246528] INFO - 
heapInitialize(): Initializing native heap with limit size 1073741824, spill ratio 0% and flags 0x00000001
2018-09-18 09:38:16 [140277897246528] INFO - 
heapInitialize(): Creating AIV heap.
2018-09-18 09:38:16 [140277897246528] INFO - 
heapInitialize(): Heap is initialized OK
2018-09-18 09:38:16 [140277897246528] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1537234696 Expiration: 0
2018-09-18 09:38:16 [140277897246528] INFO - New credentials expiration is 1537234736
2018-09-18 09:38:16 [140277897246528] DEBUG - createDeviceHandler invoked
2018-09-18 09:38:16 [140277897246528] INFO - 
createDeviceResultEvent(): Create device result event.
2018-09-18 09:38:16 [140277897246528] DEBUG - clientReadyHandler invoked
2018-09-18 09:38:16 [140277897246528] DEBUG - Awaiting for the producer to become ready...
2018-09-18 09:38:16 [140277897246528] DEBUG - Kinesis Video producer is Ready.
2018-09-18 09:38:16 [140277897246528] INFO - Creating Kinesis Video Stream examplestream4
2018-09-18 09:38:16 [140277897246528] INFO - 
createKinesisVideoStream(): Creating Kinesis Video Stream.
2018-09-18 09:38:16 [140277897246528] DEBUG - describeStreamHandler invoked
2018-09-18 09:38:16 [140277897246528] DEBUG - Awaiting for the stream to become ready...
2018-09-18 09:38:16 [140277897246528] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1537234696 Expiration: 0
2018-09-18 09:38:16 [140276709902080] DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.537149035272E9,"DataRetentionInHours":1,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-1:012274199671:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:012274199671:stream/examplestream4/1537149035272","StreamName":"examplestream4","Version":"fmDJKG2amvKBmgBNzCeO"}}
2018-09-18 09:38:18 [140276709902080] INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-east-1:012274199671:stream/examplestream4/1537149035272
2018-09-18 09:38:18 [140276709902080] INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-east-1:012274199671:stream/examplestream4/1537149035272
2018-09-18 09:38:18 [140276709902080] INFO - 
describeStreamResultEvent(): Describe stream result event.
2018-09-18 09:38:18 [140276709902080] DEBUG - tagResourceHandler invoked for stream: arn:aws:kinesisvideo:us-east-1:012274199671:stream/examplestream4/1537149035272
2018-09-18 09:38:18 [140276709902080] DEBUG - tagResourceHandler post body: {"StreamARN":"arn:aws:kinesisvideo:us-east-1:012274199671:stream/examplestream4/1537149035272","Tags":{"testTag_0_0":"testTag_0_0_Value","testTag_0_1":"testTag_0_1_Value","testTag_0_2":"testTag_0_2_Value","testTag_0_3":"testTag_0_3_Value","testTag_0_4":"testTag_0_4_Value"}}

2018-09-18 09:38:18 [140276709902080] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1537234698 Expiration: 0
2018-09-18 09:38:18 [140276701509376] INFO - 
tagResourceResultEvent(): Tag resource result event.
2018-09-18 09:38:20 [140276701509376] DEBUG - streamingEndpointHandler invoked
2018-09-18 09:38:20 [140276701509376] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1537234700 Expiration: 0
2018-09-18 09:38:20 [140276709902080] DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
2018-09-18 09:38:21 [140276709902080] INFO - streaming to endpoint: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com
2018-09-18 09:38:21 [140276709902080] INFO - 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2018-09-18 09:38:21 [140276709902080] DEBUG - streamingTokenHandler invoked
2018-09-18 09:38:21 [140276709902080] INFO - Refreshing credentials. Force refreshing: 1 Now time is: 1537234701 Expiration: 1537234736
2018-09-18 09:38:21 [140276709902080] INFO - New credentials expiration is 1537234741
2018-09-18 09:38:21 [140276709902080] INFO - 
getStreamingTokenResultEvent(): Get streaming token result event.
2018-09-18 09:38:21 [140276709902080] DEBUG - streamReadyHandler invoked
2018-09-18 09:38:21 [140276709902080] DEBUG - Kinesis Video stream examplestream4 is Ready.
DEBUG - putStreamHandler invoked
2018-09-18 09:38:22 [140277897246528] DEBUG - Created a new PutMedia request with stream upload handle: 0
2018-09-18 09:38:22 [140277897246528] INFO - 
putStreamResultEvent(): Put stream result event.
2018-09-18 09:38:22 [140277897246528] INFO - 
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 000000000112145e.
2018-09-18 09:38:22 [140277897246528] INFO - 
getKinesisVideoMetrics(): Get the memory metrics size.
2018-09-18 09:38:22 [140277897246528] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 1073741824
	>> Available storage byte size: 1073740531
	>> Allocated storage byte size: 1293
	>> Total view allocation byte size: 120072
	>> Total streams frame rate (fps): 25
	>> Total streams transfer rate (bps): 33554432 (32768 Kbps)
	>> Current view duration (ms): 40
	>> Overall view duration (ms): 40
	>> Current view byte size: 1168
[... cut ...]
2018-09-18 09:38:25 [140276579428096] =======================putFrame : 15372347055431325,4113--1
DEBUG - Wrote 4126 bytes to Kinesis Video. Upload stream handle: 0
2018-09-18 09:38:25 [140276579428096] =======================putFrame : 15372347056112046,5677--1
DEBUG - Wrote 5690 bytes to Kinesis Video. Upload stream handle: 0
2018-09-18 09:38:25 [140276579428096] INFO - Curl post body write function for stream: examplestream4 and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":1537234702478,"FragmentNumber":"91343852333181764487428761467811574485542693876"}

MushMal, 2018-09-18:
@xli022 the unit tests were designed as a test bed for streaming "synthetic" data. As such, those really don't provide CPD nor handle the callbacks properly. You might want to consider using KVS GStreamer plugin or modify the sample and add the callback handling just like in the KVS plugin.

This said, you need to extract the CPD from your media pipeline. Many media pipelines (for example Android MediaCodec) provide a secondary channel to get the CPD. Others, pre-pend the Annex-B format SPS/PPS before an I-frame. I would really need more information about your media pipeline or the elementary stream (whether it's in Annex-B and whether it has the CPD). In case your ES has the CPD pre-pended to the I-frames, you could extract the CPD (two NALu units from |0001|SPS|0001|PPS|0001I-frame| format) and submit those in KinesisVideoStream::start(cpd) API.

One more note: You seem to fail to put the first frame with status code 0x30000005 which indicates that the frame you are submitting has a timestamp less than the previous frame timestamp + the previous frame duration. Some media pipelines can give invalid timestamps or have a large jitter (frames are "clumped" together). In the case of jitter, please specify a smaller frame duration - something like 5 milliseconds for every frame duration.

Please let us know how this goes.

xli022, 2018-09-19:
I missed KinesisVideoStream::start(cpd) API.
Thank you very much.

closed 2018-09-19 by xli022