awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #130: unable to get RTSP streaming with error "Source link FAILED"

sesame437, 2018-09-18:
I have deploy this sdk on both my MAC, I could run ./kinesis_video_gstreamer_sample_app on my MAC, so I supposed this SDK installed successfully. And I have one camera and RTSP stream is worked with VLC on my mac. But it failed with ：AK SK ././kinesis_video_gstreamer_sample_rtsp_app rtsp stream name. 
The error message is "Source link FAILED". 
Please help me. Thanks.
+++++++
New RTSP source found: recv_rtp_src_0_647676507_96
Source link FAILED
New RTSP source found: recv_rtp_src_1_984905903_104
Source link FAILED
++++++

whole log message i list below
++++++++++++++
[INFO ][2018-09-18 02:43:30] Initializing curl.
[INFO ][2018-09-18 02:43:30]
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-09-18 02:43:30]
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-09-18 02:43:30]
heapInitialize(): Creating AIV heap.
[INFO ][2018-09-18 02:43:30]
heapInitialize(): Heap is initialized OK
[INFO ][2018-09-18 02:43:30] Refreshing credentials. Force refreshing: 0 Now time is: 1537238610 Expiration: 0
[INFO ][2018-09-18 02:43:30] New credentials expiration is 1537241010
[DEBUG][2018-09-18 02:43:30] createDeviceHandler invoked
[INFO ][2018-09-18 02:43:30]
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-09-18 02:43:30] clientReadyHandler invoked
[DEBUG][2018-09-18 02:43:30] Awaiting for the producer to become ready...
[DEBUG][2018-09-18 02:43:30] Kinesis Video producer is Ready.
[DEBUG][2018-09-18 02:43:30] Client is ready
[INFO ][2018-09-18 02:43:30] Creating Kinesis Video Stream xxx-stream
[INFO ][2018-09-18 02:43:30]
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-09-18 02:43:30] describeStreamHandler invoked
[INFO ][2018-09-18 02:43:30] Refreshing credentials. Force refreshing: 0 Now time is: 1537238610 Expiration: 0
[DEBUG][2018-09-18 02:43:30] Awaiting for the stream to become ready...
[DEBUG][2018-09-18 02:43:31] describeStream response: {"StreamInfo":{"CreationTime":1.515844320397E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:xxxxxxxxx:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:xxxxxxxxx:stream/xxx-stream/1515844320397","StreamName":"xxx-stream","Version":"EfAxWdDG2B66As0MEm1m"}}
[INFO ][2018-09-18 02:43:31] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:xxxxxxxxx:stream/xxx-stream/1515844320397
[INFO ][2018-09-18 02:43:31] stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:xxxxxxxxx:stream/xxx-stream/1515844320397
[INFO ][2018-09-18 02:43:31]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-09-18 02:43:31] streamingEndpointHandler invoked
[INFO ][2018-09-18 02:43:31] Refreshing credentials. Force refreshing: 0 Now time is: 1537238611 Expiration: 0
[DEBUG][2018-09-18 02:43:32] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2018-09-18 02:43:32] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2018-09-18 02:43:32]
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2018-09-18 02:43:32] streamingTokenHandler invoked
[INFO ][2018-09-18 02:43:32] Refreshing credentials. Force refreshing: 1 Now time is: 1537238612 Expiration: 1537241010
[INFO ][2018-09-18 02:43:32] New credentials expiration is 1537241012
[INFO ][2018-09-18 02:43:32]
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2018-09-18 02:43:32] streamReadyHandler invoked
[DEBUG][2018-09-18 02:43:32] Kinesis Video stream xxx-stream is Ready.
[DEBUG][2018-09-18 02:43:32] Stream is ready
[DEBUG][2018-09-18 02:43:32] Pipeline OK
[DEBUG][2018-09-18 02:43:32] rtsp://admin:password@192.168.xx.xx:554/main_stream
[DEBUG][2018-09-18 02:43:32] appsink configured
[DEBUG][2018-09-18 02:43:32] Pipeline built, elements *NOT* yet linked
[DEBUG][2018-09-18 02:43:32] Pipeline playing
New RTSP source found: recv_rtp_src_0_647676507_96
Source link FAILED
New RTSP source found: recv_rtp_src_1_984905903_104
Source link FAILED
Error received from element udpsrc3: Internal data stream error.
Debugging information: gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source/GstUDPSrc:udpsrc3:
streaming stopped, reason not-linked (-1)
[INFO ][2018-09-18 02:43:33] Freeing Kinesis Video Stream xxx-stream
[INFO ][2018-09-18 02:43:33]
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[INFO ][2018-09-18 02:43:33] Freeing Kinesis Video Stream xxx-stream
[INFO ][2018-09-18 02:43:33]
freeKinesisVideoClient(): Freeing Kinesis Video Client
[INFO ][2018-09-18 02:43:33]
commonHeapDebugCheckAllocator(): Heap is initialized
[INFO ][2018-09-18 02:43:33]
commonHeapDebugCheckAllocator(): Heap limit: 				536870912
[INFO ][2018-09-18 02:43:33]
commonHeapDebugCheckAllocator(): Heap size: 				0
[INFO ][2018-09-18 02:43:33]
commonHeapDebugCheckAllocator(): Number of allocations: 				0
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				(nil)
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x5621c008
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): Block:	0x5621c008		size:	536870887
[INFO ][2018-09-18 02:43:33]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-09-18 02:43:33]
heapRelease(): Freeing native heap.

unicornss, 2018-09-18:
Based on the error in the logs:
> Source link FAILED
> Error received from element udpsrc3: Internal data stream error.
> Debugging information: gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source/GstUDPSrc **:udpsrc3:**

Can you please add the following attribute : 

`
  "protocols", GST_RTSP_LOWER_TRANS_UDP,
`
after the line
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp#L412 .

Example:
> g_object_set(G_OBJECT (data.source),
>                  "location", rtsp_url,
>                  "short-header", true, // Necessary for target camera
> NULL);

needs to be updated as

>  g_object_set(G_OBJECT (data.source),
>                  "location", rtsp_url,
>                  "short-header", true, // Necessary for target camera
>                  "protocols", GST_RTSP_LOWER_TRANS_UDP,
> NULL);

Please run install-script.sh again after this change (from the `kinesis-video-native-build` directory); this time it will be quick as it only needs to compile this change. Let us know if this helps.

sesame437, 2018-09-18:
I refer to your suggestion but seem build failed with following error message. And I did a quick test and same error message.
++++++++++++
Scanning dependencies of target kinesis_video_gstreamer_sample_rtsp_app
[ 63%] Built target kinesis_video_gstreamer_sample_app
[ 64%] Building CXX object CMakeFiles/kinesis_video_gstreamer_sample_rtsp_app.dir/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp.o
[100%] Built target start
/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp:373:48: warning:
      format string is not a string literal (potentially insecure) [-Wformat-security]
    SNPRINTF(stream_name, MAX_STREAM_NAME_LEN, argv[2]);
                                               ^~~~~~~
/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp:373:48: note:
      treat the string as an argument to avoid this
    SNPRINTF(stream_name, MAX_STREAM_NAME_LEN, argv[2]);
                                               ^
                                               "%s",
/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp:377:45: warning:
      format string is not a string literal (potentially insecure) [-Wformat-security]
    SNPRINTF(rtsp_url, MAX_STREAM_NAME_LEN, argv[1]);
                                            ^~~~~~~
/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp:377:45: note:
      treat the string as an argument to avoid this
    SNPRINTF(rtsp_url, MAX_STREAM_NAME_LEN, argv[1]);
                                            ^
                                            "%s",
/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp:413:31: error:
      use of undeclared identifier 'GST_RTSP_LOWER_TRANS_UDP'
                 "protocols", GST_RTSP_LOWER_TRANS_UDP,
                              ^
2 warnings and 1 error generated.
make[2]: *** [CMakeFiles/kinesis_video_gstreamer_sample_rtsp_app.dir/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp.o] Error 1
make[1]: *** [CMakeFiles/kinesis_video_gstreamer_sample_rtsp_app.dir/all] Error 2
make: *** [all] Error 2

unicornss, 2018-09-18:
could you please add the required header file #include <gst/rtsp/gstrtsptransport.h>

Also, can you send us the details of camera model?

Thanks

sesame437, 2018-09-18:
#include <gst/rtsp/gstrtsptransport.h>
sorry, i don't understand this part. could you please give me information for it? 
The camera brand is EZVIZ, model is C4WI.

unicornss, 2018-09-18:
In the kinesis_video_gstreamer_sample_rtsp_app.cpp sample after the line you can add this header file include.  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp#L3 

Example:
Revise:

> #include <gst/gst.h>
#include <gst/app/gstappsink.h>
#include <string.h>
#include <chrono>
#include <Logger.h>
#include "KinesisVideoProducer.h"

to

> #include <gst/gst.h>
> #include <gst/app/gstappsink.h>
> #include <gst/rtsp/gstrtsptransport.h>
> #include <string.h>
> #include <chrono>
> #include <Logger.h>
> #include "KinesisVideoProducer.h"

Let us know if that helps with this camera.

unicornss, 2018-10-04:
Closing the issue as it has been answered. Please reopen or create a new issue with details on the support required. Thanks.