awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #131: Failed to create Kinesis Video stream - Error status: 0x5200004b

ameyat2812, 2018-09-27:
Hi @MushMal

I am using Kinesis Producer SDK version 1.4.0 and having following setup.

I have one instance of KinesisVideoProducer and have couple of KinesisVideoStreams attached to it for different purposes. The streams are created and destroyed as and when there are needed and it could happen that I might call following APIs on stream in quick succession,

1. CreateSync
2. Start
3. PutFrame for some time
4. Stop
5. wait for streamClosed msg with a timeout of couple of seconds
6. free
7. CreateSync (free and create could happen in quick succession and that is when this issue happens) 

There seems to be some state cleanup corner case in SDK in the above case, I am getting following errors
`Submitting event result for stream: 140308479699950 failed with: 1375731766 
Failed to create Kinesis Video Stream - timed out. 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Unable to create Kinesis Video stream. e63d93c9-dfba-41b0-9cb7-669a5b21ca9e_playback Error status: 0x5200004b 
Failed to create Kinesis Video stream 
Submitting event result for stream: 140309082144718 failed with: 1375731766 
Failed to create Kinesis Video Stream - timed out. 
Unable to create Kinesis Video stream. 6266161b-a342-4229-a3c6-f922cb9e7fad_playback Error status: 0x5200004b `

So I suspect some resource is either still cleaning up or has failed to be cleaned up and I am calling CreateSync on with the same stream name. This request times out and then every subsequent request to create that stream afterwords results in duplicate stream name error.

This is very random so I can't really reproduce this issue but the fact that every subsequent request to create the stream is failing makes the stream context unusable.

MushMal, 2018-09-27:
@ameyat2812 seems like you are hitting an edge case that was fixed in later releases.

I assume you are calling freeStream() from the Producer object. 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/KinesisVideoProducer.cpp#L215

Could you please upgrade your SDK to the latest version? Alternatively, you could try adding the line above.

Also, you could use the sync version of the KinesisVideoStream::stopSync()

Please let us know if this works.

ameyat2812, 2018-10-02:
Hi @MushMal , I have bumped the version to 1.5.0,  will keep monitoring if this issue shows up. Thanks for prompt reply.

MushMal, 2018-10-02:
Very cool to hear that. I am going to resolve the issue but please don't hesitate to re-open if you hit any issues. We do have a set of unit tests that allow for creation and termination of the streams.

closed 2018-10-02 by MushMal