awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #133: Problem with install on NVIDIA TX2 dev platform

duaneharkness, 2018-09-29:
I encountered two errors using install-script on an NVIDIA TX2 development platform. Both of these are related to its aarch64 architecture. I was able to work around both problems but thought I'd raise them as issues.

The first problem is the following compilation error when building the openssl dependency: 

```
engines/afalg/e_afalg.c: In function 'eventfd':
engines/afalg/e_afalg.c:110:20: error: '__NR_eventfd' undeclared (first use in this function)
     return syscall(__NR_eventfd, n);
                    ^
engines/afalg/e_afalg.c:110:20: note: each undeclared identifier is reported only once for each function it appears in
engines/afalg/e_afalg.c:111:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[1]: *** [engines/afalg/e_afalg.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

This appears to be a known problem for older versions of openssl when compiling on aarch64:

https://github.com/zorun/openssl/commit/6e2471a97b65cd76e2329572a62cd6fffe550c66

I was able to fix this by editing install-script and changing "openssl-1.1.0f" to "openssl-1.1.0g".

The second problem is with the m4 dependency which uses a very old "config.guess" file that doesn't recognize the aarch64 architecture.

I was able to fix this by editing install-script and changing "m4-1.4.10" to "m4-1.4.18".

MushMal, 2018-10-01:
@duaneharkness thanks for reporting the issue. We will look into this issue ASAP. Likely, we will update to the later versions of the open source components as you've suggested. We also do appreciate pull requests :).

unicornss, 2018-12-21:
Closing the issue for now.

closed 2018-12-21 by unicornss