awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #134: How much is the minimum live stream delay?

xli022, 2018-09-29:
How to reduce delay？

zhiyua-git, 2018-10-01:
Hi @xli022,

It depends on your use case. 

If you are higher latency, you can try our [HLS](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-hls.html) solution which will be much easier to use on different kinds of browsers and platforms.

If you want to get latency for less than 1 second, you can refer to [examples](https://github.com/aws/amazon-kinesis-video-streams-parser-library#example) in our parser library, KinesisVideoRendererExample is the one for playback. Note this is currently only available as java (maven) code/library, if you need it in other languages or platform, you may need to migrate the code.

Also can you share your use case so we can help you better? Thanks.

Hope this answers your question.
-ZH

unicornss, 2018-10-04:
Closing the issue as it has been answered. Please reopen or create a new issue with details on the support required. Thanks.

closed 2018-10-04 by unicornss