awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #135: Suspect thread deadlock on STATUS_INVALID_PARSED_ACK_TYPE error in PutMedia

ameyat2812, 2018-10-04:
Hi @MushMal,

I faced a very peculiar problem the other day where I suspect putFrame API from KinesysVideoStream was blocked forever. This happened on one of our deploys where SDK version 1.4.0 was running.

I could see following error thrown by the SDK 

ERROR[01/10/18 23:28:56.048][com.amazonaws.kinesis.video][OngoingStreamState.cpp:209] Failed to submit ACK: {"Message":"User: USER is not authorized to perform: kinesisvideo:PutMedia on resource: RESOURCE_NAME/1537842499455"} with status code: 1375731793

And I reckon the putFrame API never returned. Could it be possible to have some deadlock happening while acquiring mutexes in the putFrame() when this particular error occurs??

zhiyua-git, 2018-10-04:
Hi @ameyat2812 ,

The message ("User: USER is not authorized to perform: kinesisvideo:PutMedia on resource: RESOURCE_NAME/1537842499455) indicates you don't have PutMedia permission on your account/user, can you double check if you have setting proper permission there? You can refer to [this document](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/gs-account.html). 

You can use AmazonKinesisVideoStreamsFullAccess for full access. You can use AmazonKinesisVideoStreamsReadOnlyAccess for read-only access.

Hope this helps.
-ZH

ameyat2812, 2018-10-04:
hi @zhiyua-git. 

Thanks for your prompt reply. 

I don't think I was lacking permissions as I was uploading to the same RESOURCE for more than a day. I could also confirm that the credentials that I was using were also valid. So not really sure what caused this error.

Having said that, could this error cause putFrame() API to get into a deadlock?

zhiyua-git, 2018-10-04:
Hi @ameyat2812,

Which client are you using, GStreamer plugin or C++ producer sample? If it is C++ producer then probably you got that failure and sdk start retrying, after 5 retries with failure, it stops. You can modify [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L300) to a large number like 1000000, then it will retry for more times, but if PutMedia is consistently failing for permission issue, even unlimited retry would not help. 

Do you mind sending your full log file (at least the part you still had PERSISTED Acks and the failure) to kinesis-video-support@amazon.com so we can better analyze what was happening? Can you also share some information about how you are using kinesis video, we might point you to some better way for your use case?

Thanks,
-ZH

MushMal, 2018-10-04:
@ameyat2812 just to add more info

* The authentication errors are non-retriable errors as it makes no sense to retry.
* I am not sure where {"Message":"User: USER is not authorized to perform: kinesisvideo:PutMedia on resource: RESOURCE_NAME/1537842499455"} comes from. It is an unknown ACK from the backend hence the parsing fails. 
* Any callbacks from the PIC should be "prompt". This is the only reason putFrame or other operations can block.

Can you elaborate what your observation is? Why do you think the putFrame blocks? As @zhiyua-git  suggests could you please email the relevant logs to the support email?

MushMal, 2018-10-05:
What you are trying to achieve in general is a very supported scenario - make sure you update the SDK version if not done so. You just need to ensure you update the credentials in the CredentialProvider whenever the refresh is called but seems that you are doing it already but please make sure you set the correct expiration like in the samples.

The SDK doesn't have it's own threads - it utilizes the threads from API calls to events to drive the machinery. When you make the first call to putFrame, the thread that makes it (which is the GStreamer thread) will be used for the PIC callback of PutStream, which, will power CURL to issue a series of service calls like DescribeStream, GetMediaEndpoint, etc... and eventually the PutMedia. As soon as the first call is scheduled, the thread returns (the GStreamer thread that was used) and the state machinery will be driven by the networking thread that returns the call result.

If you are using C++ SDK without modification then it shouldn't lock. However, the putFrame could return an error status. Make sure you check for it. Also, make sure you handle the onStreamError properly.

I am having hard time understanding/re-creating the scenario as you have multiple issues. I would love if you could save the logs so we can clearly see what's going on. You could also enable GStreamer debug logs specifying `export GST_DEBUG=4`

ameyat2812, 2018-10-05:
I have bumped the version as you said just that it is not done on our prod environment just yet. Will do that.

Just to answer your questions, I am using C++ SDK untouched, and have ERROR level logs at all API failures as well as onStreamError().

I will keep monitoring this issue and will also try and reproduce by playing with resource permissions, with lower log levels to see if it happens again.

Thanks for the directions though... 👍

unicornss, 2018-10-12:
Closing the issue as it has been answered. 
Please re-open if additional support is required. Thanks

closed 2018-10-15 by unicornss