awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #136: Repeatedly getting Failed to submit frame to Kinesis Video client. status: 0x32000002 after following RPi tutorial

frenchie4111, 2018-10-10:
Continuously getting error:

```
Failed to submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp: 4879634 presentation timestamp: 4540057
Dropped frame!
```

On:
Model: Raspberry Pi 3

```
$ uname -a
Linux l53-pi-1 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux
```

Followed instructions here:

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html

Running command `./kinesis_video_gstreamer_sample_app l53-pi-1`

Snippet of logs:
```
$ ./kinesis_video_gstreamer_sample_app l53-pi-1
[INFO ][2018-10-10 21:14:14] Initializing curl.
[INFO ][2018-10-10 21:14:14] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-10-10 21:14:14] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-10-10 21:14:14] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-10-10 21:14:14] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-10-10 21:14:14] Refreshing credentials. Force refreshing: 0 Now time is: 1539206054 Expiration: 0
[INFO ][2018-10-10 21:14:14] New credentials expiration is 1539208454
[DEBUG][2018-10-10 21:14:14] createDeviceHandler invoked
[INFO ][2018-10-10 21:14:14] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-10-10 21:14:14] clientReadyHandler invoked
[DEBUG][2018-10-10 21:14:14] Awaiting for the producer to become ready...
[DEBUG][2018-10-10 21:14:14] Kinesis Video producer is Ready.
[DEBUG][2018-10-10 21:14:14] Client is ready
[INFO ][2018-10-10 21:14:14] Creating Kinesis Video Stream l53-pi-1
[INFO ][2018-10-10 21:14:14] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-10-10 21:14:14] describeStreamHandler invoked
[INFO ][2018-10-10 21:14:14] Refreshing credentials. Force refreshing: 0 Now time is: 1539206054 Expiration: 0
[DEBUG][2018-10-10 21:14:14] Awaiting for the stream to become ready...
[DEBUG][2018-10-10 21:14:14] describeStream response: {"StreamInfo":{"CreationTime":1.539204514969E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:340301842845:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:340301842845:stream/l53-pi-1/1539204514969","StreamName":"l53-pi-1","Version":"J0O4OEOat5NtGSMISrNT"}}
[INFO ][2018-10-10 21:14:14] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:340301842845:stream/l53-pi-1/1539204514969
[INFO ][2018-10-10 21:14:14] stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:340301842845:stream/l53-pi-1/1539204514969
[INFO ][2018-10-10 21:14:14] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-10-10 21:14:14] streamingEndpointHandler invoked
[INFO ][2018-10-10 21:14:14] Refreshing credentials. Force refreshing: 0 Now time is: 1539206054 Expiration: 0
[DEBUG][2018-10-10 21:14:14] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2018-10-10 21:14:14] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2018-10-10 21:14:14] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2018-10-10 21:14:14] streamingTokenHandler invoked
[INFO ][2018-10-10 21:14:14] Refreshing credentials. Force refreshing: 1 Now time is: 1539206054 Expiration: 1539208454
[INFO ][2018-10-10 21:14:14] New credentials expiration is 1539208454
[INFO ][2018-10-10 21:14:14] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2018-10-10 21:14:14] streamReadyHandler invoked
[DEBUG][2018-10-10 21:14:14] Kinesis Video stream l53-pi-1 is Ready.
[DEBUG][2018-10-10 21:14:14] Stream is ready
[INFO ][2018-10-10 21:14:15] 
kinesisVideoStreamFormatChanged(): Stream format changed.
[DEBUG][2018-10-10 21:14:15] putStreamHandler invoked
[... 119 lines cut ...]
	>> Current transfer rate (bps): 21481688 (20978 Kbps)
[DEBUG][2018-10-10 21:14:17] Wrote 463 bytes to Kinesis Video. Upload stream handle: 0
[ERROR][2018-10-10 21:14:17] Failed to submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp: 23389071 presentation timestamp: 23048832
Dropped frame!
[DEBUG][2018-10-10 21:14:17] Wrote 2442 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 2723 bytes to Kinesis Video. Upload stream handle: 0
[INFO ][2018-10-10 21:14:17] Curl post body write function for stream: l53-pi-1 and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":461,"FragmentNumber":"91343852333181447257866054353933030195990487054"}

[DEBUG][2018-10-10 21:14:17] fragmentAckReceivedHandler invoked
[DEBUG][2018-10-10 21:14:17] Wrote 3630 bytes to Kinesis Video. Upload stream handle: 0
[INFO ][2018-10-10 21:14:17] Curl post body write function for stream: l53-pi-1 and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":2311,"FragmentNumber":"91343852333181447262817814511074551760000606190"}

[DEBUG][2018-10-10 21:14:17] fragmentAckReceivedHandler invoked
[DEBUG][2018-10-10 21:14:17] Wrote 2799 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 2913 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 3409 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 2571 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 2778 bytes to Kinesis Video. Upload stream handle: 0
[INFO ][2018-10-10 21:14:17] Curl post body write function for stream: l53-pi-1 and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":461,"FragmentNumber":"91343852333181447257866054353933030195990487054"}

[DEBUG][2018-10-10 21:14:17] fragmentAckReceivedHandler invoked
[DEBUG][2018-10-10 21:14:17] Wrote 3470 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 2531 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2018-10-10 21:14:17] Wrote 2717 bytes to Kinesis Video. Upload stream handle: 0
```

frenchie4111, 2018-10-10:
Same issue after:

1. Changed line 220-223 to:

```
// if (data->h264_stream_supported) {
    buffer->pts = buffer->dts;
//}
```

2. reran `./install-script`

MushMal, 2018-10-10:
@frenchie4111 the special case handling in LN220-223 contains a workaround for some "bad" media pipelines which produce Idr frame timestamps that contain pts != dts. What @chehefen suggested is to comment all 3 lines 220 - 223, that is not to set the pts = dts. This might produce an issue if the encoder is returning pts and dts which are not on the same timeline. Please post your observations after commenting the assignment and if you can, also include some information about the version of the firmware and what type of camera is attached.

frenchie4111, 2018-10-10:
Ah okay. I misunderstood.

I have updated code as mentioned, (commented out the assignment as well) and it is now dropping frames more often.

```
$ /opt/vc/bin/vcgencmd version
Oct  8 2018 17:47:48 
Copyright (c) 2012 Broadcom
version 656741eb5ba785fc4f1014a3a3b1c0e9c2cc8487 (clean) (release)
```

Camera: https://www.amazon.com/dp/B012V1HEP4/ref=pe_2640190_232748420_TE_item

Logs:
```
$ ./kinesis_video_gstreamer_sample_app l53-pi-1
[INFO ][2018-10-10 21:38:38] Initializing curl.
[INFO ][2018-10-10 21:38:38] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-10-10 21:38:38] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-10-10 21:38:38] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-10-10 21:38:38] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-10-10 21:38:38] Refreshing credentials. Force refreshing: 0 Now time is: 1539207518 Expiration: 0
[INFO ][2018-10-10 21:38:38] New credentials expiration is 1539209918
[DEBUG][2018-10-10 21:38:38] createDeviceHandler invoked
[INFO ][2018-10-10 21:38:38] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-10-10 21:38:38] clientReadyHandler invoked
[DEBUG][2018-10-10 21:38:38] Awaiting for the producer to become ready...
[DEBUG][2018-10-10 21:38:38] Kinesis Video producer is Ready.
[DEBUG][2018-10-10 21:38:38] Client is ready
[INFO ][2018-10-10 21:38:38] Creating Kinesis Video Stream l53-pi-1
[INFO ][2018-10-10 21:38:38] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-10-10 21:38:38] describeStreamHandler invoked
[INFO ][2018-10-10 21:38:38] Refreshing credentials. Force refreshing: 0 Now time is: 1539207518 Expiration: 0
[DEBUG][2018-10-10 21:38:38] Awaiting for the stream to become ready...
[DEBUG][2018-10-10 21:38:38] describeStream response: {"StreamInfo":{"CreationTime":1.539204514969E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:340301842845:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:340301842845:stream/l53-pi-1/1539204514969","StreamName":"l53-pi-1","Version":"J0O4OEOat5NtGSMISrNT"}}
[INFO ][2018-10-10 21:38:38] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:340301842845:stream/l53-pi-1/1539204514969
[INFO ][2018-10-10 21:38:38] stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:340301842845:stream/l53-pi-1/1539204514969
[INFO ][2018-10-10 21:38:38] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-10-10 21:38:38] streamingEndpointHandler invoked
[INFO ][2018-10-10 21:38:38] Refreshing credentials. Force refreshing: 0 Now time is: 1539207518 Expiration: 0
[DEBUG][2018-10-10 21:38:38] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2018-10-10 21:38:38] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2018-10-10 21:38:38] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2018-10-10 21:38:38] streamingTokenHandler invoked
[INFO ][2018-10-10 21:38:38] Refreshing credentials. Force refreshing: 1 Now time is: 1539207518 Expiration: 1539209918
[INFO ][2018-10-10 21:38:38] New credentials expiration is 1539209918
[INFO ][2018-10-10 21:38:38] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2018-10-10 21:38:38] streamReadyHandler invoked
[DEBUG][2018-10-10 21:38:38] Kinesis Video stream l53-pi-1 is Ready.
[DEBUG][2018-10-10 21:38:38] Stream is ready
[INFO ][2018-10-10 21:38:38] 
kinesisVideoStreamFormatChanged(): Stream format changed.
[DEBUG][2018-10-10 21:38:38] putStreamHandler invoked
[DEBUG][2018-10-10 21:38:38] Created a new PutMedia request with stream upload handle: 0
[INFO ][2018-10-10 21:38:38] 
putStreamResultEvent(): Put stream result event.
[ERROR][2018-10-10 21:38:38] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 4221518 presentation timestamp: 4148619
Dropped frame!
[INFO ][2018-10-10 21:38:38] Creating new connection for Kinesis Video stream: l53-pi-1
[INFO ][2018-10-10 21:38:38] Refreshing credentials. Force refreshing: 0 Now time is: 1539207518 Expiration: 0
[ERROR][2018-10-10 21:38:38] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 4603259 presentation timestamp: 4228798
Dropped frame!
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 4870568 presentation timestamp: 4537295
Dropped frame!
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 5235745 presentation timestamp: 4845766
Dropped frame!
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 5484390 presentation timestamp: 5154250
Dropped frame!
[DEBUG][2018-10-10 21:38:39] Curl post header write function returned:HTTP/1.1 100 Continue

[DEBUG][2018-10-10 21:38:39] Curl post header write function returned:

[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 5794679 presentation timestamp: 5462757
Dropped frame!
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 6132888 presentation timestamp: 5771227
[... cut ...]
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 7085813 presentation timestamp: 6696668
Dropped frame!
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 7335249 presentation timestamp: 7005156
Dropped frame!
[ERROR][2018-10-10 21:38:39] Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 7686826 presentation timestamp: 7313625
```

frenchie4111, 2018-10-10:
After applying your patch that error message seems to have gone away, but I am still getting "No Fragments Found" on the aws stream page for my kv stream.

frenchie4111, 2018-10-10:
Looks like I had the region wrong. It's working now! Thanks for the help guys.

Should we leave this open to track that patch.txt?

chehefen, 2018-10-10:
Glad to hear that. You can close the issue. We will track it internally.

closed 2018-10-10 by frenchie4111