awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #137: How do I configure the cpp demo to send producer timestamp as current system time

frenchie4111, 2018-10-12:
How do I configure the cpp stream demo to send producer timestamp as current system time?

It seems to be sending time since stream start as the producer timestamp right now. Is it easy to send system time instead?

chehefen, 2018-10-12:
The frame timestamp is set here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L180. Right now, it's using the timestamp that gstreamer produces.

frenchie4111, 2018-10-12:
It looks like it's probably using MONOTONIC time. Any easy way to change this to realtime? I tried the below code to no avail:

```
    GstClock* gstSystemClk = gst_system_clock_obtain();
    g_object_set(gstSystemClk, "clock-type", GST_CLOCK_TYPE_REALTIME, NULL);
```

frenchie4111, 2018-10-12:
This seems to be working for me.

```
    frame->decodingTs = static_cast<UINT64>(std::chrono::system_clock::now().time_since_epoch().count()) / DEFAULT_TIME_UNIT_IN_NANOS;
    frame->presentationTs = static_cast<UINT64>(std::chrono::system_clock::now().time_since_epoch().count()) / DEFAULT_TIME_UNIT_IN_NANOS;
```

MushMal, 2018-10-12:
@frenchie4111 using the same timestamp for both pts and dts might break you on B-frames. True, most streaming encoders will not produce B-frames but if they do then the pts != dts.

Also, you can use frame->pts = frame->dts = timestamp to avoid a duplicate call to system clock.

Could you please share some information on what you are trying to do? If you are writing your own consumer then you could perhaps add the stream time to the relative time that's produced?

frenchie4111, 2018-10-12:
I have a second set of code running on the device that is producing data tagged with system timestamps, and I am trying to line them up as closely as possible.

frenchie4111, 2018-10-15:
This setup is working for me (with @MushMal's enhancement as well). I am going to close this

closed 2018-10-15 by frenchie4111