awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #138: Video stream is not working in AWS kinesis

satish0708, 2018-10-15:
Hi Team,

I have installed producer sdk cpp on raspberry pi and created test in kinesis stream.
when I  execute following command it's working on raspberry pi but video stream is not showing in kinesis video stream when I click on test and it's getting an following error.

No fragments found
No video fragments were found. Verify that video is currently being streamed and that the fragment timecodes are correct.
**my stream name is : test**
 I have checked on raspberry pi the camera is supported and detected without any issues.

$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=5120000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="YourStreamName"  access-key="YourAccessKey" secret-key="YourSecretKey" 

below are the logs:

essage from syslogd@raspberrypi at Oct 14 15:35:00 ...
kernel:[ 2247.202381] bee0: 7f0ae720 7f0b26bc 7f0ae64c 00e16c04 b5ed0ed0 ad449480 40045612 0000000a

Message from syslogd@raspberrypi at Oct 14 15:35:00 ...
kernel:[ 2247.202399] bf00: a7b5bf7c a7b5bf10 8029e72c 7f0ae658 ad449480 80242adc a7b5bf74 a7b5bf28

Could you please help me on this if any one has any idea on this and it would be great help.
Thanks in advance

MushMal, 2018-10-15:
@satish0708, it's not immediately clear whether there is an issue on the producer side. Could you please either paste the logs here or email those to kinesis-video-support@amazon.com which can help to understand the root cause.

Could you please also check 

1) Whether you have the correct credentials supplied to the sample application
2) Whether you are using the correct region in the console that matches the applications region (the default in the application is us-west-2).

johankumps, 2019-02-09:
Hi,

I have a similar issue. My Raspberry Pi originated stream is not appearing in Kinesis. Please consider the logs attached. Can somebody please help me with this one?

Thanks in advance!
Johan,
[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2848327/log.txt)

MushMal, 2019-02-10:
@johankumps the logs you've attached have the following error:

[DEBUG][2019-02-09 22:25:21] Curl post body write function for stream: test and upload handle: 0 returned: {"EventType":"ERROR","FragmentTimecode":0,"FragmentNumber":"91343852333183680496745165025638096697789402774","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}

Could you please try to provide more information about what setup you use on which device? What version of the SDK do you use?

johankumps, 2019-02-10:
@MushMal 

Hi MushMal,

I use a Raspberry Pi 3 + the Pi camera.  This with the SDK from Git : https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

What else do you want to know?

Kind regards,
Johan,

chehefen, 2019-02-10:
Hi, can you do `export DEBUG_DUMP_FRAME_INFO=TRUE` and then run the sdk for a minute or two and send us the log? Thanks.

johankumps, 2019-02-11:
Hi,

I'm afraid we are having some timezone issues here... :) I'm located in belgium so UTC +1. I'll send you the logs tomorrow morning, meaning in about 12 hours as soon as I arrive at my desk.
Thanks in advance for helping me out.

Kind regards,
Johan,

johankumps, 2019-02-15:
Up

chehefen, 2019-02-16:
It's this frame that cause all the issue `pts: 3966768, dts: 4005977, duration: 10000, size: 26, trackId: 0, isKey: 0`. Can you pull the latest sdk and run the sample again? If that didnt get rid of the issue, can you apply this patch and let us know how it goes? Thanks. 
[patch.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2871445/patch.txt)

johankumps, 2019-02-18:
@chehefen ,

I first updated from git, executed the install scripts an executed the test. No luck.
Then I applied the path using git apply patch.txt in the /amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build folder. Executed the install script and the test. No luck.

I attached the logging. Can you have a look at it again?

Should I use some stable release branch instead of using the master?

Thanks!

johankumps, 2019-02-21:
Any updates please?

chehefen, 2019-02-21:
Hi @johankumps, the patch should be applied at /amazon-kinesis-video-streams-producer-sdk-cpp folder by running `git apply patch.txt`. If the previous patch still dont work then can you remove that patch by doing `git checkout -- .` at /amazon-kinesis-video-streams-producer-sdk-cpp folder and apply this patch. Then run the sample app for 1 minute and send us the log so that we can diagnose what that frame was. Thanks
[patch.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2890661/patch.txt)

johankumps, 2019-02-25:
Hi @chehefen,

None of the patches seem to fix the issue. Please find logs in attachment. In the Kinesis console I see this : LIVE-98.0 s. Mind the negative number is that correct?

Thanks for your help. I assume you have Pi installations working fine. Is there a stable build of the SDK that I can use?

Kind regards,
Johan,
[25022019_patch2.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2902407/25022019_patch2.log)

chehefen, 2019-02-25:
can you add this block of code 
```
(GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DISCONT) && GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_HEADER) && GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DELTA_UNIT)) ||
```
after this line https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L351. It should fix the issue for you.

BWrites, 2019-02-25:
@chehefen I was getting this identical issue also running on a RasPi 3 and a RasPi camera.  I've just implemented this fix and it has worked. Thanks!!

johankumps, 2019-02-26:
@chehefen This is indeed working! I still have the second patch applied too. Is this as expected?
Maybe this extra block can be added to the source in git too.

Thanks for your help!

Johan,

msjoshi, 2019-03-21:
@johankumps 

I'm having a similar issue with the latest version of the Producer SDK. Can you confirm which solution worked for you? I see a couple of patches and one code change. I'm assuming the patch would be already applied in the latest version.

BWrites, 2019-03-21:
@msjoshi

I didn’t have to apply any of the patches to fix the issue - I only had to make the code change [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/138#issuecomment-467167640)

Not sure if this fix has made its way into the code base yet...

msjoshi, 2019-03-21:
@BWrites 

I'm not using the sample app but using `get-launch-1.0` command. If its not made its way in the SDK code it won't work for me.

Raghava248, 2019-05-02:
Hi,
Iam trying to run this 
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="" storage-size=512 access-key="" secret-key="" aws-region=""
WARNING: erroneous pipeline: no element "h264parse"
ater specfying all the keys and stream name it is giving me this error
basically Iam trying to stream video using amazon kinesis

chehefen, 2019-05-02:
can you tell us how you installed the sdk? Did you use install-script or min-install-script?

nagendra092918, 2019-05-03:
Thank You I solved the issue by exporting the path

Raghava248, 2019-05-04:
Hi Iam trying to live stream using gstreamer plugin in amazon kinesis
rohan@rohan-HP-15-Notebook-PC:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format =I420,width=640,height=480,framerate=30/1 ! vtenc_h264_hw allow-frame-recording=FALSE realtime= TRUE ma-keyframe-interval=45 bitrate=500 ! h264parse ! video/-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name= "Aerologiks2" storage-size= 1024 aws-region = us-east-1 log-config=kvslog
WARNING: erroneous pipeline: no element "vtenc_h264_hw"

But Iam stuck here please help me in doing this

MushMal, 2019-05-05:
@Raghava248 vtenc_h264_hw is a hardware codec which is available on OSx only. Please try your platform specific encoder or use a software encoder like x264

MushMal, 2019-05-05:
Folks, I am resolving this issue as there are already multiple threads in it. Please open a separate issue to discuss a particular issue.

closed 2019-05-05 by MushMal

biranchi2018, 2019-09-08:
> @chehefen I was getting this identical issue also running on a RasPi 3 and a RasPi camera. I've just implemented this fix and it has worked. Thanks!!

where to add ?

   buffer = gst_sample_get_buffer(sample);
    isHeader = GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_HEADER);
    isDroppable = GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_CORRUPTED) ||
                  GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DECODE_ONLY) ||
                  (GST_BUFFER_FLAGS(buffer) == GST_BUFFER_FLAG_DISCONT) ||
                  (GST_BUFFER_FLAGS(buffer) == (GST_BUFFER_FLAG_DISCONT | GST_BUFFER_FLAG_HEADER)) ||
                  // drop if buffer contains header only and has invalid timestamp
                  (isHeader && (!GST_BUFFER_PTS_IS_VALID(buffer) || !GST_BUFFER_DTS_IS_VALID(buffer)));

chehefen, 2019-09-09:
You can do this:
```
buffer = gst_sample_get_buffer(sample);
isHeader = GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_HEADER);
isDroppable = GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_CORRUPTED) ||
GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DECODE_ONLY) ||
(GST_BUFFER_FLAGS(buffer) == GST_BUFFER_FLAG_DISCONT) ||
(GST_BUFFER_FLAG_IS_SET(GST_BUFFER_FLAG_DISCONT) && GST_BUFFER_FLAG_IS_SET(GST_BUFFER_FLAG_DELTA_UNIT)) ||
// drop if buffer contains header only and has invalid timestamp
(isHeader && (!GST_BUFFER_PTS_IS_VALID(buffer) || !GST_BUFFER_DTS_IS_VALID(buffer)));
```

Fernando-Oliveira86, 2020-01-02:
> You can do this:
> 
> ```
> buffer = gst_sample_get_buffer(sample);
> isHeader = GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_HEADER);
> isDroppable = GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_CORRUPTED) ||
> GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DECODE_ONLY) ||
> (GST_BUFFER_FLAGS(buffer) == GST_BUFFER_FLAG_DISCONT) ||
> (GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DISCONT) && GST_BUFFER_FLAG_IS_SET(buffer, GST_BUFFER_FLAG_DELTA_UNIT)) ||
> // drop if buffer contains header only and has invalid timestamp
> (isHeader && (!GST_BUFFER_PTS_IS_VALID(buffer) || !GST_BUFFER_DTS_IS_VALID(buffer)));
> ```

Hello, I have tried to implement this code as well as the others mentioned in this thread but none succeeded.
The error message "FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS" keeps popping up, and cutting off the video stream. Any other solutions?

chehefen, 2020-01-02:
@Fernando-Oliveira86 can you open a new issue going over what your video source is and how are you streaming to kvs? Thanks.