awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #139: liblog4cplus-1.2.so.5 Error tried every mentioned solution but to no avail.

surbhitrao, 2018-10-16:
./kinesis_video_gstreamer_sample_app: error while loading shared libraries: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory

surbhitrao, 2018-10-16:
That error is gone.But now it's showing a new error.

`log4cplus:ERROR could not open file kvs_log_configuration

log4cplus:ERROR No appenders could be found for logger (com.amazonaws.kinesis.video.gstreamer).
log4cplus:ERROR Please initialize the log4cplus system properly.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.
Aborted`

unicornss, 2018-10-16:
> terminate called after throwing an instance of 'std::runtime_error'
> what(): Failed to create Kinesis Video Stream **- timed out.**

Could you provide details on the following to debug further.

1.  folder (current working directory) where you are running the demo application
2.  command line you are using to start the demo application and 
3.  the environment  ( OS version running in laptop or  virtualbox ) 
4.  confirm if you are using valid credentials
5.  check if all environment variables are set as in the [Readme](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#setting-credentials-in-environment-variables)  instructions

surbhitrao, 2018-10-16:
1. /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/
2 and 3. I am using Raspberry Pi 3 which runs on Raspbian OS and I am using the provided terminal.
4 and 5. checked.

unicornss, 2018-10-16:
Can you tell us how are you planning to use the Kinesis Video Streaming? 
> Failed to create Kinesis Video Stream - timed out. 

error suggests either network is not available (If there are two network interfaces available - please make sure the network connectivity exists for both Ethernet and wireless) or credentials and region are not set properly. Use env command to check the required access key,  secret key are available.

Are you starting the demo application as `./kinesis_video_gstreamer_sample_app yourstreamname` from the` /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/` folder ?

unicornss, 2018-10-16:
In order to help  debug your issue, can you run the demo sample application from the `kinesis-video-native-build` folder? 

`./kinesis_video_gstreamer_sample_app  -w 640 -h 480 my-stream`

and send us the full output you are seeing...

surbhitrao, 2018-10-16:
Same error.

> pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ ./kinesis_video_gstreamer_sample_app -w 640 -h 480 my-stream
./kinesis_video_gstreamer_sample_app: error while loading shared libraries: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory

surbhitrao, 2018-10-16:
After exporting the LD_LIBRARY_PATH 
I got this 

> pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ ./kinesis_video_gstreamer_sample_app -w 640 -h 480 my-stream
[INFO ][2018-10-16 20:43:22] Initializing curl.
[INFO ][2018-10-16 20:43:22] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-10-16 20:43:22] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-10-16 20:43:22] 
heapInitialize(): Creating AIV heap.
[INFO ][2018-10-16 20:43:22] 
heapInitialize(): Heap is initialized OK
[INFO ][2018-10-16 20:43:22] Refreshing credentials. Force refreshing: 0 Now time is: 1539722602 Expiration: 0
[INFO ][2018-10-16 20:43:22] New credentials expiration is 1539725002
[DEBUG][2018-10-16 20:43:22] createDeviceHandler invoked
[INFO ][2018-10-16 20:43:22] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-10-16 20:43:22] clientReadyHandler invoked
[DEBUG][2018-10-16 20:43:22] Awaiting for the producer to become ready...
[DEBUG][2018-10-16 20:43:22] Kinesis Video producer is Ready.
[DEBUG][2018-10-16 20:43:22] Client is ready
[INFO ][2018-10-16 20:43:22] Creating Kinesis Video Stream my-stream
[INFO ][2018-10-16 20:43:22] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-10-16 20:43:22] describeStreamHandler invoked
[DEBUG][2018-10-16 20:43:22] Awaiting for the stream to become ready...
[INFO ][2018-10-16 20:43:22] Refreshing credentials. Force refreshing: 0 Now time is: 1539722602 Expiration: 0
[WARN ][2018-10-16 20:43:24] HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20181016/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=3f4228c133980eee658b1f235202d1b99be4b73147b4916361f8fd5cfef2f5a0
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.5.0 GCC/6.3.0 Linux/4.14.71-v7+ armv7l
    X-Amz-Date: 20181016T204322Z
[DEBUG][2018-10-16 20:43:24] describeStream response: {"message":"The security token included in the request is invalid."}
[INFO ][2018-10-16 20:43:24] Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
[INFO ][2018-10-16 20:43:24] 
describeStreamResultEvent(): Describe stream result event.
[ERROR][2018-10-16 20:43:24] Submitting event result for stream: 26329574 failed with: 1375731728
[ERROR][2018-10-16 20:43:24] Reporting stream error. Errored timecode: 0 Status: 1375731728
[INFO ][2018-10-16 20:43:52] Freeing Kinesis Video Stream my-stream
[INFO ][2018-10-16 20:43:52] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[ERROR][2018-10-16 20:43:52] Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.
Aborted

zhiyua-git, 2018-10-16:
Hi @surbhitrao,

[WARN ][2018-10-16 20:43:24] HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}

This indicates your credentials provided is not correct. Can you confirm you have already set the credentials properly?
export AWS_ACCESS_KEY_ID=<your access key>
export AWS_SECRET_ACCESS_KEY=<your secret access key>
export AWS_DEFAULT_REGION=<your region>

Thanks,
-ZH

surbhitrao, 2018-10-16:
Yes the credentials are correct.I am trying to run the command but I get this error now

> pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
$ ./kinesis_video_gstreamer_sample_app -w 640 -h 480 my-stream
[INFO ][2018-10-16 20:52:41] Initializing curl.
[INFO ][2018-10-16 20:52:41] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-10-16 20:52:41] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-10-16 20:52:41] 
heapInitialize(): Creating AIV heap.
[ERROR][2018-10-16 20:52:41] 
aivHeapInit(): Failed to allocate heap with limit size 536870912
[ERROR][2018-10-16 20:52:41] 
heapInitialize(): Failed to initialize native heap.
[INFO ][2018-10-16 20:52:41] 
heapRelease(): Freeing native heap.
[INFO ][2018-10-16 20:52:41] 
freeKinesisVideoClient(): Freeing Kinesis Video Client
Segmentation fault

unicornss, 2018-10-16:
> aivHeapInit(): Failed to allocate heap with limit size 536870912

The log shows that the default memory size of 512MB is not available in your Pi setup.
 
 
1. You can change the demo application storage  size in the [Line number 126](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L126)
  `device_info.storageInfo.storageSize = 512 * 1024 * 1024;`  to  `device_info.storageInfo.storageSize = 64 * 1024 * 1024;`

2. run install-script again (it should be fast as it has to compile on this change)

3. Start the demo app with all the required environment variables are set including the LD_LIBRARY_PATH