awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #141: Not getting any result from the data stream using sharditerator

surbhitrao, 2018-10-21:
Hi I am not getting any result from the rekogniton data stream. Although I can see the video stream in my browser.

 
 String shardIterator;
        GetShardIteratorRequest getShardIteratorRequest = new GetShardIteratorRequest();
        getShardIteratorRequest.setStreamName("AmazonRekogntionResults");
        getShardIteratorRequest.setShardId("shardId-000000000000");
        getShardIteratorRequest.setShardIteratorType("LATEST");
        GetShardIteratorResult getShardIteratorResult = client.getShardIterator(getShardIteratorRequest);
         shardIterator = getShardIteratorResult.getShardIterator();
        List<Record> records;
        while (true) {
        
            GetRecordsRequest getRecordsRequest = new GetRecordsRequest();
            getRecordsRequest.setShardIterator(shardIterator);
            GetRecordsResult result = client.getRecords(getRecordsRequest);
            Log.i("Result",(result.toString()));
            records = result.getRecords();
            Log.i("Record",(records.toString()));
            for (Record r : records) {
                Log.i("Output1",(r.getSequenceNumber()));
                Log.i("Output2",(r.getPartitionKey()));

                System.out.println(r.getSequenceNumber());
                System.out.println(r.getPartitionKey());
                byte[] bytes = r.getData().array();
                System.out.println(new String(bytes));
                Log.i("Outputi",(r.getData().array().toString()));

            }
            try {
                Thread.sleep(1000);
            } catch (InterruptedException exception) {
                throw new RuntimeException(exception);
            }
            shardIterator = result.getNextShardIterator();
        }

I am expecting the face matched Json response in my Android system. The kinesis stream is being stream from my Raspberry Pi's camera.

vijaygj, 2018-10-21:
Just see if indeed any data is being put into Kinesis Data stream using AWS Cloudwatch metrics. So you can debug if there are no faces being recognized or there code snippet is not going through the records fast enough with 1 second sleep and latest as the start time.

bdhandap, 2018-10-22:
@surbhitrao  Can you do a describe stream processor and check the status of your rekognition stream processor? It should be in RUNNING state to process the KVS stream

https://docs.aws.amazon.com/rekognition/latest/dg/API_DescribeStreamProcessor.html

https://docs.aws.amazon.com/rekognition/latest/dg/API_StreamProcessor.html

surbhitrao, 2018-10-22:
Yes I checked using 

> aws rekognition list-stream-processors --region us-west-2

It's in the running state.

bdhandap, 2018-10-22:
@surbhitrao Can you check the CloudWatch metrics like @vijaygj  suggested? Also can you send your AWS account id, kinesis video stream name and rekognition stream processor id to kinesis-video-support@amazon.com?

closed 2018-10-23 by surbhitrao