awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #142: No fragment found

fabolandry, 2018-10-21:
It's seem to be working but I have the error message:
No Fragments found
No video fragments were found. Verify that video is currently being streamed and that the fragment timecodes are correct.

No Live feed is showing from my Rpi-3

Below you can see a screen shoot of what I have on the terminal 
[image]

zhiyua-git, 2018-10-22:
Hi @fabolandry ,

Can you double check whether you are using the correct region in the console that matches the applications region (the default in the application is us-west-2)?

Also, could you please either paste the logs here or email those to kinesis-video-support@amazon.com which can help to understand the root cause?

Thanks,
-ZH

fabolandry, 2018-10-23:
I changed the region to us-west-2. It's working. Thank you so much. Great Project !!!!!!!

unicornss, 2018-10-23:
Glad to hear that @fabolandry.

closed 2018-10-23 by unicornss