awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #143: log4cplus issues on installation

jdbuss, 2018-11-01:
I have attempted to set this application up both with a windows installation as well as utilizing the docker installation.  Here are some issues that I've ran into
1) First attempt at installing on Windows, there was an error in the installation of Strawberry perl that couldn't get access to a subfolder that the installation creates under downloads.
2) Another windows attempt and I received an error that the path was too long.  (I moved the whole installation folder to then be directly under C
3) After what looked like a clean installation, I received error messages when trying to run the app that about a handful of .dll files couldn't be found
4) I attempted the docker installation and the build failed with the following:  Checking log4cplus at /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus.dylib/.so
log4cplus lib not found.

In looking back at some of the logging on the windows installation I found reference to log4cplus not found messages as well.  Also when I looked at the docker installation error and followed the path.  I noticed that under downloads, there is no 'local' folder?

jdbuss, 2018-11-01:
In following the Readme directly under the docker folder, I used those docker files and received the log4cplus error above.  I also attempted to use the docker files under the windows folder and it failed faster with:  

Sending build context to Docker daemon  8.192kB
Step 1/9 : FROM microsoft/windowsservercore
latest: Pulling from microsoft/windowsservercore
3889bb8d808b: Pulling fs layer
c0f6863f3dd4: Pulling fs layer
image operating system "windows" cannot be used on this platform

zhiyua-git, 2018-11-01:
Hi @jdbuss,

Sorry to hear you running into these issues. Can you provide some more information on one issue?
1. subfolder cannot be accessed due to path too long in Windows is a known issue.
2. The windows install script is adding directory to path, so if it is being executed for multiple time, it can screw up the cmd terminal with path too long. Please open a new cmd when you start the build.
3. This one seems promising, can you try to run it again and tell us what .dll it could not find? 
4. Can you share some log on this one so we can see how it failed? We will also try to reproduce it on our windows machine.
5. As you can run docker script using linux image (the one failed with 'log4cplus lib not found.'), seems you are running linux container on your docker, it is normal to fail with windows image. If you want to try windows docker image, you would need to switch your docker setting to use windows container.

Let me know if there answer some of your questions.

Thanks,
-ZH

jdbuss, 2018-11-01:
5. I'm running Docker for Windows on my local desktop, that's not Windows based?  What version of the docker file(s) are in the base directory right under docker_native_scripts?  Those get me the furthest, I tried both the files under Windows AND amazonlinux-docker and neither work.

zhiyua-git, 2018-11-01:
3. Can you try running sample app after adding these?
set path=%path%;C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\bin
or 
set path=%path%;C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86\bin  if your windows is 32 bit.

5. You can right click on the docker tray icon and see if you have **Switch to Linux containers...**/**Switch to Windows containers...**. If you have Switch to Windows containers..., then it means your docker is using Linux container. For Linux container you would need to use [this Dockerfile](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docker_native_scripts/Dockerfile). Dockerfile under amazon linux directory is specific for customer using amazon linux images in EC2 instances.

jdbuss, 2018-11-01:
3. Received some additional files not found, but different errors in a popup window, however it did get further and try to run the application.  I'm going to digest the logs to double check connectivity/security is all in sync.

4. I was able to switch and working on the installation process. I'll update later.

zhiyua-git, 2018-11-01:
3. Good to hear you can get the application executed. Can you share screenshot for those errors popped up? We might see those errors before.
5. Actually you don't need to run the docker build from scratch. Try following the steps [here](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker) to pull down the image from our ECR repo may save you lots of time. Specifically for windows, see Download the Docker image for Windows.

jdbuss, 2018-11-02:
so, both options now are getting close.  I do have the project installed directly to the C: drive.  Here is beginning of the log from the docker image:

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstdecklink.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstdecklink.dll': The specified procedure could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstdirectsound.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstdirectsound.dll': The specified module could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstdirectsoundsrc.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstdirectsoundsrc.dll': The specified module could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstopengl.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstopengl.dll': The specified module could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstopenglmixers.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstopenglmixers.dll': The specified module could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstopenh264.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstopenh264.dll': The specified procedure could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstsoundtouch.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstsoundtouch.dll': The specified procedure could not be found.
WARNING: no real random source present!

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgsttaglib.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgsttaglib.dll': The specified procedure could not be found.

(kinesis_video_gstreamer_sample_rtsp_app.exe:1984): GStreamer-WARNING **: Failed to load plugin 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstwinks.dll': 'C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstwinks.dll': The specified module could not be found.

Why would it be looking at a folder C:\opt\ ?

jdbuss, 2018-11-02:
Remainder of the log information:  [INFO ][2018-11-02 12:41:12] Initializing curl.
[INFO ][2018-11-02 12:41:12]
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-11-02 12:41:12]
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-11-02 12:41:12]
heapInitialize(): Creating AIV heap.
[INFO ][2018-11-02 12:41:12]
heapInitialize(): Heap is initialized OK
[INFO ][2018-11-02 12:41:12] Refreshing credentials. Force refreshing: 0 Now time is: 1541162472 Expiration: 0
[INFO ][2018-11-02 12:41:12] New credentials expiration is 1541164872
[DEBUG][2018-11-02 12:41:12] createDeviceHandler invoked
[INFO ][2018-11-02 12:41:12]
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-11-02 12:41:12] clientReadyHandler invoked
[DEBUG][2018-11-02 12:41:12] Awaiting for the producer to become ready...
[DEBUG][2018-11-02 12:41:12] Kinesis Video producer is Ready.
[DEBUG][2018-11-02 12:41:12] Client is ready
[INFO ][2018-11-02 12:41:12] Creating Kinesis Video Stream streamProcessorForBlog
[INFO ][2018-11-02 12:41:12]
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-11-02 12:41:12] describeStreamHandler invoked
[DEBUG][2018-11-02 12:41:12] Awaiting for the stream to become ready...
[INFO ][2018-11-02 12:41:12] Refreshing credentials. Force refreshing: 0 Now time is: 1541162472 Expiration: 0
[ERROR][2018-11-02 12:41:18] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5625 milliseconds
[WARN ][2018-11-02 12:41:18] HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ZZZZZZZZZZZZZZZZZZZZZ/20181102/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, 
	Signature=YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.5.0 MSVC/1915 Windows/6.2.9200 AMD64
    X-Amz-Date: 20181102T124112Z
[DEBUG][2018-11-02 12:41:18] describeStream response:
[INFO ][2018-11-02 12:41:18] Describe stream did not find the stream streamProcessorForBlog in Kinesis Video (stream will be created)
[INFO ][2018-11-02 12:41:18]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-11-02 12:41:18] describeStreamHandler invoked
[INFO ][2018-11-02 12:41:18] Refreshing credentials. Force refreshing: 0 Now time is: 1541162478 Expiration: 0
[ERROR][2018-11-02 12:41:24] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5625 milliseconds
[WARN ][2018-11-02 12:41:24] HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ZZZZZZZZZZZZZZZZZZZZZ/20181102/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=a0adb3a959b16b79447d524fa101b158373dbf90dba2faf234ef8346032f2855
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.5.0 MSVC/1915 Windows/6.2.9200 AMD64
    X-Amz-Date: 20181102T124118Z
[DEBUG][2018-11-02 12:41:24] describeStream response:
[INFO ][2018-11-02 12:41:24] Describe stream did not find the stream streamProcessorForBlog in Kinesis Video (stream will be created)
[INFO ][2018-11-02 12:41:24]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-11-02 12:41:24] describeStreamHandler invoked
[INFO ][2018-11-02 12:41:24] Refreshing credentials. Force refreshing: 0 Now time is: 1541162484 Expiration: 0
[ERROR][2018-11-02 12:41:24] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo.us-west-2.amazonaws.com
[WARN ][2018-11-02 12:41:24] HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ZZZZZZZZZZZZZZZZZZZZZ/20181102/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, 
	Signature=YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.5.0 MSVC/1915 Windows/6.2.9200 AMD64
    X-Amz-Date: 20181102T124124Z
[DEBUG][2018-11-02 12:41:24] describeStream response:
[INFO ][2018-11-02 12:41:24] Describe stream did not find the stream streamProcessorForBlog in Kinesis Video (stream will be created)
[INFO ][2018-11-02 12:41:24]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-11-02 12:41:24] describeStreamHandler invoked
[... 28 lines cut ...]
[INFO ][2018-11-02 12:41:27] Describe stream did not find the stream streamProcessorForBlog in Kinesis Video (stream will be created)
[INFO ][2018-11-02 12:41:27]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-11-02 12:41:27] describeStreamHandler invoked
[INFO ][2018-11-02 12:41:30] Refreshing credentials. Force refreshing: 0 Now time is: 1541162490 Expiration: 0
[... cut ...]
describeStreamResultEvent(): Describe stream result event.
[ERROR][2018-11-02 12:41:36] Submitting event result for stream: 1654719490206 failed with: 1375731729
[ERROR][2018-11-02 12:41:36] Reporting stream error. Errored timecode: 0 Status: 1375731729
[INFO ][2018-11-02 12:41:42] Freeing Kinesis Video Stream streamProcessorForBlog
[INFO ][2018-11-02 12:41:42]
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[ERROR][2018-11-02 12:41:42] Failed to create Kinesis Video Stream - timed out.

jdbuss, 2018-11-02:
Here is the error from running the application (along with VERY similar output from above)

[image]

jdbuss, 2018-11-02:
what would the RTSP of the camera on my laptop be?

zhiyua-git, 2018-11-02:
Hi @jdbuss ,

Your log shows you windows host cannot find our frontend host. 
[ERROR][2018-11-02 12:41:24] curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result **Couldn't resolve host name: Could not resolve host: kinesisvideo.us-west-2.amazonaws.com**

Can you double check your network connectivity? If there are two network interfaces available - please make sure the network connectivity exists for both Ethernet and wireless.

jdbuss, 2018-11-02:
SO, I found a gstream command and at test rtsp URL and ran it and it was successfully communicating with AWS Kinesis that I have set up.

**c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\bin**>gst-launch-1.0 rtspsrc location="rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="X" storage-size=512 access-key="X" secret-key="X" aws-region="X"

Notice that I made sure I ran the command from the directory where the executable is at.

I had to reboot my laptop and when I went back to re-try running the application I went back a step where it couldn't find the libraries again.  Here is the path's that I have in my Path environment variable.  

[image]

I also noticed that when using the Docker image, there is another folder that is put in there "OPT"???  To run the docker do I need to put that path in the Environment variable as well?

I'm working off of wireless at the moment, 1 connection.

zhiyua-git, 2018-11-02:
Have you set the gstreamer environments? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md#running-the-sample-demo-applications-to-stream-video-to-kinesis-video-streams
You shouldn't need to run from the directory where the executable is at after this.

To run the docker with gstreamer plugin, you would need to put the environment variables in start_rtsp_in_docker.bat.
set GST_PLUGIN_PATH=C:\opt\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release
set GST_PLUGIN_SYSTEM_PATH=C:\opt\Downloads\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0

After that, you can substitute the 'kinesis_video_gstreamer_sample_rtsp_app.exe %3 %4' with  start_rtsp_in_docker.bat to gstreamer plugin command you want.

jdbuss, 2018-11-05:
Update: I was able to get the demo to work with a Raspberry Pi 3 device.

For my laptop, I have an initial issue with both the windows and linux docker images.  Are either of these solutions runnable to use the camera built into my laptop?
  1) Windows docker image, I need the rtsp URL to run the docker image, what would that be for an internal camera?
  2) Linux docker image, I can't run the "docker run -it --network="host" --device=/dev/video0" command because in my windows command prompt, it doesn't recognize the camera device location?

The raspberry pi couldn't initially recognize the /dev/video0 path but found a spot in documentation to assign it and then it worked.

zhiyua-git, 2018-11-05:
Hi @jdbuss ,

  From the search we made before, Docker doesn't support usb pass through or camera access on Windows. 
  As you have the application built successfully on your laptop, I would recommend using it instead of docker solution to access to your built-in camera. 
  Also you would need to [enable your camera](https://support.microsoft.com/en-us/help/4028647/windows-open-the-camera-in-windows-10) first for usage. 

-ZH

jdbuss, 2018-11-06:
OK, looks like things are resolved, multiple things going on.
1) can't use the docker solutions if trying to utilize internal laptop/pc camera
2) moving target issues, firewall had allowed traffic through to amazon at one point, but then had automatically blocked the traffic 
3) I did have camera on, but it wasn't enabled to utilize with other applications
4) had a little confusion on setting Paths (which still didn't seem to work as I had to navigate and run the final command from the gstreamer executable path).
5) despite seeing some errors/warnings on installation, some of them didn't affect the overall processing.

Issues have been resolved, appreciate the time and responsiveness getting through these.

closed 2018-11-06 by jdbuss