awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #144: Error Running the gst-launch-1.0 command to start RTSP.

lionmat, 2018-11-15:
Hi, I am trying to run the gst-launch-1.0 commnad to start RTSP.
But I am not able to start. I use below command to use kvssink, but it end up with error.
I am using docker image from amazon ECR.

```
GST_DEBUG=FIXME gst-launch-1.0 rtspsrc location=<rtsp stream> ! queue ! short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au kvssink stream-name= <stream name> storage-size=512 access-key=<access-key> secret-key=<secret-key> aws-region=<region>
```

```
0:00:00.016308656  1241      0x1ce56c0 WARN                 default grammar.y:1137:priv_gst_parse_yyerror: Error during parsing: syntax error, unexpected ASSIGNMENT, expecting IDENTIFIER or BINREF or '('
0:00:00.016338040  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:1061:priv_gst_parse_yyparse: syntax error
0:00:00.016348245  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:1061:priv_gst_parse_yyparse: syntax error
0:00:00.016353089  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:1061:priv_gst_parse_yyparse: syntax error
0:00:00.016901028  1241      0x1ce56c0 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "video"!
0:00:00.016913948  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:816:priv_gst_parse_yyparse: no element "video"
0:00:00.016923953  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:901:priv_gst_parse_yyparse: link has no sink [source=@0x1d22080]

(gst-launch-1.0:1241): GStreamer-CRITICAL **: gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed
0:00:00.016956276  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:971:priv_gst_parse_yyparse: no source element for URI "/x-h264,"
0:00:00.016966035  1241      0x1ce56c0 WARN                 default grammar.y:1137:priv_gst_parse_yyerror: Error during parsing: syntax error, unexpected ASSIGNMENT
0:00:00.016971972  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:1061:priv_gst_parse_yyparse: syntax error
0:00:00.016977967  1241      0x1ce56c0 ERROR           GST_PIPELINE grammar.y:1061:priv_gst_parse_yyparse: syntax error
```
How can I solve this error? it looks like yyparse error ?
thanks in advane.

unicornss, 2018-11-15:
Hi @lionmat 

Seems like one of the required ! is missing . Please try : 
`GST_DEBUG=FIXME gst-launch-1.0 rtspsrc location=<rtsp stream> ! queue ! short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name= <stream name> storage-size=512 access-key=<access-key> secret-key=<secret-key> aws-region=<region>`

lionmat, 2018-11-16:
Hi @unicornss, thanks for your help.

I fixed my command like below and it works but I got an another error.
```
gst-launch-1.0 rtspsrc location=<rtsp stream> short-header=TRUE \
! queue \
! rtph264depay \
! video/x-h264, format=avc,alignment=au \
! kvssink stream-name=<stream name>  storage-size=512 access-key=<access-key> secret-key=<secret-key> aws-region=<region>
```

looks like curl ssl error.
```
curl perform failed for url https://kinesisvideo.ap-northeast-1.amazonaws.com/describeStream with result SSL connect error: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
```
so I followrd the [link](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#curl-ssl-issue---unable-to-get-local-issuer-certificate
)  to rebuild local curl.
but I couldn't find /etc/ssl/cert.pem.
Do I have to copy <sdk-path>/kinesis-video-native-build/certs/cert.pem to /etc/ssl/cert.pem ?

I also tried to install with -c option, but it looks like similar problem.
```
cert.pem file does not exist in the /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build directory
```

unicornss, 2018-11-16:
Can you please refer the steps here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#install-the-certificate-in-the-operating-system-certificate-store 

Thanks

lionmat, 2018-11-16:
Thanks for your help.

I followed the instruction and downloaded pem file to /etc/ssl/cert.pem.
and then I use below command to re-install curl.

```
$ cd /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
$ rm downloads/local/lib/libcurl*
$ rm downloads/local/bin/curl*
$ cd downloads
$/usr/bin/curl -L "https://curl.haxx.se/download/curl-7.57.0.tar.xz" -o "curl-7.57.0.tar.xz"
$ tar -xvf curl-7.57.0.tar.xz
$ cd curl-7.57.0
$ export DOWNLOADS=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
$ ./configure --prefix=$DOWNLOADS/local/ --enable-dynamic --disable-rtsp --disable-ldap --without-zlib --with-ssl=$DOWNLOADS/local/ --with-ca-bundle=/etc/ssl/cert.pem
$ make
$ make install

$ which curl
/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/bin/curl
$ local/bin/curl-config --ca
/etc/ssl/cert.pem
```

it looks ok... but I still have error.
```
$ curl https://kinesisvideo.ap-northeast-1.amazonaws.com
curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number
```

lionmat, 2018-11-16:
[Solved]
I needed to set proxy inside docker container process.
I could send video stream to kvs, thanks for your help @unicornss !

closed 2018-11-16 by lionmat