awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #145: Stream MP4 file on kinesis with C++ SDK

TsunamiBit, 2018-11-17:
Hi all boys
I'm an absolute beginner in using aws and C++, and I approached aws kinesis to try and make a simple stream of an Mp4 file.
I'm using the sdk in c ++ right now and I've seen that you can use the GStreamer tool to stream.
In the examples provided in the documentation, however, I noticed that only streams of webcams or cameras are used as sources of the stream ... my question is: I can make a stream from an MP4 file (or other format) through GStreamer, encode it in H / 264 and send it to my stream?
I tried to use in the GStreamer pipeline with the "filesrc location = ..... / file.mp4" plugin but I get an error "ERROR: pipeline does not want to preroll."

Do you know if there is an alternative way to upload a video file?
Thanks a lot!

unicornss, 2018-11-19:
Hi @TsunamiBit 

Can you please provide the GStreamer pipeline you are using to demux the mp4 ? Thanks

TsunamiBit, 2018-11-20:
Hi @unicornss 
Thank you for the reply!
the pipeline is: 
"$ gst-launch-1.0 filesrc do-timestamp=TRUE location="/absolute/path/of/video.mp4" ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="my-stream" storage-size=512 access-key="MyKey" secret-key="MySecretKey" aws-region="AWSRegion"

Thank You!

chehefen, 2018-11-20:
can you try this:
"$ gst-launch-1.0 filesrc location="/absolute/path/of/video.mp4" ! qtdemux ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="my-stream" storage-size=512 access-key="MyKey" secret-key="MySecretKey" aws-region="AWSRegion"

if that didnt work, then try this:
"$ gst-launch-1.0 filesrc location="/absolute/path/of/video.mp4" ! decodebin ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="my-stream" storage-size=512 access-key="MyKey" secret-key="MySecretKey" aws-region="AWSRegion"

TsunamiBit, 2018-11-23:
hi @chehefen 
Thank you soooo much!
The second command work perfectly! 
thanks again to all guys.
you have been very kind!

unicornss, 2018-11-23:
Hi @TsunamiBit  Glad to hear that the pipeline worked.

closed 2018-11-23 by unicornss