awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #146: SDk for ios

ketansweta, 2018-11-19:
I don't see any SDK for ios ? any clue if thats going to be released.

closed 2018-12-21 by unicornss

Sneakr, 2019-10-24:
Hi, 

It has been 1 year since this was requested. How is it possible to not have iOS support yet?

kdjani, 2019-12-03:
@MushMal is there any update on the ios sdk? @unicornss

MushMal, 2019-12-03:
We are looking at the customer feedback to prioritize this work. Nothing concrete yet. As per my message above - many customers just compile the existing SDK for iOS

aeperaltan, 2020-04-12:
Hello,

I am working on a solution that requires support of Kinesis video for IOS. Can you provide details about the procedure to convert the code from C++ to objective-C / swift? Thanks!

MushMal, 2020-04-13:
@aeperaltan unfortunately, we don't have an example of binding to objective-c/swift for the core SDK. I would imagine that you could simply use the core of the SDK as is - native and write high-level binding for the public APIs like create stream and put frame that you can use in your application.

aeperaltan, 2020-04-13:
> @aeperaltan unfortunately, we don't have an example of binding to objective-c/swift for the core SDK. I would imagine that you could simply use the core of the SDK as is - native and write high-level binding for the public APIs like create stream and put frame that you can use in your application.

Thank you for the quick response. By the way, is it possible to know if publishing the IOS SDK for Kinesis video is in the roadmap? Regards.

MushMal, 2020-04-13:
@aeperaltan, unfortunately, we don't have any timelines for the iOS SDK. We are still evaluating the market for the iOS SDK and it's feature set.

Salarsoleimani, 2020-08-13:
> Hello,
> 
> I am working on a solution that requires support of Kinesis video for IOS. Can you provide details about the procedure to convert the code from C++ to objective-C / swift? Thanks!

Hey, Did you find anything useful for video streaming channel? or library that we can use?

Ankish, 2020-09-08:
@MushMal : ```In the meanwhile, the code itself can be compiled for iOS using pre-compiled dependencies like curl/ssl/gstreamer/log4cpus/etc... from existing repositories in the web. The code itself for the SDK will be the same but the dependencies and the sample apps would be different in this case.```

It will of great help, if you could provide us with simple step by step guide to achieving the same in iOS till you release an iOS Specific library. It would save a ton of time for us figuring out things.

SanthoshMKunthe, 2020-10-07:
hi @MushMal . . .  i looked around the web. . . there is no single clue, example, stackoverflow question about this topic. . . https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-android/issues/36 here sir you suggested to use either WebRTC and create a seperate channel or use C/C++ producer library. . . kindly provide atleast some data that can help us get started. . . 

Kindly shed some light on how "many customers just compile the existing SDK for iOS" can be done. . .

MushMal, 2020-10-07:
Unfortunately, I don't have an example. We currently don't support iOS as a platform. The code is quite portable and people can create build scripts to compile for iOS and a think Objective-C or SWIFT layer can be written to wrap around it.

I don't have any further pointers on this. Perhaps other members of the community could chime in?

SanthoshMKunthe, 2020-10-07:
Respected @ketansweta @Sneakr @kdjani @aeperaltan  @Salarsoleimani @Ankish . . . if you haven't found the solution yet kindly thumbs up this question. . . 

and if sir you have found the solution, if there is possibility of sharing, kindly do the same. . .

Ashish0151, 2020-10-12:
Hello @ketansweta @Sneakr @aeperaltan @Salarsoleimani @Ankish ...
I am working on a solution that requires support of Kinesis video for IOS.
Any one of you provide me procedure to convert the code from C++ to objective-C / swift?
I am also trying to connect with Support, Yet not get any reply... Very disappointed.
Please help to find out solution..

oceanvsagar, 2021-04-30:
"Platform Independent Codebase (PIC)" to "iOS"
or
"amazon-kinesis-video-streams-producer-c" to "iOS"

is not working. kindly provide updates if any.