awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #147: Surface Pro Intel avstream camera 2500 MAX_FRAGMENT_DURATION_REACHED

eorlovsky, 2018-11-21:
Having SDK build for **Surface Pro** we are not able to stream to Kinesis Video Stream. Most of video fragments get lost because of "ErrorCode":"**MAX_FRAGMENT_DURATION_REACHED**","ErrorId":4002

We've tried 2 methods of run Producer and stream in Kinesis Video Streams. Both of them failed with the same error

1) By command gst-launch-1.0 ksvideosrc do-timestamp=FALSE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key=

2) With app kinesis_video_gstreamer_sample_app.exe LiveRekognitionVideoAnalysis

eorlovsky, 2018-11-21:
[logs.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2604604/logs.txt)

Logs attached

unicornss, 2018-11-21:
Hi @eorlovsky  

Could you please try with lower `key-int-max` value e.g. `key-int-max=10`

`gst-launch-1.0 ksvideosrc do-timestamp=FALSE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key= `

You can also check if the `frame-timestamp=1` helps 

`gst-launch-1.0 ksvideosrc do-timestamp=FALSE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key= frame-timestamp=1`

Additional reference: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin-parameters.html

Thanks

eorlovsky, 2018-11-22:
@unicornss Many thanks to you ! This helps us to launch Producer SDK on a Surface Pro

closed 2018-11-23 by eorlovsky