awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #148: Surface Pro Intel avstream camera 2500 green stream img

Fuzzywuzzy1991, 2018-11-22:
Having SDK build for Surface Pro we are not able to stream to Kinesis Video Stream.
Camera identified by Windows: Intel avstream camera 2500.
We successfully run Stream on our Surface Pro with command gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key=

But we got green video on our stream. I have attached a screenshot. 
[image]

Usually photo with this camera looks like better and not green.

Any solutions to improve quality of stream image?

eorlovsky, 2018-11-23:
@unicornss do you know what kind of this could be? Have the same issue on my side

unicornss, 2018-11-23:
Hi @Fuzzywuzzy1991 @eorlovsky

 Could you try the following settings

gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key=

Also, if you could find the codec format that the camera supports, it will be helpful.  Thanks

eorlovsky, 2018-11-24:
@unicornss Thanks for the reply. But unfortunately,  your command gives the same result? Is possible to debug the problem? Maybe to enabling DEBUG log? 
Also, what do you mean by "codec format that the camera supports"? Do you have a command to execute to get this info?

eorlovsky, 2018-11-24:
[image]

unicornss, 2018-11-26:
Thanks @eorlovsky.

Could you please check if the camera has the latest driver update? or is it possible to try a USB camera if USB ports are available?
Unfortunately,  we do not have the same camera model to debug why this camera model video has green tint.

Fuzzywuzzy1991, 2018-12-10:
Hi @unicornss.
How we can use usb camera in our command gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key= 
in Windows 10.
Can you help?

zhiyua-git, 2018-12-10:
Hi @Fuzzywuzzy1991 ,

You can run gst-device-monitor-1.0 to find the device name of your USB device. If your device name is HD Webcam XXX, then your command need to be:

gst-launch-1.0 ksvideosrc **device-name="HD Webcam XXX"** do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name=“LiveRekognitionVideoAnalysis” access-key= secret-key=

unicornss, 2018-12-21:
Closing this issue for now as @zhiyua-git  has provided the details. Please reopen or create a new request if additional support is required.

closed 2018-12-21 by unicornss