awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #149: Update Documentation regarding Regions

ohlr, 2018-11-28:
In the documentation (see link bellow) there is no hint on how to set the region of the stream.
Moreover it's not clear that it has to be set in order to make the streaming work.

Would be nice to add some lines here:
[](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html)

With information from that comment:
[](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/138#issuecomment-442429315)

unicornss, 2018-11-28:
Thanks @ohlr  for your input.

We will update the [documentation](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html) to capture the `AWS_DEFAULT_REGION` setting in the same way as its captured in the [Readme](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#running-examples)  and [here](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html)

unicornss, 2019-01-04:
Thanks @ohlr.
Closing this issue as the Documentation has been updated.

closed 2019-01-04 by unicornss