awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #150: Architectural Setup - Questions

ohlr, 2018-11-30:
Hi,

I successfully managed to test two of your tutorials [Raspberry-Pi](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html) and [G-Streamer](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html).

However some things are still unclear to me:

1. Here you mention a list of ip-cameras that you tested successfully:[system-requirements](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/system-requirements.html)
Is it possible to install this sdk directly on those cameras? Or do I need an additional pc?

2. Let's say I have multiple ip cameras (2-100) in my network + a PC that runs the GStreamer-C++-SDK, how to scale efficiently from 1 camera (shown in your examples) to multiple cameras? 
Things that come to my mind:

- Would you execute separate threads for each camera, or is there a builtin functionality already in the GStreamer application? 
- Would you preallocate streams for each camera? 
- Will the pc become a (network) speed limiting factor

3. With this command `gst-launch-1.0 rtspsrc location="rtsp://username:password@camera_ip"..`

-  Is the connection between the pc and the camera encrypted? Or can anyone in the same network read username and password?
- Is it possible to use https if the camera supports that?

unicornss, 2018-12-01:
> Would you execute separate threads for each camera, or is there a builtin functionality already in the GStreamer application?

Please check if the [sample](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_multistream_app.cpp ) helps to address your scenario.

For additional use cases and questions please reach out to this email id:  kinesis-video-team@amazon.com

ohlr, 2018-12-04:
Hi,

the sample you provided seems to be exactly what I was looking for.

Is there some documentation on how to get it running?

The only thing I found was this:
`                "Usage: AWS_ACCESS_KEY_ID=SAMPLEKEY AWS_SECRET_ACCESS_KEY=SAMPLESECRET ./kinesis_video_gstreamer_sample_app base-stream-name rtsp-url-file-name\n" <<
                "base-stream-name: the application will create one stream for each rtsp url read. The base-stream-names will be suffixed with indexes to differentiate the streams\n" <<
                "rtsp-url-file-name: name of the file containing rtsp urls separated by new lines"`

Can it be executed with something like `gst-launch-1.0 ...` ?

ohlr, 2018-12-04:
So from looking I think I need to create a file eg. `rtsp_url.txt`:
`rtsp://ip_ip-cam1`
`rtsp://ip_ip-cam2`

and streams: 

> TestStream_0, TestStream_1

and then I can launch with` ./kinesis_video_gstreamer_sample_multistream_app TestStream rtsp_url.txt`

closed 2018-12-21 by aaron-freshwater-aws