awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #153: max fragment duration reached with low frame rate

laurenkemperman, 2018-12-04:
Streaming video encoded with omxh264 encoder at bitrate 5120000. Set frame rate to be compatible with high definition 1920 by 1080 at 10 frames per second, and received maximum fragment duration error. Highest resolution where streaming worked without this error was 1280 by 720 at 15 frames per second.

zhiyua-git, 2018-12-04:
Hi @laurenkemperman ,

What is the I frame interval you set for the encoder? To avoid this maximum fragment duration error, you would need to have at least one I frame every 10 seconds.

If you can send your log and video streaming use case to kinesis-video-support@amazon.com, we can provide further assistance.

-ZH

laurenkemperman, 2018-12-04:
> If you can send your log and video streaming use case to kinesis-video-support@amazon.com <mailto:kinesis-video-support@amazon.com>, we can provide further assistance.

Hi, 

The iframe interval has been set to 60. I am getting an error which states that there is no corresponding frame found. I will send the log and use case, thanks for your help! 

Lauren

MushMal, 2018-12-20:
@laurenkemperman thanks for the info. It seems that you are producing 15 fps stream. Having key-int-max 60 means you will produce GoP size of 12 seconds which is above the current max fragment duration limit. Try setting key-int-max to a smaller value like 45 or have FPS increased to ensure the fragment duration is less than 10 seconds

unicornss, 2018-12-21:
Closing the issue for now.
Please reopen or contact kinesis-video-support@amazon.com if needed. 
Thanks

closed 2018-12-21 by unicornss

tstamper, 2021-01-09:
This sounds inaccurate? how can GOP size of 60 with 15FPS produce an iframe in 12 seconds time duration? It sounds like its 5FPS not 15FPS. Was that a typo?

MushMal, 2021-01-09:
Yes indeed, 5 fps. The issue is again, you either have a low fps or your encoder blocks