awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #154: Greengrass Core Integration

richardzyx, 2018-12-05:
Hi,

Just wondering if there's a way or instruction on how to integrate KVS with Greengrass core? It seems like it would be supported by using Lambda executable, but the only tutorial[1] online requires a module from DeepLens for integrating with Python.

Is there any discussion or standardization around this topic? Thanks!

[1]: https://aws.amazon.com/blogs/machine-learning/video-analytics-in-the-cloud-and-at-the-edge-with-aws-deeplens-and-kinesis-video-streams/

unicornss, 2018-12-21:
Hi @richardzyx , 
Please contact kinesis-video-support@amazon.com for your specific use case/application.  
Thanks

closed 2018-12-21 by unicornss

jcornouiller, 2020-12-10:
Hi there,

have you got an answer for that?

Thanks