awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #155: public docker image

matti, 2018-12-07:
please publish the images - for example this https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docker_native_scripts/Dockerfile takes a long time to build

zhiyua-git, 2018-12-07:
Hi @matti ,

The images are already published. You can retrieve them following https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker-download.

Hope this helps.
-ZH

matti, 2018-12-07:
they are not public and it seems like this one https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docker_native_scripts/Dockerfile is not published

zhiyua-git, 2018-12-07:
You need to follow the authentication steps first in the document: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker-authenticate.

The docker file you referred is built and published as Ubuntu image: 
Download the Docker Image for Ubuntu
sudo docker pull 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest

matti, 2018-12-07:
yeah, I did see the authentication - would be more convenient to have fully public images.